JanM321/esphome-lg-controller #1: Pick and Placement Centroid file

dncbert, 2023-07-11:
Not really an issue... just not sure how else to reach out. 

I appreciate the work that you put into this project. I was able to follow your plans and build a breadboard version of this project. It works great although soldering the TI Lin Receiver pushed the boundaries of my soldering skills. 

I would like to order a more polished project board from PCBWay. I was able to use the gerber file in the repository for the board layout. I would also like them to install the components if possible. In that process, they requested a centroid file to delineate the location of the components. I am a total novice at this so I do not know how to generate the file. You mentioned in your writeup that you also had PCBWay build your board. Would you happen to have a Centroid file? And if so, are you willing to share it?

Thanks,

Daryl

closed 2023-07-11 by JanM321

JanM321, 2023-07-11:
> I appreciate the work that you put into this project. I was able to follow your plans and build a breadboard version of this project. It works great although soldering the TI Lin Receiver pushed the boundaries of my soldering skills.

I was hoping others would be able to use this so that's really great to hear. If you don't mind sharing, which LG AC model are you using it with?

> I would like to order a more polished project board from PCBWay. I was able to use the gerber file in the repository for the board layout. I would also like them to install the components if [...]

I added it as pcb-placement.csv. I also added pcb-bom.csv for the list of components. Thanks for asking, it is more complete this way.

JanM321, 2023-07-11:
Two more things:

* These CSVs were exported from EasyEDA Designer, the tool I used for the hardware design. I'm not sure if they're standard CSV files, but EasyEDA is pretty popular and PCBWay accepted them.
 
* PCBWay told me their supplier didn't have a lot of TSRN 1-2433 components in stock (when I ordered in May). If they can't get them anymore it's probably fine to use a TSR 1-2433 instead, but I haven't tested this.

dncbert, 2023-07-12:
Thanks for sharing the pick and placement file. I had managed to create the bom file from your parts list. 

I have a ten year old single unit LG heat pump with wall mounted indoor unit. Model# AS-W123MMM9. I added a wall thermostat (PREMTB10U) about eight years ago. I have replaced the wall thermostat with this esphome controller. 

I am curious whether you have the same thermostat and if so, have you tried running the esphome controller along side the wall thermostat. i.e. in some kind of master-slave configuration. I would interested in that setup to allow guests to make adjustments at the wall when needed.

jourdant, 2023-07-16:
Might be worth me opening another issue on this to discuss further - but I have a single AC unit outside ducted to multiple zones. Each zone has its own PREMTB001 thermostat. Ideally what I'd like to do is have the ESP32 sitting on the network/controller interface to reflect the state of each zone inside Home Assistant. On top of that I'd like to be able to inject control messages back to the relevant zones and show on the panels too.

From everything I've read in your repo and others, it seems that this should be possible. There may need to be some code changes required as you mentioned above for it running as a slave device for example. 

I'm also looking at merging your schematic with this schematic https://oshwlab.com/quickstar77/m5sensorshield so that I can build a single shield for an [M5 Atom Lite](https://shop.m5stack.com/products/atom-lite-esp32-development-kit) device. It's small and works nicely with esphome too.

Would you have any pointers on changes to the library (if needed) for multiple zones or anything else I've mentioned?

Really grateful for the work you've done here so far.

JanM321, 2023-07-16:
@jourdant Are most of your PREMTB001 thermostats for the zones in slave mode? Or can there be multiple master controllers if they're for separate zones?

I haven't paid a lot of attention to zones so far because I'm not familiar with them. [This blog post](https://www.instructables.com/Hacking-an-LG-Ducted-Split-for-Home-Automation/) and the comments have some information but there it sounds like they use one controller where each zone just has a boolean on/off flag? But you instead have one controller per zone where you select the zone in the settings?

I still have the setup to hook up my PREMTB001 to a laptop so I may poke at that a little today because now I'm curious how that mode affects the protocol :)

> I'm also looking at merging your schematic with this schematic https://oshwlab.com/quickstar77/m5sensorshield so that I can build a single shield for an [M5 Atom Lite](https://shop.m5stack.com/products/atom-lite-esp32-development-kit) device. It's small and works nicely with esphome too.

I'm using an M5 Atom Lite as ESPHome [bluetooth proxy](https://esphome.github.io/bluetooth-proxies/) for the temperature sensors and I've had the same thought. My current setup works great so it's hard to justify doing the work for me, but if you manage to get it working I'd love to hear about it.

jourdant, 2023-07-16:
Awesome! I'm new to building PCBs and I know it's a fairly ambitious project to start with, but given I have two references to work from and I just need to combine them - how hard can it be? 😅 I'll definitely make it all available when I've got it working.

I haven't had a chance to look at the zones and whether they are master or slave in the configuration. I need to open the installer mode and figure that out.

Am I right in thinking that we could also be impersonating other devices on the network of controllers to change settings? I need to do a bit more reading on the protocol to understand it more.

Also I know the built in thermistor per panel is not ideal, but I'd like to still collect that data too.

JanM321, 2023-07-16:
> Am I right in thinking that we could also be impersonating other devices on the network of controllers to change settings? I need to do a bit more reading on the protocol to understand it more.

Fortunately the protocol is pretty simple and forgiving in my experience. There are multiple message types and if a controller or the AC unit sends a message, all other devices on the same bus will update their state based on it. Worth checking if your controllers are on the same bus or separate ones (each connected to some kind of global controller or something).

I updated protocol.md with an undocumented zone setting in the settings menu. I'm not sure what this is for but I can set it to 1-3. From what I can tell zones are usually just open/closed though.

JanM321, 2023-07-16:
Oh it's also possible you just have multiple indoor units connected to a single outdoor unit. In that case each indoor unit has its own master controller and communication bus. That's similar to my multi split setup.

jourdant, 2023-07-16:
> Oh it's also possible you just have multiple indoor units connected to a single outdoor unit. In that case each indoor unit has its own master controller and communication bus. That's similar to my multi split setup.

Ohhh it could actually be this. It's ducted so I'm not actually sure what is in the walls. All I know is that I have a single outdoor unit outside with model number ARUN040GSS0. So potentially we're saying that there are separate indoor units with their own controllers (which in this case would all be masters), but they'd all be attached to the single outdoor unit?

JanM321, 2023-07-16:
Correct. A quick search suggests that's indeed a multi split outdoor unit.

Question is which controller protocol is used by the indoor units, the one implemented here or the one implemented [in this repo](https://github.com/Flameeyes/esphome-lg-pqrcuds0) because the PREMTB001 supports both. The hardware is the same so you could also just try both of them, but I don't know if the other protocol supports connecting multiple controllers.

commit a2b910b3 (2023-07-11): Add BOM and placement CSV files for PCB, fixes #1