JanM321/esphome-lg-controller #2: M5 Atom Lite target

jourdant, 2023-07-18:
@JanM321 I'll keep updating this thread going forward as I had pretty much hijacked the other closed issue. For reference in case anyone else is reading: #1.

I had another idea which is potentially a little ambitious (and completely out of my skill range). It would be cool to potentially add a lipo battery to the circuit and have it trickle charge over the 12v supply from the unit. I understand that the 12v on the interface is probably very sensitive and probably not designed to run even an ESP32 off it - but if we could trickle charge a battery and supplement the 12v interface that could be interesting. Alternatively a super small solar cell could potentially charge that battery slowly? What are your thoughts on powering this from the 12v line even without the battery? 

I have a test PREMTB001 that I wired up to a 12v power supply and also used a buck converter to 5v for the Atom lite. Unfortunately because there is no unit attached to it (and I haven't got the emulator hooked up to a TTL+LIN adapter) I haven't made any headway there other than being able to power them both.

I am not in the house with the multiple PREMTB001 panels yet, but should be in the next couple of weeks. So I can't easily test them to see if they are all in master configuration, or if this is some kind of master/slave + zone configuration. In the meantime, I am working on adding your design onto the layout of the M5 Atom Lite.

JanM321, 2023-07-18:
> What are your thoughts on powering this from the 12v line even without the battery?

My design does power everything from the 12V line.

I'm not sure how much current you can draw but I think it's possible to connect multiple controllers and the latest ones have color LCD screens (see PREMTB100) so one ESP32 should be fine. I haven't tested this with multiple controllers though. My firmware also doesn't enable the bluetooth proxy feature that ESPHome supports because I didn't want to complicate the design and use more power than necessary.

jourdant, 2023-07-30:
Bit of an update - my progress on learning PCB design isn't moving fast enough so for this one I thought I'd list a quick gig on fiverr (a risk I know 😅).  The requirements were to keep the SHT40 as in the reference size design (so it can be used instead of thermistor if you don't want an external temperature sensor in the room) and to merge in your design including the use of the 12v line for power.

I also had the JST-XH 3-pin socket added on there so that in theory you could build a splitter cable and use the existing plugs that the LG units are using.

I've had a look as much as I can at the spec and it looks ok, but I'm open to someone else with better PCB knowledge taking a look before I get a small prototype run done. 

[image]
[image]
[image]

jourdant, 2023-09-02:
Just an update - I've had some sample PCBs made up and set up a basic esphome config with:

```
uart:
  tx_pin: GPIO19
  rx_pin: GPIO22
  baud_rate: 104
  debug:
    direction: BOTH
    dummy_receiver: true
    after:
      delimiter: "\n"
    sequence:
      - lambda: UARTDebug::log_string(direction, bytes);
```

I haven't plugged it into the actual aircon unit yet, but I do have it plugged into my spare PREMTB001 that I have on a 12v power brick and I'm seeing data coming in from the LIN transceiver! I haven't made any particular sense of what's coming yet, but I am seeing the messages sent every couple of seconds or so. So far so good.

``` 
[14:14:53][D][uart_debug:158]: <<< "\x94\x00\x00\x00\x00\xC1"
[14:14:55][D][uart_debug:158]: <<< "\x94\x00\x00\x00\x00\xC1"
[14:14:57][D][uart_debug:158]: <<< "\x94\x00\x00\x00\x00\xC1"
[14:14:59][D][uart_debug:158]: <<< "\x94\x00\x00\x00\x00\xC1"
[14:15:01][D][uart_debug:158]: <<< "\x05\x14\"\x00 \x0E"
[14:15:03][D][uart_debug:158]: <<< "\x05\x15!\x00 \x0E"
[14:15:06][D][uart_debug:158]: <<< "\x05\x15\"\x00 \t"
[14:15:08][D][uart_debug:158]: <<< "\x05\x16\"\x00 \b"
[14:15:10][D][uart_debug:158]: <<< "\x05\x16r\x00 \xF8"
[14:15:12][D][uart_debug:158]: <<< "\x05\x16r\x00 \xF8"
[14:15:14][D][uart_debug:158]: <<< "\x05\x16r\x00 \xF8"
[14:15:16][D][uart_debug:158]: <<< "\x05\x172\x00 ;"
[14:15:18][D][uart_debug:158]: <<< "\x05\x172\x00 ;"
[14:15:20][D][uart_debug:158]: <<< "\x15\x17\x02\x00 \e"
[14:15:22][D][uart_debug:158]: <<< "\x15\x17\x02\x00 \e"
[14:15:24][D][uart_debug:158]: <<< "\x14\x17\x02\x00 \x18"
[14:15:27][D][uart_debug:158]: <<< "\x14\x17\x02\x00 \x18"
[14:15:29][D][uart_debug:158]: <<< "\x14\x17\x02\x00 \x18"
[14:15:31][D][uart_debug:158]: <<< "E\x17.\x00 \xFF"
[14:15:33][D][uart_debug:158]: <<< "E\x18.\x00 \xFE"
[14:15:35][D][uart_debug:158]: <<< "D\x18.\x00 \xFF"
[14:15:37][D][uart_debug:158]: <<< "D\x18.\x00 \xFF"
[14:15:39][D][uart_debug:158]: <<< "D\x18.\x00 \xFF"
[14:15:41][D][uart_debug:158]: <<< "5\x18)\x00 \xC3"
[14:15:43][D][uart_debug:158]: <<< "5\x18)\x00 \xC3"
[14:15:45][D][uart_debug:158]: <<< "4\x18)\x00 \xC0"
[14:15:47][D][uart_debug:158]: <<< "4\x18)\x00 \xC0"
[14:15:50][D][uart_debug:158]: <<< "%\x18)\x00 \xD3"
[14:15:52][D][uart_debug:158]: <<< "%\x18)\x00 \xD3"
[14:15:54][D][uart_debug:158]: <<< "$\x18)\x00 \xD0"
[14:15:56][D][uart_debug:158]: <<< "\x05\x182\x00 :"
[14:15:58][D][uart_debug:158]: <<< "\x05\x182\x00 :"
[14:16:00][D][uart_debug:158]: <<< "\x04\x182\x00 ;"
[14:16:02][D][uart_debug:158]: <<< "\x04\x182\x00 ;"
```

Next steps is to run add the c code in and update the esphome config to talk to the actual unit.

[image]

jourdant, 2023-09-03:
I've got it connected up to the HVAC now. I'm having power issues where the 12v line drops to ~9v when I plug it in (even without the panel also connected)... So for now doing testing with usb plugged in to help with the power supply issues. Open to any ideas here.

Regardless though, I've got messages coming back that match the format in the [protocol.md](https://github.com/JanM321/esphome-lg-controller/blob/58857923281c16042d5d8d1fb410b4346fd2f46a/protocol.md) file!

Here's an extract if anyone is interested in what I'm seeing from my unit:

### 0xC8 example:
```[D][uart_debug:196]: <<< 0b11001000 (0xC8) 0b00010000 (0x10) 0b00000000 (0x00) 0b00000000 (0x00) 0b01000000 (0x40) 0b00000000 (0x00) 0b00100101 (0x25) 0b00001101 (0x0D) 0b00000000 (0x00) 0b00000000 (0x00) 0b00000000 (0x00) 0b00000011 (0x03) 0b00011000 (0x18)```

### 0xCC example:
```[D][uart_debug:196]: <<< 0b11001100 (0xCC) 0b00101101 (0x2D) 0b00001001 (0x09) 0b00000000 (0x00) 0b00000000 (0x00) 0b00000000 (0x00) 0b00000000 (0x00) 0b00000000 (0x00) 0b00000000 (0x00) 0b00100000 (0x20) 0b00000000 (0x00) 0b00000000 (0x00) 0b01110111 (0x77)```

### 0xCF
```[D][uart_debug:196]: <<< 0b11001111 (0xCF) 0b00000000 (0x00) 0b00000000 (0x00) 0b00000000 (0x00) 0b00000000 (0x00) 0b00000000 (0x00) 0b00000000 (0x00) 0b00000000 (0x00) 0b00000000 (0x00) 0b00000000 (0x00) 0b00000000 (0x00) 0b00000000 (0x00) 0b10011010 (0x9A)```

^ any idea what 0xCF is so far?

JanM321, 2023-09-03:
> I've got it connected up to the HVAC now. I'm having power issues where the 12v line drops to ~9v when I plug it in (even without the panel also connected)... So for now doing testing with usb plugged in to help with the power supply issues. Open to any ideas here.

You seem to have a linear voltage regulator which isn't as efficient as a switching regulator / buck converter. No idea if that's it though.

> Regardless though, I've got messages coming back that match the format in the [protocol.md](https://github.com/JanM321/esphome-lg-controller/blob/58857923281c16042d5d8d1fb410b4346fd2f46a/protocol.md) file!

That's interesting, I don't think I've seen any CF/AF messages before. Is it sending these regularly? My best guess is they ran out of bits with the C8 or CC messages so started using CF.

jourdant, 2023-09-03:
Very true on the LDO, I'll be switching that back to a buck converter if I can find a decent SMT/SMD one. Any recommendations there? I'll be going 12v -> 3.3v this time as I'm also thinking of switching to ESP8285 for lower power too. I think it makes sense especially if I'm not going to be using the bluetooth on the ESP32.

Here's a more verbose log of the messages with 0xCF in the mix, I'll come back to these at some point once I have the base functionality working properly with the C8/CC messages first. I'm also going to do some testing as 0x2x rather than 0xAx.

These are in the order they were received:

```
11001000 (0xC8) 00010000 (0x10) 00000000 (0x00) 00000000 (0x00) 01000000 (0x40) 00000000 (0x00) 00100101 (0x25) 00001101 (0x0D) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000011 (0x03) 00011000 (0x18)

11001000 (0xC8) 00010000 (0x10) 00000000 (0x00) 00000000 (0x00) 01000000 (0x40) 00000000 (0x00) 00100101 (0x25) 00001100 (0x0C) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000011 (0x03) 00011001 (0x19)

11001100 (0xCC) 00101101 (0x2D) 00001001 (0x09) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00100000 (0x20) 00000000 (0x00) 00000000 (0x00) 01110111 (0x77)

11001100 (0xCC) 00101101 (0x2D) 00001001 (0x09) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00100000 (0x20) 00000000 (0x00) 00000000 (0x00) 01110111 (0x77)

11001111 (0xCF) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 10011010 (0x9A)

11001111 (0xCF) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 10011010 (0x9A)

11001000 (0xC8) 00010000 (0x10) 00000000 (0x00) 00000000 (0x00) 01000000 (0x40) 00000000 (0x00) 00100101 (0x25) 00001101 (0x0D) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000011 (0x03) 00011000 (0x18)

11001000 (0xC8) 00010000 (0x10) 00000000 (0x00) 00000000 (0x00) 01000000 (0x40) 00000000 (0x00) 00100101 (0x25) 00001101 (0x0D) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000011 (0x03) 00011000 (0x18)

11001100 (0xCC) 00101101 (0x2D) 00001001 (0x09) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00100001 (0x21) 00000000 (0x00) 00000000 (0x00) 01110110 (0x76)

11001100 (0xCC) 00101101 (0x2D) 00001001 (0x09) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00100001 (0x21) 00000000 (0x00) 00000000 (0x00) 01110110 (0x76)

11001111 (0xCF) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 10011010 (0x9A)

11001111 (0xCF) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 10011010 (0x9A)

11001000 (0xC8) 00010000 (0x10) 00000000 (0x00) 00000000 (0x00) 01000000 (0x40) 00000000 (0x00) 00100101 (0x25) 00001101 (0x0D) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000011 (0x03) 00011000 (0x18)

11001000 (0xC8) 00010000 (0x10) 00000000 (0x00) 00000000 (0x00) 01000000 (0x40) 00000000 (0x00) 00100101 (0x25) 00001101 (0x0D) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000011 (0x03) 00011000 (0x18)

11001100 (0xCC) 00101101 (0x2D) 00001001 (0x09) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00100001 (0x21) 00000000 (0x00) 00000000 (0x00) 01110110 (0x76)

11001100 (0xCC) 00101101 (0x2D) 00001001 (0x09) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00100001 (0x21) 00000000 (0x00) 00000000 (0x00) 01110110 (0x76)

11001111 (0xCF) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 10011010 (0x9A)

11001111 (0xCF) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 00000000 (0x00) 10011010 (0x9A)
```

JanM321, 2023-09-04:
> Very true on the LDO, I'll be switching that back to a buck converter if I can find a decent SMT/SMD one. Any recommendations there? I'll be going 12v -> 3.3v this time as I'm also thinking of switching to ESP8285 for lower power too.

I like the Traco Power ones. TSR 1-2433SM is a surface mount version. They don't require external capacitors or other components so they're very convenient and worth the money for me.

jourdant, 2023-09-08:
I've been trying to compile the code into the esphome firmware. So far just adding the includes lg-controller.h file.

```
In file included from src/main.cpp:30:
src/lg-controller-original.h:8:38: error: expected class-name before '{' token
 class LgSwitch final : public Switch {
                                      ^
src/lg-controller-original.h:16:29: error: 'SwitchRestoreMode' has not been declared
   void restore_and_set_mode(SwitchRestoreMode mode) {
                             ^~~~~~~~~~~~~~~~~
src/lg-controller-original.h:14:8: error: 'void LgSwitch::write_state(bool)' marked 'override', but does not override
   void write_state(bool state) override;
        ^~~~~~~~~~~
src/lg-controller-original.h: In member function 'void LgSwitch::restore_and_set_mode(int)':
src/lg-controller-original.h:17:5: error: 'set_restore_mode' was not declared in this scope
     set_restore_mode(mode);
     ^~~~~~~~~~~~~~~~
src/lg-controller-original.h:17:5: note: suggested alternative: 'uart_set_mode'
     set_restore_mode(mode);
     ^~~~~~~~~~~~~~~~
     uart_set_mode
src/lg-controller-original.h:18:22: error: 'get_initial_state_with_restore_mode' was not declared in this scope
     if (auto state = get_initial_state_with_restore_mode()) {
                      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Compiling /data/lounge-ac-1/.pioenvs/lounge-ac-1/libb70/ESPAsyncWebServer-esphome/AsyncEventSource.cpp.o
src/lg-controller-original.h: At global scope:
src/lg-controller-original.h:24:35: error: 'climate' has not been declared
 class LgController final : public climate::Climate, public Component {
                                   ^~~~~~~
src/lg-controller-original.h:24:44: error: expected '{' before 'Climate'
 class LgController final : public climate::Climate, public Component {
                                            ^~~~~~~
src/lg-controller-original.h:24:53: error: expected unqualified-id before 'public'
 class LgController final : public climate::Climate, public Component {
                                                     ^~~~~~
src/lg-controller-original.h: In member function 'void LgSwitch::write_state(bool)':
src/lg-controller-original.h:539:5: error: 'publish_state' was not declared in this scope
     publish_state(state);
     ^~~~~~~~~~~~~
src/lg-controller-original.h:539:5: note: suggested alternative: 'write_state'
     publish_state(state);
     ^~~~~~~~~~~~~
     write_state
src/lg-controller-original.h:540:16: error: invalid use of incomplete type 'class LgController'
     controller_->set_changed();
                ^~
src/lg-controller-original.h:24:7: note: forward declaration of 'class LgController'
 class LgController final : public climate::Climate, public Component {
       ^~~~~~~~~~~~
Compiling /data/lounge-ac-1/.pioenvs/lounge-ac-1/libb70/ESPAsyncWebServer-esphome/AsyncWebSocket.cpp.o
*** [/data/lounge-ac-1/.pioenvs/lounge-ac-1/src/main.cpp.o] Error 1
```

Ignore the references to lg-controller-original.h - that's just the untouched version I grabbed off GitHub. I'm not too familiar with the esphome API but when I read through the docs it looks like most of these errors should be non-issues? maybe a few namespaces to be added for climate for example? I'm running esphome as a container on my HA node.

JanM321, 2023-09-09:
@jourdant The command line instructions in the readme still work for me. I just updated my controllers to ESPHome 8.3 to test this.

Are you using everything in base.yaml? I think that affects what gets pulled into the global namespace.

bzumik1, 2023-10-05:
Hi, this looks really interesting, I hope you would be able to get this working, it would be really nice.

JanM321, 2023-11-08:
Note that there's now also a hardware-tiny/ PCB from @florianbrede-ayet. It's both smaller and cheaper.

bzumik1, 2023-11-08:
Where can I buy it?

bzumik1, 2023-11-08:
Thanks a lot!

jourdant, 2023-11-20:
@JanM321  just a quick update - I've also started trying a new path - integrating with the outdoor unit using the Central Control pins that use the LGAP protocol. I've been slowly figuring out what all the values mean and you can check it out here: https://github.com/jourdant/esphome-lgap. This is my current approach so that I can use a single off-the-shelf device to integrate with the outdoor unit and control 1-to-many zones. 

I really appreciate your help and sharing your designs for this project, if this is also of interest feel free to check it out!

JanM321, 2023-11-20:
> @JanM321 just a quick update - I've also started trying a new path - integrating with the outdoor unit using the Central Control pins that use the LGAP protocol. I've been slowly figuring out what [...]

It's really cool that you can control indoor units through the outdoor unit. It will be interesting to see what information is exposed about the outdoor unit and whether you'll be able to control it better. 

Nice to see that it uses the same checksum algorithm as the two indoor unit protocols (sum and then xor with 0x55).

I looked into communicating with the outdoor unit before I went with the CN-REMO controller, because my MU2R17 outdoor unit seemed to have limited options for this (unlike the bigger outdoor units).

jourdant, 2023-11-20:
> Nice to see that it uses the same checksum algorithm as the two indoor unit protocols (sum and then xor with 0x55).

Definitely encouraging to see the consistencies throughout!

> I looked into communicating with the outdoor unit before I went with the CN-REMO controller, but my MU2R17 outdoor unit seemed to have limited options for this (unlike the bigger outdoor units).

I've looked up the installer manual for your unit and you're in luck - your unit is actually compatible with the PI-485 communications board. Essentially for older outdoor units that don't have Central Control pins onboard, you can use the PI-485 board to present an RS485 LGAP interface for your unit. See here: https://www.usermanuals.au/lg/mu2r17ul0/manual?p=22 page 22.

JanM321, 2023-11-21:
> I've looked up the installer manual for your unit and you're in luck - your unit is actually compatible with the PI-485 communications board. Essentially for older outdoor units that don't have [...]

Thanks for checking, but I think those units are the bigger ones that support at least 3 indoor units. My MU2R17 is the 16k version and the MU2R15 is 14k. Starting with the MU3R19 (18k) the outdoor units get a bit more advanced. I think the duo-split units are a bit of an awkward combination of single-split and multi-split.

jourdant, 2023-11-22:
> Thanks for checking, but I think those units are the bigger ones that support at least 3 indoor units. My MU2R17 is the 16k version and the MU2R15 is 14k. Starting with the MU3R19 (18k) the outdoor [...]

After looking on the model number, I see what you're talking about now. That's really unfortunate!

I've spent a bit more time expanding on the LGAP protocol here: https://github.com/jourdant/esphome-lgap/blob/main/protocol.md, I've used the same format you have in this repo too. It's a good way to lay things out for seeing the relevant bytes etc.

JanM321, 2023-11-22:
Great. There are still some unknown bits in the indoor unit messages, so maybe this will help with that too.

JanM321, 2024-02-03:
I'll close this issue because you've moved on to the LGAP project :) The hardware-tiny PCB also addresses some of the same issues. Feel free to reopen or file a new issue if you want to get back to this.

closed 2024-02-03 by JanM321

Gunner251192, 2024-02-26:
> > Where can I buy it?
> > The production files are prepared for jlcpcb, but pcbway should work as well.
> > I ordered mine here:
> > https://cart.jlcpcb.com/quote/
> 
> * Upload the `zip` as gerber file, leave all settings default
> * Enable the `Assembly` option and the bottom of the _quote_ page and proceed
> * Upload the `bom.csv` for the bom file and the `positions.csv´ for placement / cpl
> * I already corrected the positions / especially the THT parts by hand, so at least for jlcpcb they should fit perfectly
> 
> For me it was ~42 USD for 5 pieces assembled + shipping + possibly customs

Hi All, Does someone have one assembled piece at spare which I can take over? I live in NL.
I tried to order them trough JLPC assembled but with your supplied files it is currently not possible since:

C3 | 22uF | CP_Elec_4x5.8 | EEEFPC220UARC40175822uF 16V 160mA@100kHz 850mΩ@100kHz ±20% SMD,D4xL5.8mm Aluminum Electrolytic Capacitors - SMD ROHS | 5 |   | Extended | $0.00 | 5 shortfall
-- | -- | -- | -- | -- | -- | -- | -- | --

The 22uF capacitors are not available at JLPCB, I don't know with wat to replace them, I also don't need 5 boards to be honest.

belly-89, 2024-02-28:
> > > Where can I buy it?
> > > The production files are prepared for jlcpcb, but pcbway should work as well.
> > > I ordered mine here:
> > > https://cart.jlcpcb.com/quote/
> > 
> > 
> > 
> > * Upload the `zip` as gerber file, leave all settings default
> > * Enable the `Assembly` option and the bottom of the _quote_ page and proceed
> > * Upload the `bom.csv` for the bom file and the `positions.csv´ for placement / cpl
> > * I already corrected the positions / especially the THT parts by hand, so at least for jlcpcb they should fit perfectly
> > 
> > For me it was ~42 USD for 5 pieces assembled + shipping + possibly customs
> 
> Hi All, Does someone have one assembled piece at spare which I can take over? I live in NL. I tried to order them trough JLPC assembled but with your supplied files it is currently not possible since:
> 
> C3	22uF	CP_Elec_4x5.8	EEEFPC220UARC40175822uF 16V 160mA@100kHz 850mΩ@100khz ±20% SMD,D4xL5.8mm Aluminum Electrolytic Capacitors - SMD ROHS	5	 	Extended	$0.00	5 shortfall
> The 22uF capacitors are not available at JLPCB, I don't know with what to replace them, I also don't need 5 boards to be honest.

Hi, Yes i have a spare PCB, had to order 5 pieces at once and only needed two. So i can send you one. Send me a message to make arrangement. 

Thanks for mentioning me @JanM321

Gunner251192, 2024-02-28:
> Hi, Yes i have a spare PCB, had to order 5 pieces at once and only needed two. So i can send you one. Send me a message to make arrangement.
>
> Thanks for mentioning me @JanM321

Hi [belly-89](https://github.com/belly-89),

Please let me know how I can reach out to you.

Regards Nick.

belly-89, 2024-02-28:
> Hi [belly-89](https://github.com/belly-89),
>
> Please let me know how I can reach out to you.
>
> Regards Nick.

Are you on tweakers.net? You can send me an PM there, username belly89

belly-89, 2024-04-03:
> Hey @belly-89 Do you happen to still have one of these PCBs laying around? I would like to buy one of them if you still do. Thanks!

No sorry, I don't have any PCB boards left.

jourdant, 2024-04-26:
@JanM321 you've made so much awesome progress on your integration! I've got my LGAP protocol fully working now as an external component presenting a climate interface - using off the shelf parts. Feel free to check it out sometime, interested in any feedback, especially if you have any suspicions on the protocol, there's still a few bytes I haven't mapped that may look familiar to you 🙂 https://github.com/jourdant/esphome-lgap

JanM321, 2024-04-28:
> I've got my LGAP protocol fully working now as an external component presenting a climate interface - using off the shelf parts.

Nice work. I like the HA screenshot showing multiple climate entities :)

> especially if you have any suspicions on the protocol, there's still a few bytes I haven't mapped that may look familiar to you

Nothing stands out. It might be worth exposing values to HA with a temporary sensor so you can see if there’s a pattern. Pipe temperature values in particular should stand out that way - they might use a similar encoding to the one I reverse engineered from the indoor unit controllers.

jbutlerdev, 2024-08-23:
> The production files are prepared for jlcpcb, but pcbway should work as well. I ordered mine here: https://cart.jlcpcb.com/quote/
>
> * Upload the `zip` as gerber file, leave all settings default
> * [...]

Is it possible to get an updated bom? It seems like some parts are no longer available and I'm not sure what substitute to pick.

reconfigrr, 2025-01-05:
I am also hoping for an updated BOM. Im not an electrical engineer, but I have attempted to find some substitutes for the parts no longer available. Hoping someone might be able to confirm if these are ok.

VZH100M1CTR-0405 replacing EEE-FC1C100AR
0402B332K500NT replacing CL05B332KB5NNNC

Any help would be appreciated.

mindless2, 2026-01-12:
Anyone have a spare custom PCB board??

artik246a, 2026-10-22:
Cześć, czy ma ktoś na zbyciu jedną płytkę PCB z wysyłką do Polski?

artik246a, 2026-10-22:
> 31\. Cześć, czy ktoś ma na sprzedaż płytki PCB z możliwością wysyłki do Polski? A jeśli ktoś chce kupić jedną z tych płytek, to czy mogę zamówić kilka i odsprzedać cztery?

Jestem zainteresowany zakupem 1 szt.

gittmf, 2026-10-22:
> requests on the go with GitHub Mobile for iOS and Android. Download it today!

Posiadam z wysyłką z Polski. Najlepiej do paczkomatu. 70 blik + przesylka
TMF Poland
Maciej Filipowicz
http://www.tmf.net.pl/
tel 603724716

piot84zet, 2026-10-26:
> 28\. Hi, does anyone have PCBs for sale with shipping to Poland? or if anyone wants to buy one of these boards, can I order some and resell four of them?

Hello, did you bought it? jestem zainteresowany odkupem :)

artik246a, 2026-11-08:
Napisałem do gittmf czekam na odpowiedź

andrzejboron, 2026-11-08:
Mam 2 rodzaje pcb

artik246a, 2026-11-08:
potrzebuję kontroler wifi do klimatyzatora LG S24EQ.NSK

artik246a, 2026-11-08:
szukam płytki do kontrolera LG z projektu esphome-lg-controller (JanM321). Chcę wariant "hardware-tiny" pod ESP32 DevKitC-32E, najlepiej już zmontowany (z transceiverem TLIN1027 i przetwornicą), ale rozważę też inną płytkę, jeśli jest z nią zgodna.

Proszę o informację:
1. Które z Pana dwóch płytek to wariant "tiny" pod DevKitC-32E?
2. Czy jest zmontowana, a jeśli tak, jaki ma transceiver (TLIN1027 / TJA1027)?
3. Cena, wysyłka do Polski (najlepiej paczkomat) i forma płatności.
4. Czy płytka była sprawdzona na jakiejś klimie LG, a jeśli tak, to na jakim modelu? Mam jednostkę S24EQ.NSK

kleczarp, 2026-11-13:
Cześć @gittmf  @andrzejboron @artik246a 

szukam 1 szt. płytki z wysyłką w Polsce (najlepiej paczkomat).

1. Czy to wariant "hardware-tiny" pod ESP32 DevKitC-32E?
2. Czy płytka jest zmontowana (TPS82130 + TJA1027)?
3. Czy złącze JST-XH i listwy 1x19 są przylutowane?
4. Jaka jest cena z wysyłką i forma płatności?

Płytkę chcę użyć do pompy ciepła LG Therma V (jednostka NK4), nie do klimatyzatora. Modbus na mojej jednostce nie działa, więc chcę spróbować przez złącze CN-REMO sterownika przewodowego. Jeśli ktoś już próbował z Therma V, chętnie dowiem się, czy działało.