JanM321/esphome-lg-controller #3: Current power consumption

drbugfinder, 2023-08-17:
This isn't really an issue, more of a question:
Is the current power consumption of the air conditioning unit also sent via the protocol (as it can also be seen in the app)? Or is this data not transmitted over the bus?

Thank you for the great work 👍 - I am facing the same problem of local control of three S12ET air conditioners and would like to connect them via MQTT. This is a great base to build on! PCBs are already ordered... 😃

drbugfinder, 2023-08-17:
My bad... looks like it's being transmitted, as you wrote: 

https://github.com/JanM321/esphome-lg-controller/blob/main/protocol.md#type-0xc-0xac0xcc

**Type 0xC (0xAC/0xCC)**
`bytes 3-5: energy usage 06:49, 12.9 kWh`

In the app it is reported to the exact Watt. Are you sure about the field length / additional scaler filed etc.?

drbugfinder, 2023-08-17:
That's good news! I couldn't find any information about the power consumption information in any of the adjacent repos...
(But it's important information for me - otherwise I could simply use an IR blaster)

> That part of my notes is based on me sending random data to the LG controller from my laptop and checking how it affects things.
> 
> >
Do you maybe have more information to share? I wonder how you were able to interpret the information - did you then see the consumption in the app that you sent on the bus?

I will keep you updated about that feature and the overall support for the S12ET units. I can't wait to get my hands on that project

JanM321, 2023-08-17:
For measuring power consumption of my AC I use a HomeWizard socket. It integrates with Home Assistant and is similar to a Shelly plug.

I just checked the LG PREMTB001 manual and I think `06:49, 12.9 kW` might be 64912.9 kWh? If I have some time the coming weeks I could look into this a bit more if needed. I can still connect the LG PREMTB001 to my laptop so it should be pretty easy to figure this out if you see these messages on the bus.

drbugfinder, 2023-08-17:
I see. I wasn't aware of the feature of the PREMTB001 to display the total power consumption. I'm rather looking for a momentary consumption, but it's better than nothing. Maybe there is more? What's with bytes 6-11 for example... 🤔

I don't really like to add another component for power measuring, if the unit itself already has this built-in, so that's one reason why I want to go this road.

JanM321, 2023-10-07:
@drbugfinder did you make any progress on this?

drbugfinder, 2023-10-13:
Hi @JanM321,

I apologize for my delayed response; I've been quite busy lately and haven't had the opportunity to delve deeper into the topic. However, shortly after my last message, all the necessary components and PCBs arrived, and I painstakingly hand-soldered everything together for my three AC units.

As a side note, I recommend using a larger format C1 capacitor to have at least a fair chance for hand-soldering :-) 

The TSR1-2433 has been performing well on all units.

What I've observed in the past few weeks of use is that when you turn the unit off, it often automatically turns back on due to the last set or last known state from the ESP module. So, in my case, I can only reliably control them via the ESP module.

I will keep you updated. 
[image]
[image]

JanM321, 2023-10-13:
> I apologize for my delayed response; I've been quite busy lately and haven't had the opportunity to delve deeper into the topic. However, shortly after my last message, all the necessary components and PCBs arrived, and I painstakingly hand-soldered everything together for my three AC units.

No worries :)

> The TSR1-2433 has been performing well on all units.

Nice. I was expecting this but it's good to have confirmation.

> What I've observed in the past few weeks of use is that when you turn the unit off, it often automatically turns back on due to the last set or last known state from the ESP module. So, in my case, I can only reliably control them via the ESP module.

I haven't seen this yet with my units. When you turn the unit off, it should almost immediately send a message to the controller which will then update its state. I wonder if your S12ET units have more of a delay before sending.

drbugfinder, 2023-10-13:
> I haven't seen this yet with my units. When you turn the unit off, it should almost immediately send a message to the controller which will then update its state. I wonder if your S12ET units have more of a delay before sending.

It appears to be a timing issue to me. I've noticed a delay with the PREMTB001 as well. Interestingly, when I use the PREMTB001, the delay is present (~ 1-4 seconds), but the unit maintains the last set state without any problems. This is not a major issue for me since I intend to solely use the ESP modules for individual control. However, there is certainly something different.

drbugfinder, 2024-01-21:
@JanM321 Is there polling needed to get CC or CF messages? https://github.com/JanM321/esphome-lg-controller/blob/main/protocol.md#type-f-0xaf0xcf
Or will they be regularly send by the AC, if it supports them? 
I think I have seen CC messages in the past, but I cannot find the old logs at the moment. I'm still interested in getting the power consumption from the units...

landrysik, 2024-03-06:
Hi I have single-split LG Standard Plus PC12SK. I just got the hardware up and running and was interested in seeing the power consumption aswell. Do I understand correctly that if I don't see any CC or CF message in a log (checked last 20minutes), there is no chance of getting the power consumption? I only see messages A8... and C8....
I was naively thinking that the wired remote will show at least all the data the wifi module transmitts over to LG Thinq app...

JanM321, 2024-03-07:
> Do I understand correctly that if I don't see any CC or CF message in a log (checked last 20minutes), there is no chance of getting the power consumption?

That's correct, as far as I know.

>  I was naively thinking that the wired remote will show at least all the data the wifi module transmitts over to LG Thinq app...

No that's not true unfortunately. Two very different protocols and both support a different set of features. To support all the features, the controller would have to connect to the WIFI port too...

dzungpv, 2024-07-23:
Any update on this? @drbugfinder you find something new?