JanM321/esphome-lg-controller #4: pcbway ordering trouble

marcvandam2, 2023-08-20:
Hello, 

First of all thanks for all the research in this project! 

Here in Netherlands in the winter of 2022-2023 I used a LG AP12RK airco to partly heat our livingroom. This year I wanted to improve the setup and found this project, Getting the annoying beep away, and not being cloud dependent are high on my list. 

I also own a PREMTB001 controller and also now have the colored version.

Now I'm trying to order the PCB from pcbway but run into trouble. 

What is the size of the PCB? 
[image]

If I just fill in rest at pcbway "as default" and include gerber zip file and pcb-bom and pcb-placement all seems fine. Are there extra changes needed when ordering?

I will separately order the Espressif ESP32-DevKitC-32E for example from [Conrad](https://www.conrad.nl/nl/p/espressif-esp32-devkitc-32e-printplaat-voor-ontwikkelaars-esp32-devkitc-32e-2814007.html?utm_source=google&utm_medium=surfaces&utm_campaign=shopping-feed&utm_content=free-google-shopping-clicks&utm_term=2814007&adcampaign=google&tid=17210340624_pla-2814007&gad=1&gclid=Cj0KCQjw84anBhCtARIsAISI-xcWmBnApJ7C1fyFsQwrwcNUp7Ijn1S-q-OLFNaMr46uQ5dvjj-t7PoaAoGlEALw_wcB)

JanM321, 2023-08-20:
@marcvandam2 Goedemiddag :) Try this page: https://www.pcbway.com/QuickOrderOnline.aspx

This will use the dimensions from the gerber files. IIRC I didn't change any of the default values on that page.

JanM321, 2023-08-20:
> I will separately order the Espressif ESP32-DevKitC-32E for example from [...]

I used DigiKey for this project and they've been fast and reliable. They sell these for less than 9 euros (but I don't remember how much the shipping costs were). This matters less if you need just one controller though.

> I also own a PREMTB001 controller and also now have the colored version.

I've been thinking about getting a PREMTB100 for reverse engineering the remaining parts of the protocol, but it has been hard to justify spending more money (and time..) on this project considering that what I have now works well for me...

Please let me know how it goes :) I should probably add a PCBWay section to the README because you're not the first one to ask about this.

drbugfinder, 2023-08-21:
I used jlcpcb for ordering the PCBs and the measurements were automatically set when I uploaded the gerber files. Still waiting for the delivery. The cost there were 5,50€ (including EuroPacket shipping and VAT).

I was absolutely impressed of Digikey, where I ordered the parts - it took only two (!!) days from ordering in USA to hold it in my hands in Germany. I want to build this for three a/c units, so I exactly reached the 50€ mark for free shipping. I ordered the TSR1-2433 instead of the TSRN1-2433, because the TSRN was not available. 
I ordered the ESP32-DevKitC-32E there, too. They were 8,84€ each (w/o VAT).

The main reason for ordering at Digikey for me was that the LIN transciever was only available there and I couldn't find it at Conrad or Reichelt or anywhere locally.

JanM321, 2023-10-07:
@marcvandam2 Were you able to order the PCBs?

marcvandam2, 2023-10-08:
Yes, sorry for my late response.

The pcb was delivered and works really well!
I put it in airco nicely with a plastic sealing. 

I found two issues with it:

The setup is set to have external home assistant sensor. 
The temperature reported by airco even with that somehow still reports the internal airco temp sensor. This temp comes in from serial line of airco and is written to home assistant. Because of this home assistant reports wide temperatures, as it constantly switches to ‘home assistant temp’ and temp coming from airco. I also looked closely to temperature reported in LG app is constant the home assistant sensor. I think will do some extra test in this area later also with PREM controller.

Second thing I found was. Somehow rhe home assistant LG integrations offers to put airco in 2 extra modus: LOW-MID and MID-HIGH. With 20% steps until 100% so to say. PREM controller doesnt offer that. 

But overall this product is already as it is really usefull! This winter the airco in our livingroom will be switching without any noise. Super! Really nice. I have build complex home assistant automations to control the airco and extra convector heaters.

JanM321, 2023-10-08:
Thanks for the feedback! I'm glad it works and is useful.

> The setup is set to have external home assistant sensor. The temperature reported by airco even with that somehow still reports the internal airco temp sensor. This temp comes in from serial line of [...]

That's interesting. I don't see that with my units: if I set it to use an external sensor, I don't see the value of the internal one anywhere (I have to change the setting temporarily to see it).
 
> Second thing I found was. Somehow rhe home assistant LG integrations offers to put airco in 2 extra modus: LOW-MID and MID-HIGH. With 20% steps until 100% so to say. PREM controller doesnt offer that.

Yeah. My PREMTB001 supports LOW-MID and MID-HIGH values but only if the unit supports it. Unfortunately my units (incorrectly) report that they don't. I tried sending the value for these settings anyway but the unit doesn't accept it.

JanM321, 2023-11-04:
The reported temperature issue might be fixed with the latest version of the code.

JanM321, 2024-02-03:
Closing this issue because the PCBs arrived months ago :) I also improved the README a bit.

closed 2024-02-03 by JanM321