JanM321/esphome-lg-controller #5: LG FM19AH Inverter

111lisu, 2023-09-21:
hello, do you have any information or experience on how to start the LG FM19AH Inverter outdoor unit from ESPHome / modbus ...
Without indoor unit?

JanM321, 2023-09-21:
> hello, do you have any information or experience on how to start the LG FM19AH Inverter outdoor unit from ESPHome / modbus ... Without indoor unit?

I know it's possible to connect to some outdoor units with the LGAP protocol, but I don't know anything about this.

111lisu, 2023-09-21:
The control is by one wire as in your solution +10V and 2 transoptors for each channel (3 chanel multisplit / 2006 year) it only needs heating at max (buffor). I could control the rest differently, but I don't want to modify the control of this inverter ( FSAM20SM60A & IGBT ).
id you the control codes from some documentation, or did you observe communication between the units?

JanM321, 2023-11-17:
I just looked at the communication from the wall controller and the indoor unit. I don't know anything about the outdoor unit's communication protocol.

closed 2023-11-17 by JanM321