JanM321/esphome-lg-controller #6: My CN-REMO has no socket

kozfelipe, 2023-10-18:
[image]

As you can see my CN-REMO has no JST-XH interface. Should I assume my board has no wired controller support or should I try soldering a socket?

EAX35907219 PCB

JanM321, 2023-10-18:
> As you can see my CN-REMO has no JST-XH interface. Should I assume my board has no wired controller support or should I try soldering a socket?

Soldering a socket might work, but I'm not sure. Someone else ran into the same issue, [see here](https://community.home-assistant.io/t/lg-ac-wired-controller-integration-via-esphome-esp32/582954/2).

kozfelipe, 2023-10-18:
oh nice,
It seems those pins aren't dead at all. If it throws a 12VAC feed is a good news
thanks for replying, ill share my solution if I decide to move on

closed 2023-10-18 by kozfelipe

kozfelipe, 2023-11-27:
@dzungpv thank you for your suggestion. Unfortunately the main board is a pain to remove and it would break the foil seal in a attempt to revive those pins. I manage to control it using IR bridge. It is not the best control but it will workaround until I get my official AC.