JanM321/esphome-lg-controller #8: Is there a file for the case ???

mshrem, 2023-10-24:
Hello first of all wow , amazing work. 

A long time ago i decided to just drop the idea of controlling my AC because the inverter on my bedroom doesnt support the LG wifi module. 

I am so pumped that you made this work, i already submitted the files to PCBway so they can build and assemble for me , however i was wondering if you happen to have the file for the accrylic case ? I kinda want to have everything made by them and i do not have any 3d printer . 

Thanks

JanM321, 2023-10-24:
It's just an Adafruit 2359 Raspberry Pi case :) When I designed the PCB I was hoping it'd fit in a Pi case and this one works well.

It's also in the list of materials in hardware/materials.md

mshrem, 2023-10-24:
Thanks , i somehow missread that part. 

One last thing i am a bit confused , do we have to power the unit itself with USB C and then connect to the HVAC ???? or does it use the power from the HVAC unit ?

JanM321, 2023-10-25:
> One last thing i am a bit confused , do we have to power the unit itself with USB C and then connect to the HVAC ???? or does it use the power from the HVAC unit ?

It's powered by the HVAC unit. You only need the USB port to flash the firmware on it the first time before connecting it to the AC.

mshrem, 2023-11-07:
So my pcbs are finally done but i think some of the orientation of the smds might be wrong . 

Can someone take a look at this? 

[image]

florianbrede-ayet, 2023-11-11:
Do you think the R values are incorrect?
Because otherwise resistors have no polarity, their orientation doesn't matter.

mshrem, 2023-11-16:
I am patienly waiting now for the order to be delivered , i went ahead flashed the firmware already on both of my esphomes, they are recognized in home assistant however i cannot add them in the ESPhome addon , i want to manage them in the esphome addon from home assistant for easier updates and such. 

Ps i also orde your pcbs @florianbrede-ayet , they arrive soon

mshrem, 2023-11-17:
Finally everything is here altho i am encountering some issues. 

I am attaching some pictures i know that the cables do not match , but i think i got it correctly , the espmodule doesnt even come on. 

[image]
[image]

JanM321, 2023-11-17:
You need to switch the red and yellow wires on the PCB.

If you look at the colors of the wires *from the AC* it should be (from left to right on your picture) Red, Yellow, Black. Because the cable has different colors, it should be Yellow, Black, Red from left to right in the picture you posted.

I should have added some labels to the PCB about the colors, but my first goal was to get it working at all and back then I didn't expect this PCB to become so popular.

mshrem, 2023-11-18:
I have tried to swit many cables around and nothing happens , the esp light doesnt even come on , i can power the esp with a USB cable without problems and its recognized already in HA. Does the led on the ESPmodule is supposed to come on ? 

[image]

Maybe the CN-REMO connector is not plugged into the board ? 

[image]
[image]

JanM321, 2023-11-18:
> I have tried to swit many cables around and nothing happens , the esp light doesnt even come on , i can power the esp with a USB cable without problems and its recognized already in HA. Does the led on the ESPmodule is supposed to come on ?

The LED will be off when connected to the AC. It will only be on when connected to USB.

mshrem, 2023-11-18:
> > 
> 
> The LED will be off when connected to the AC. It will only be on when connected to USB.

Yup ,so stupid of me LOL i am kinda newbie to this esp world , just figured it out , works perfectly however commands take a few seconds to arrive to the unit when pressing them on the HA app is this the same behaviour for you ??

JanM321, 2023-11-18:
> Yup ,so stupid of me LOL i am kinda newbie to this esp world , just figured it out , works perfectly 

Great!  🎉 

> however commands take a few seconds to arrive to the unit when pressing them on the HA app is this the same behaviour for you ??

Yes that's expected. The LG communication protocol is slow (sending a complete message takes at least a second) and the controller internally "updates" every 6 seconds so there can be a delay of about 1-6 seconds depending on this 6 second cycle.

mshrem, 2023-11-18:
> 
> Yes that's expected. The LG communication protocol is slow (sending a complete message takes at least a second) and the controller internally "updates" every 6 seconds so there can be a delay of about 1-7 seconds depending on this 6 second cycle.

Awesome job man , i cant thank you enough for all of the help , would be amazing if we can just use esphome addon on home assistant to just update the firmware , guess i need to learn more linux lol

closed 2023-11-18 by mshrem

JanM321, 2023-11-18:
> Awesome job man , i cant thank you enough for all of the help

No problem! It's great to see people all over the world using this. If you don't mind sharing, what kind of LG units do you have? I have a list of supported units in the README.

> would be amazing if we can just use esphome addon on home assistant to just update the firmware , guess i need to learn more linux lol

It may be possible to upload the firmware within Home Assistant and/or the ESPHome dashboard / add-on, but I don't really know this part of HA/ESPHome. When I was writing the software it was just easiest to use the command-line tools for testing. Maybe someone else reading this knows more about it.

mshrem, 2023-11-18:
> No problem! It's great to see people all over the world using this. If you don't mind sharing, what kind of LG units do you have? I have a list of supported units in the README.
> 
Yup , i have the [ARNU18GSCR4](https://www.lgbusiness.it/climatizzazione/altri-codici/arnu18gscr4/) , which worked 

Also works on the [ARNU36GSVA4](https://www.lgbusiness.es/climatizacion/multi-v/interiores/pared/pared-standard/arnu30gsva4-arnu36gsva4/)

Altho i broke the pin on one of the JSX connectors so i need to wait until monday to fix it since i am away for the weekend. I only have 1 working so far but tested on both devices and it worked. 

> It may be possible to upload the firmware within Home Assistant and/or the ESPHome dashboard / add-on, but I don't really know this part of HA/ESPHome. When I was writing the software it was just [...]

Will be a plus but its not necessary so happy that i did not purchased the LG wifi module , plus the ARTCOOL Mirror doesnt even support it now i need to get some temperature sensors so i can automate the climate to always be on the perfect temperature , pumped about this.

JanM321, 2023-11-18:
Wow, the ARNU36GSVA4 is very powerful.

I added these to the README; maybe it will help someone with similar units.