JanM321/esphome-lg-controller #10: Bug In Temperature Reporting

florianbrede-ayet, 2023-11-03:
I noticed that my LG 4-Way-Cassette is always reporting the internal thermistor value in `C9`.
My Artcool Gallery behaves differently and responds with the value set by the controller.

This is a display issue though, the Cassette reacts correctly to whatever I set the *room temperature* in both `Controller` and `2TH` mode.
I guess it just keeps reporting its internal sensor to display it in PREMBT001 along with the controller sensor:

```
[23:06:26][D][lg-controller:371]: sending A8.12.00.08.40.00.12.8F.00.00.00.00.F6 (13)
[23:06:26][D][climate:380]: 'LG Schlafzimmer' - Sending state:
[23:06:26][D][climate:383]:   Mode: HEAT
[23:06:26][D][climate:388]:   Fan Mode: LOW
[23:06:26][D][climate:400]:   Swing Mode: OFF
[23:06:26][D][climate:403]:   Current Temperature: 17.50°C
[23:06:26][D][climate:409]:   Target Temperature: 17.00°C
[23:06:26][W][component:214]: Component <unknown> took a long time for an operation (0.54 s).
[23:06:26][W][component:215]: Components should block for at most 20-30ms.
[23:06:31][D][lg-controller:525]: update
[23:06:31][D][lg-controller:387]: received A8.12.00.08.40.00.12.8F.00.00.00.00.F6 (13)
[23:06:31][D][lg-controller:396]: verified send
[23:06:37][D][lg-controller:525]: update
[23:06:50][D][lg-controller:371]: sending A8.12.00.08.40.00.12.8F.00.00.00.00.F6 (13)
[23:06:50][W][component:214]: Component <unknown> took a long time for an operation (0.51 s).
[23:06:50][W][component:215]: Components should block for at most 20-30ms.
[23:06:55][D][lg-controller:525]: update
[23:06:55][D][lg-controller:387]: received A8.12.00.08.40.00.12.8F.00.00.00.00.F6 (13)
[23:06:55][D][lg-controller:396]: verified send
[23:07:01][D][lg-controller:525]: update
[23:07:13][D][lg-controller:582]: line busy, not sending yet
[23:07:19][D][lg-controller:525]: update
[23:07:19][D][lg-controller:582]: line busy, not sending yet
[23:07:25][D][lg-controller:525]: update
[23:07:25][D][lg-controller:387]: received C8.12.00.08.40.00.12.9A.00.00.00.00.9B (13)
[23:07:25][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[23:07:25][D][climate:380]: 'LG Schlafzimmer' - Sending state:
[23:07:25][D][climate:383]:   Mode: HEAT
[23:07:25][D][climate:388]:   Fan Mode: LOW
[23:07:25][D][climate:400]:   Swing Mode: OFF
[23:07:25][D][climate:403]:   Current Temperature: 23.00°C
[23:07:25][D][climate:409]:   Target Temperature: 17.00°C
[23:07:25][D][climate:380]: 'LG Schlafzimmer' - Sending state:
[23:07:25][D][climate:383]:   Mode: HEAT
[23:07:25][D][climate:388]:   Fan Mode: LOW
[23:07:26][D][climate:400]:   Swing Mode: OFF
[23:07:26][D][climate:403]:   Current Temperature: 23.00°C
[23:07:26][D][climate:409]:   Target Temperature: 17.00°C
[23:07:26][D][lg-controller:387]: received C8.12.00.08.40.00.12.9A.00.00.00.00.9B (13)
```

My fix for this is simple:
https://github.com/florianbrede-ayet/esphome-lg-controller/commit/b3d24963acff562d88caefb8056cbf7a7d67a934

Since I already added some functionality (hysteresis to fix the stupid LG behaviour in heating mode, setting installer messages in yaml, configurable setpoint thresholds), I wasn't sure if you're interested in a full PR.

JanM321, 2023-11-04:
Hm this sounds similar to the behavior the reporter in issue #4 was seeing. I actually wish my units would still report the internal temperature... I can land your fix.

> Since I already added some functionality (hysteresis to fix the stupid LG behaviour in heating mode, setting installer messages in yaml, configurable setpoint thresholds), I wasn't sure if you're interested in a full PR.

In case it helps: a few days ago I set installer setting 15 (heating temperature thresholds) to 4 (-0,5C/+0,5C). This makes my units work a lot more like how I want them to. In HA I use a filter sensor to average the temperature over 10 minutes to make it not turn on/off too aggressively.

closed 2023-11-04 by JanM321

florianbrede-ayet, 2023-11-04:
I need to experiment more with the overheating.
The artcool gallery is complete design failure, its layout makes it impossible to work well for heating.

It has an airflow loop from the side vents to the intake.
It would help to control them individually, but that seems to be unsupported.

When I set it to `4`, it won't heat anymore, because the ambient heat from the copper pipes is enough to heat the thermistor above 20°C at all times even with the fan running.

Do you know if the installer setting `(0: default for unit, 1: +4C/+6C, 2: +2C/+4C, 3: -1C/+1C, 4: -0.5C/+0.5C)` should apply to both unit and external controller thermistor?

Because by default, the device overheats by at least `6°C`. 
As soon as I set it to external controller however, it stops at `>+2°C` without changing the setting above.

JanM321, 2023-11-04:
> Do you know if the installer setting `(0: default for unit, 1: +4C/+6C, 2: +2C/+4C, 3: -1C/+1C, 4: -0.5C/+0.5C)` should apply to both unit and external controller thermistor?
>
> Because by default, [...]

My units behave the same in this regard with internal or external sensor. The default setting for wall units is +1/+3, so when I started using an external sensor I had to set it to 19C to make it turn on at 20C and off at 22C. Changing this installer setting to 3 or 4 fixed that; I can now set it to 20.5 or 21.

My wall units also aren't ideal for heating, especially for my unit that's near a corner. Fortunately the external sensor fixes most of that.

commit dedbfe83 (2023-11-04): Ignore room temperature from unit if using external sensor. Fixes #10