JanM321/esphome-lg-controller #11: Capabilities & Stock Installer Messages For LG Artcool Gallery & 4-Way-Cassettes (Multi)

florianbrede-ayet, 2023-11-03:
These are my Artcool Gallery and 4-Way-Cassette messages for anyone interested:

**LG ARTCOOL GALLERY (Multi) ORIGINAL MESSAGES:**
```
[21:38:26][D][lg-controller:363]: received C9.84.E8.5F.82.71.00.84.13.60.04.81.56 (13)
[21:38:26][D][lg-controller:363]: received CA.00.00.00.00.00.00.00.00.12.F1.21.BB (13)
[21:38:32][D][lg-controller:363]: received CB.80.00.64.68.FF.00.00.40.04.00.00.0F (13)
```

If anyone with an Artcool Gallery stumbles upon this, `23` is the minimum LOW fan configuration that doesn't keep the compressor from stopping (overheat) and is a lot more bearable than the default fan noise:
```
AA.00.00.23.28.36.00.00.00.12.F1.21.xx
```

**LG 4 WAY CASSETTE (Multi) ORIGINAL MESSAGES:**
```
[21:54:36][D][lg-controller:387]: received C9.B9.EB.3F.BB.30.20.84.F7.D7.C4.A3.25 (13)
[21:54:36][D][lg-controller:387]: received CA.00.00.00.00.00.00.13.61.00.F1.01.65 (13)
[21:54:36][D][lg-controller:387]: received CB.80.00.77.73.00.00.00.01.01.40.00.22 (13)
```

JanM321, 2023-11-04:
Thanks for sharing these. I should probably add this to protocol.md.

The CA message from your Artcool is exactly the same as the one from my wall units. Good to know that you were able to set the fan speeds. Because my units send zeros for these bytes, I was thinking they don't support this setting but it'd probably work then. I wish I could see the current values before changing them, though...

The CB message: interesting that both have byte 1 set to 0x80, I haven't seen that bit being set yet. Bytes 3-4 are similar for all messages I've seen so far: maybe some kind of sensor values but I don't know what it is. Byte 5 is 0xFF for my wall units and your Artcool, but your cassette has it set to 0. No idea what it is. Your units also send model information, mine just send zeros for these bytes.

florianbrede-ayet, 2023-11-04:
@JanM321 do you have plans to use `C9` for feature selection?
I'm thinking of persisting / updating `C9` to unlock the additional fan modes that 4-Way-Cassette supports for example.
Also I'll eventually expose the vane configuration to HA as I use different settings during summer & winter for better air distribution.

Another question - if you're using your AC for heating as well - I'm wondering why you set the min. setpoint to `18°C`.

My Artcool Gallery does not support a cooling setpoint of 16°C, while the 4-Way-Cassette does.
However, both support a heating setpoint of 16°C, so either that's available for all units or there must be another undecoded parameter.

JanM321, 2023-11-04:
> Another question - if you're using your AC for heating as well - I'm wondering why you set the min. setpoint to 18°C.

I never use setpoint < 18, so I just didn't think about it. My apartment stays pretty warm so I turn off the AC if I'm gone for a few days.

Do you know if your Artcool handles a cooling setpoint of 16 by simply changing the setting to 18? In that case we could consider setting the minimum to 16 for now and letting the AC handle it.

For my uses the software is mostly feature complete at this point. I can look into vane control + maybe a YAML parameter for the over heating installer setting when I have some time. I think having these in place would make it easier for people to add their own settings.

florianbrede-ayet, 2023-11-05:
> Do you know if your Artcool handles a cooling setpoint of 16 by simply changing the setting to 18? In that case we could consider setting the minimum to 16 for now and letting the AC handle it.
 
Yes it does, I set it to 19°C in cooling via thinkQ and then requested 16°C with the controller. 
It will clamp to 18°C:
```
[12:25:17][D][lg-controller:524]: update
[12:25:17][D][lg-controller:387]: received A8.02.00.00.00.00.13.14.00.00.00.00.84 (13)
[12:25:17][D][lg-controller:396]: verified send
[12:25:20][D][climate:011]: 'LG Bild Wohnzimmer' - Setting
[12:25:20][D][climate:040]:   Target Temperature: 16.00
[12:25:20][D][climate:380]: 'LG Bild Wohnzimmer' - Sending state:
[12:25:20][D][climate:383]:   Mode: COOL
[12:25:20][D][climate:388]:   Fan Mode: LOW
[12:25:20][D][climate:400]:   Swing Mode: OFF
[12:25:20][D][climate:403]:   Current Temperature: 20.00°C
[12:25:20][D][climate:409]:   Target Temperature: 16.00°C
[12:25:23][D][lg-controller:524]: update
[12:25:24][D][lg-controller:371]: sending A8.03.00.00.00.00.11.14.00.00.00.00.85 (13)
[12:25:24][W][component:214]: Component <unknown> took a long time for an operation (0.51 s).
[12:25:24][W][component:215]: Components should block for at most 20-30ms.
[12:25:29][D][esp32.preferences:114]: Saving 1 preferences to flash...
[12:25:29][D][esp32.preferences:143]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
[12:25:29][D][lg-controller:524]: update
[12:25:29][D][lg-controller:387]: received A8.03.00.00.00.00.11.14.00.00.00.00.85 (13)
[12:25:29][D][lg-controller:396]: verified send
[12:25:35][D][lg-controller:524]: update
[12:25:35][D][lg-controller:387]: received C8.03.00.00.00.04.13.14.00.00.00.00.A3 (13)
[12:25:35][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[12:25:35][D][binary_sensor:036]: 'Outdoor Unit': Sending state ON
[12:25:35][D][climate:380]: 'LG Bild Wohnzimmer' - Sending state:
[12:25:35][D][climate:383]:   Mode: COOL
[12:25:35][D][climate:388]:   Fan Mode: LOW
[12:25:35][D][climate:400]:   Swing Mode: OFF
[12:25:35][D][climate:403]:   Current Temperature: 20.00°C
[12:25:35][D][climate:409]:   Target Temperature: 18.00°C
[12:25:35][D][lg-controller:387]: received C8.03.00.00.00.04.13.14.00.00.00.00.A3 (13)
[12:25:35][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[12:25:35][D][climate:380]: 'LG Bild Wohnzimmer' - Sending state:
[12:25:35][D][climate:383]:   Mode: COOL
[12:25:35][D][climate:388]:   Fan Mode: LOW
[12:25:35][D][climate:400]:   Swing Mode: OFF
[12:25:35][D][climate:403]:   Current Temperature: 20.00°C
[12:25:35][D][climate:409]:   Target Temperature: 18.00°C
[12:25:35][D][lg-controller:387]: received C8.03.00.00.00.00.13.14.00.00.00.00.A7 (13)
[12:25:35][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[12:25:35][D][climate:380]: 'LG Bild Wohnzimmer' - Sending state:
[12:25:35][D][climate:383]:   Mode: COOL
[12:25:35][D][climate:388]:   Fan Mode: LOW
[12:25:35][D][climate:400]:   Swing Mode: OFF
[12:25:35][D][climate:403]:   Current Temperature: 20.00°C
[12:25:35][D][climate:409]:   Target Temperature: 18.00°C
```

> For my uses the software is mostly feature complete at this point. I can look into vane control + maybe a YAML parameter for the over heating installer setting when I have some time. I think having these in place would make it easier for people to add their own settings.

No worries, I'll just update my fork to my requirements, there is no need to upstream everything :+1:

JanM321, 2023-11-05:
I just committed some changes for vertical vane position (just a single vane since that's all I have here) and a YAML setting to change installer setting 15.

I had been thinking about doing this for a while, but it required some refactoring to handle other message types.

florianbrede-ayet, 2023-11-06:
I've tested the 0.5 overheat settings on my units and they seem to perform well, so I've reset my fork to your codebase and created a small PR for the setpoint.

Also I've taken a look at your latest commit and have a few questions:

- Have you ever tried finding a way to force the indoor unit to report 0xC9, 0xCA and 0xCB?
- Related to (1), your implementation requires power-cycling the AC whenever the ESP32 reboots, otherwise it's impossible to set the vane(s). 
My units never send capabilities or installer messages voluntarily without a power cycle, so wouldn't it make sense to persist the latest revision of these 3 messages?
- Have you found any reference in the data books regarding this for heating mode:
`XX00_0000: fan operation in cooling mode and thermal off conditions, installer setting 35 (0: fan low, 1: fan off, 2: fan setting)`
That would we very useful for me, because the Cassette always cycles its vanes between thermal off and stops the fan which would help to distribute heat from the ceiling.
- How does your MU2R17 behave in heating mode?
My MU3R19 starts defrosting (a few times per day) at around 10°C outdoor temperature:
[image]
Additionally, it mostly refuses to properly modulate above approx. 5°C outdoor temperature:
[image]

JanM321, 2023-11-06:
> Have you ever tried finding a way to force the indoor unit to report 0xC9, 0xCA and 0xCB?

Yes, I made a change on Saturday to set the request-settings flag when the ESP32 powers on. This matches the LG controller and makes my units send all of the settings. Does that work?

> Have you found any reference in the data books regarding this for heating mode:

Maybe installer setting 46, "Setting the fan continuous"? My unit doesn't support this option with the official controller so I wouldn't try this myself. Probably worth looking at the PREMTB001 and PREMTC00U manuals. Unfortunately a lot of these settings are poorly documented and I had to get information from various manuals.

>  My MU3R19 starts defrosting (a few times per day) at around 10°C outdoor temperature:

This is an issue with LG multi split systems for heating unfortunately. They call this an "oil return cycle" and it happens in heating mode every ~3 hours. With the -0.5/+0.5 installer setting I haven't seen it yet because the unit makes much shorter runs.

Modulation isn't great for me either. My MU2R17 is less powerful than your MU3R19 which helps a bit I think. This is for last night with just one unit turned on:

[image]

This is with fan speed 4 of 5 (it uses less power at higher fan speeds because less heat is needed if the fan helps spread it).

Typical stable power consumption after running for at least 10 minutes is < 500W, but sometimes it's 800W or so and I don't really know why. I think I do see fewer peaks with the over heating setting set to 4, but I'm not sure.

JanM321, 2023-11-06:
I just realized there's still a minor issue with the initialization code: I should set `is_initializing_` to `false` when we get a `C9`/`CA`/`CB` message instead of `C8`, to avoid a 'false positive' when the AC manages to sneak in a `C8` message just when the ESP32 is restarting. I'll test and change this tonight.

> For me it's really hit-and-miss. It's kinda sad, I heat a large house with 3 Multisplits and the others (Toshiba RAS-3M18U, 2M14U) do perform much better and module to below 200W easily.
> 
> The Toshiba 3M18U (pink) has equal capacity to the MU3R19 (blue) and is handling around 60% of the thermal load here:

Yeah, it seems that for LG the heating feature is more of an afterthought and their main focus is on cooling. MHI, ME, Daikin, Panasonic are also better at this. I've heard the LG single split units work better than the multis as well.

In March I got really annoyed with the internal temperature sensor (wall units are located near the ceiling so the sensor was often *way* off and rooms were either too cold or too warm). After that I started reverse engineering an LG wall controller and I'm really glad this all worked out in time for this winter. The behavior still isn't perfect, but the external temperature sensor has made a huge difference.

drbugfinder, 2023-11-12:
Hello everyone,

I'm currently pressed for time and unable to delve into my unit further, but I wanted to share the C9/CA/CB messages from my three S12ET units with you. Perhaps they might be of interest:

```
C9.C4.E8.1F.81.71.00.C0.02.40.04.81.58
CA.00.00.00.00.00.00.00.00.02.F1.21.8B
CB.00.00.FF.FF.65.00.00.00.00.00.00.7B

CB.00.00.FF.FF.7C.00.00.00.00.00.00.10

CB.00.00.FF.FF.7F.00.00.00.00.00.00.1D
```

If it's too off-topic for this issue, just ignore it.

JanM321, 2023-11-12:
> I'm currently pressed for time and unable to delve into my unit further, but I wanted to share the C9/CA/CB messages from my three S12ET units with you. Perhaps they might be of interest:

Thanks, that's helpful. These are the messages for my AP09RT unit (before I changed installer setting 15, "over heating"):
```
C9.C4.EA.1F.81.71.00.80.02.40.04.81.9A
CA.00.00.00.00.00.00.00.00.12.F1.21.BB
CB.00.00.78.78.FF.00.00.00.00.00.00.EF
```
The C9 message is almost identical to your units (as expected because they're similar wall units), except:
* Byte 2 is 0xEA instead of 0xE8 because it has the air purifier setting.
* Byte 7 is 0x80 instead of 0xC0, so mine is missing the 0x40 bit that your units have set. Wild guess: maybe something that single split units have because Florian's multi-split units also don't have this.
* (Byte 12 is the checksum.)

The CA message is similar except for byte 9 where mine has an extra unknown bit set. The Artcool has this set too but not the cassette.

The CB messages are very interesting. Until now I've seen bytes 3 and 4 have similar values that change regularly and byte 5 is either 0x00 or 0xFF. With your units, that's kind of reversed: bytes 3 and 4 are 0xFF but byte 5 has the kind of value that other units have in bytes 3-4.

I really wonder what these 3 bytes are for. I've been thinking about adding some sensors to report these values to Home Assistant every N minutes, to see if tracking them over time tells us something...

drbugfinder, 2023-11-12:
Thanks for the explanation of the various bytes. Regarding byte 2: You mean your unit has air purifier built in, mine does not, correct?

> I really wonder what these 3 bytes are for. I've been thinking about adding some sensors to report these values to Home Assistant every N minutes, to see if tracking them over time tells us something...

If you could provide me with a modified version in another branch or just a diff, I could test it out for you.

JanM321, 2023-11-12:
> Regarding byte 2: You mean your unit has air purifier built in, mine does not, correct?

Correct.

> If you could provide me with a modified version in another branch or just a diff, I could test it out for you.

Thanks. I haven't made these changes yet but if I do I'll let you know 👍 

---

Below is a list of known CB message bytes 3-5. Looking at these values and when I collected some of them, it could be a temperature value. `temperature = value / 10 + 10` (or `+ 12` or something) would fit pretty well. This is just a guess though.

```
My units, most are probably from June:
78.78.FF
71.72.FF
9B.93.FF
9C.94.FF
9E.96.FF
A0.98.FF
74.76.FF
8A.8B.FF
86.87.FF
85.87.FF
66.63.FF

Artcool:
64.68.FF

Cassette:
77.73.00

Comment on instructables blog post:
58.2E.00
60.34.00
ED.E7.00
64.65.00

S12ET:
FF.FF.65
FF.FF.7C
FF.FF.7F
```

florianbrede-ayet, 2023-11-12:
The air purifier flag does not indicate whether an plasma ionizer is built in though.
My cassette reports it, PREMTB001 allows enabling it - but nothing happens as the component is an add-on that has to be purchased & installed separately:
[image]

Here are some CB messages from my Cassette:
```
[21:54:36][D][lg-controller:387]: received CB.80.00.77.73.00.00.00.01.01.40.00.22 (13)
[21:54:36][D][lg-controller:387]: received CB.00.00.77.73.00.00.00.01.01.40.00.A2 (13)
[01:43:37][D][lg-controller:423]: received CB.00.20.5A.5C.00.00.00.01.01.40.00.B6 (13)
[01:43:43][D][lg-controller:423]: received CB.00.20.5A.5C.00.00.00.01.01.40.00.B6 (13)
[15:10:37][D][lg-controller:440]: received CB.80.20.2F.2D.00.00.00.01.01.40.00.5C (13)
[15:10:37][D][lg-controller:440]: received CB.00.20.2F.2E.00.00.00.01.01.40.00.DF (13)
[07:56:24][D][lg-controller:440]: received CB.80.20.78.79.00.00.00.01.01.40.00.CB (13)
[07:56:24][D][lg-controller:440]: received CB.00.20.78.79.00.00.00.01.01.40.00.4B (13)
[07:56:24][D][lg-controller:440]: received CB.80.20.78.79.00.00.00.01.01.40.00.CB (13)
[07:56:24][D][lg-controller:440]: received CB.00.20.78.79.00.00.00.01.01.40.00.4B (13)
[07:56:24][D][lg-controller:440]: received CB.80.20.78.79.00.00.00.01.01.40.00.CB (13)
[07:56:30][D][lg-controller:440]: received CB.00.20.78.79.00.00.00.01.01.40.00.4B (13)
[07:56:30][D][lg-controller:440]: received CB.80.20.78.79.00.00.00.01.01.40.00.CB (13)
[07:56:30][D][lg-controller:440]: received CB.00.20.78.79.00.00.00.01.01.40.00.4B (13)
[07:56:36][D][lg-controller:440]: received CB.80.20.78.7A.00.00.00.01.01.40.00.CA (13)
[07:56:36][D][lg-controller:440]: received CB.80.20.78.7A.00.00.00.01.01.40.00.CA (13)
[07:56:36][D][lg-controller:440]: received CB.00.20.78.7A.00.00.00.01.01.40.00.4A (13)
[07:56:42][D][lg-controller:440]: received CB.00.20.78.7A.00.00.00.01.01.40.00.4A (13)
[07:56:42][D][lg-controller:440]: received CB.80.20.78.7A.00.00.00.01.01.40.00.CA (13)
[07:56:48][D][lg-controller:440]: received CB.00.20.78.7A.00.00.00.01.01.40.00.4A (13)
[07:56:54][D][lg-controller:440]: received CB.80.20.78.7A.00.00.00.01.01.40.00.CA (13)
[07:56:54][D][lg-controller:440]: received CB.00.20.78.79.00.00.00.01.01.40.00.4B (13)
[03:05:38][D][lg-controller:535]: received CB.80.20.E0.E0.00.00.00.01.01.40.00.38 (13)
[03:05:38][D][lg-controller:535]: received CB.00.20.E0.E0.00.00.00.01.01.40.00.B8 (13)
[13:38:53][D][lg-controller:535]: received CB.00.20.4D.4A.00.00.00.01.01.40.00.91 (13)
[13:38:59][D][lg-controller:535]: received CB.80.20.4D.4A.00.00.00.01.01.40.00.11 (13)
[13:38:59][D][lg-controller:535]: received CB.00.20.4D.4A.00.00.00.01.01.40.00.91 (13)
[13:38:59][D][lg-controller:535]: received CB.80.20.4D.4A.00.00.00.01.01.40.00.11 (13)
[13:39:05][D][lg-controller:535]: received CB.00.20.4D.4A.00.00.00.01.01.40.00.91 (13)
[19:40:58][D][lg-controller:535]: received CB.80.20.56.54.00.00.00.01.01.40.00.02 (13)
[19:40:58][D][lg-controller:535]: received CB.00.20.56.54.00.00.00.01.01.40.00.82 (13)
[19:42:21][D][lg-controller:535]: received CB.80.20.59.58.00.00.00.01.01.40.00.0B (13)
[19:42:21][D][lg-controller:535]: received CB.00.20.59.58.00.00.00.01.01.40.00.8B (13)
[02:54:00][D][lg-controller:667]: received CB.80.20.45.43.00.00.00.01.01.40.00.60 (13)
[02:54:06][D][lg-controller:667]: received CB.00.20.45.43.00.00.00.01.01.40.00.E0 (13)
[13:31:58][D][lg-controller:672]: received CB.80.20.42.38.00.00.00.01.01.40.00.72 (13)
[13:31:58][D][lg-controller:672]: received CB.00.20.41.39.00.00.00.01.01.40.00.F2 (13)
[00:21:24][D][lg-controller:672]: received CB.80.20.59.58.00.00.00.01.01.40.00.0B (13)
[00:21:24][D][lg-controller:672]: received CB.00.20.59.58.00.00.00.01.01.40.00.8B (13)
```

And from my Artcool:
```
[02:22:42][D][lg-controller:532]: received CB.00.20.58.55.FF.00.00.40.04.00.00.8E (13)
[02:48:39][D][lg-controller:535]: received CB.80.20.4F.4E.FF.00.00.40.04.00.00.1E (13)
[02:48:39][D][lg-controller:535]: received CB.80.20.4F.4E.FF.00.00.40.04.00.00.1E (13)
[02:49:59][D][lg-controller:535]: received CB.80.20.51.4E.FF.00.00.40.04.00.00.18 (13)
[02:49:59][D][lg-controller:535]: received CB.80.20.51.4F.FF.00.00.40.04.00.00.1B (13)
[02:51:55][D][lg-controller:535]: received CB.80.20.53.51.FF.00.00.40.04.00.00.07 (13)
[02:51:55][D][lg-controller:535]: received CB.80.20.53.51.FF.00.00.40.04.00.00.07 (13)
[02:52:13][D][lg-controller:535]: received CB.00.20.53.51.FF.00.00.40.04.00.00.87 (13)
[02:52:13][D][lg-controller:535]: received CB.00.20.53.51.FF.00.00.40.04.00.00.87 (13)
[02:56:02][D][lg-controller:535]: received CB.80.20.56.55.FF.00.00.40.04.00.00.0C (13)
[02:56:02][D][lg-controller:535]: received CB.80.20.56.55.FF.00.00.40.04.00.00.0C (13)
[02:57:16][D][lg-controller:535]: received CB.80.20.58.56.FF.00.00.40.04.00.00.09 (13)
[02:57:16][D][lg-controller:535]: received CB.80.20.58.56.FF.00.00.40.04.00.00.09 (13)
[02:57:52][D][lg-controller:535]: received CB.00.20.59.56.FF.00.00.40.04.00.00.88 (13)
[02:57:52][D][lg-controller:535]: received CB.00.20.59.56.FF.00.00.40.04.00.00.88 (13)
[03:00:12][D][lg-controller:535]: received CB.80.20.5C.5B.FF.00.00.40.04.00.00.30 (13)
[03:00:12][D][lg-controller:535]: received CB.80.20.5C.5B.FF.00.00.40.04.00.00.30 (13)
[03:01:53][D][lg-controller:535]: received CB.80.20.50.4F.FF.00.00.40.04.00.00.18 (13)
[03:01:53][D][lg-controller:535]: received CB.80.20.50.4F.FF.00.00.40.04.00.00.18 (13)
[03:14:53][D][lg-controller:543]: received CB.80.20.5F.52.FF.00.00.40.04.00.00.0A (13)
[... cut ...]
[01:37:10][D][lg-controller:671]: received CB.00.20.51.52.FF.00.00.40.04.00.00.84 (13)
[01:37:10][D][lg-controller:671]: received CB.00.20.51.52.FF.00.00.40.04.00.00.84 (13)
[18:20:58][D][lg-controller:671]: received CB.80.20.51.53.FF.00.00.40.04.00.00.07 (13)
[18:20:58][D][lg-controller:671]: received CB.80.20.51.53.FF.00.00.40.04.00.00.07 (13)
[18:21:16][D][lg-controller:671]: received CB.00.20.51.54.FF.00.00.40.04.00.00.86 (13)
[18:21:22][D][lg-controller:671]: received CB.00.20.51.54.FF.00.00.40.04.00.00.86 (13)
```

JanM321, 2023-11-13:
It does look like two similar temperature/humidity/something sensor values.
```
[03:05:38][D][lg-controller:535]: received CB.80.20.E0.E0.00.00.00.01.01.40.00.38 (13)
[03:05:38][D][lg-controller:535]: received CB.00.20.E0.E0.00.00.00.01.01.40.00.B8 (13)
```
E0 is pretty high. Maybe the heating was on here?

I think it's unlikely that we can use these bytes for anything interesting. It would be nice to have a newer controller like the PREMTB100 to see if that tells us more, but they're pretty expensive so I've found it hard to justify getting one - we can already implement pretty much all of the available features.

JanM321, 2023-12-17:
Progress! These 0xCB bytes are pipe temperature values. I documented them in protocol.md

I figured this out by playing more with the PREMTB100. I'm not sure yet which flag enables the menu option for this, because with my unit's settings it doesn't show this.

This is pretty interesting information to monitor. I might add sensors for these values, but I don't know yet if I can get my unit to send 0xCB regularly.

JanM321, 2023-12-17:
To request a `CB` message, the controller can send an `AB 80` message (the high bit is set, but otherwise the same as a regular `AB 00` message). Fortunately my units support this.

This also works in the other direction: the PREMTB100 will immediately send `AB 00` messages when the controller sends `CB 80` (but not `CB 00`). I used this to build the table mapping byte values to temperature.

I have this implemented now to request a `CB` message every 5 minutes and to report the temperatures in `CB` to HA. My units support the In and Out sensors, not Mid.

I'll monitor this for a bit to see if it works as expected.

JanM321, 2023-12-18:
OK I made these changes. The controller will request and then report the pipe temperature values every 10 minutes. In heating mode these values can get pretty high (30-50 degrees) but when the units are off I get values close to the room temperature.

@drbugfinder based on your CB messages, your units only report the 'Mid' value. Most other units report 'In' and 'Out' but not 'Mid'. If you have a chance to verify this sensor works too that'd be great.

florianbrede-ayet, 2023-12-18:
Great finding!
I'll update my controllers to monitor the temperatures, might also help me debugging the issues I still have.

florianbrede-ayet, 2023-12-19:
Okay this is a tiny bit confusing.

**Artcool Gallery**
[image]

**My (problematic) 4-Way-Cassette**
[image]

Are the sensors reversed (like swapped connectors) or did I somehow (unlikely) manage to reverse the pipes?
I've never disconnected the sensors from the PCB, but on the other hand, my ODU also had some "manufacturing defects" I had to fix initially...

And yes, In is always 1+°C below Out for all temperature ranges I saw.

JanM321, 2023-12-20:
> And yes, In is always 1+°C below Out for all temperature ranges I saw.

It's the same with my units. I read it as "air coming out is hotter than air going in" but not sure if that makes sense. Typical stable values are In: 37 / Out: 42 or In: 30 / Out: 33. Both drop to around-zero when there's an oil return cycle.

drbugfinder, 2023-12-20:
> OK I made these changes. The controller will request and then report the pipe temperature values every 10 minutes. In heating mode these values can get pretty high (30-50 degrees) but when the units [...]

I see the following output:
```
[08:49:30][D][lg-controller:1016]: update
[08:49:30][D][lg-controller:701]: received 00.00.00.00.00.00.00.00.00.00.00.00.00 (13)
[08:49:30][D][lg-controller:708]: Ignoring padding message sent by unit
[08:49:30][D][lg-controller:701]: received CB.00.00.FF.FF.47.00.00.00.00.00.00.45 (13)
[08:49:30][D][sensor:094]: 'Pipe Temperature Mid': Sending state 43.00000 °C with 0 decimals of accuracy
[08:49:30][D][lg-controller:701]: received 00.00.00.00.00.00.00.00.00.00.00.00.00 (13)
[08:49:30][D][lg-controller:708]: Ignoring padding message sent by unit
[08:49:30][D][lg-controller:701]: received 00.00.00.00.00.00.00.00.00.00.00.00.55 (13)
[08:49:30][D][lg-controller:701]: received 00.00.00.00.00.00.00.00.00.00.00.00.00 (13)
[08:49:30][D][lg-controller:708]: Ignoring padding message sent by unit
[08:49:30][D][lg-controller:701]: received CD.00.00.00.00.00.00.00.00.00.00.00.98 (13)
[08:49:30][D][lg-controller:701]: received 00.00.00.00.00.00.00.00.00.00.00.00.00 (13)
[08:49:30][D][lg-controller:708]: Ignoring padding message sent by unit
[08:49:31][D][lg-controller:657]: sending AA.00.00.00.00.00.00.00.00.02.F1.21.EB (13)
```
So I think this works 👍🏻