JanM321/esphome-lg-controller #9: Tiny PCB With Kicad Files

florianbrede-ayet, 2023-11-03:
Hi,
first thanks for the project and your effort, I've finally been able to reduce my "LG Artcool Gallery (Multi)" to bearable fan speeds, which helps a lot.
I still have different issues with my MU3R19 setup, because in heating it doesn't modulate correctly. 
But unfortunately, no error is ever reported, so I have to continue looking for an LGMV dongle.

Anyway, the Artcool Gallery is tight on space and the parts of your PCB are quite hard to source and rather expensive, I designed a tiny pcb with an ESP32 DevKit footprint (around $8 per piece assembled at jlcpcb):
[image]

The files are currently in my fork, but if you're interested I could create a PR:
https://github.com/florianbrede-ayet/esphome-lg-controller/tree/main/hardware-tiny

JanM321, 2023-11-04:
This is really nice.

This was my first PCB so I tried to follow the project I based the hardware on as much as possible to not break something, but it's good to have an alternative design that works. I thought the TI LIN transceiver was the only one that didn't have the timeout feature but you found another one :)

If you could do a PR for the hardware-tiny directory, that'd work. I could also link to your fork in the README file - let me know which option you prefer.

I never expected this project to get this much attention :) I'm now thinking about rewriting the software a little to make it easier to send/receive other message types.

mshrem, 2023-11-07:
> Hi, first thanks for the project and your effort, I've finally been able to reduce my "LG Artcool Gallery (Multi)" to bearable fan speeds, which helps a lot. I still have different issues with my [...]

Ohhhhh wow this is awesome i wish i had seen this before i placed the order with pcbway. I also have some Artcool Gallery so this will help me. 
I think they soldered some [parts incorrectly ](https://github.com/JanM321/esphome-lg-controller/issues/8#issuecomment-1798005828) but i am waiting on verification otherwise if they did i will kindly ask them to change them for this layout.

JanM321, 2023-11-07:
@florianbrede-ayet did you want to create a pull request for this? I think this could help a lot of people.

florianbrede-ayet, 2023-11-07:
@JanM321 just did, it got lost with my fork reset.

JanM321, 2023-11-08:
Thanks! I merged the PR and added a comment to the README about this.

closed 2023-11-08 by JanM321