JanM321/esphome-lg-controller #15: Restoring fan speed and swing mode after power-cycle fails

florianbrede-ayet, 2023-11-13:
I noticed that restoring the previous fan state and swing mode fails after a power-cycle of the heatpumpt (and thus the ESP32).

This is the log for reference (original swing mode was OFF and fan mode was LOW), I can take a look as soon as I have time:
```
[03:13:13][D][lg-controller:934]: update
[03:13:17][D][lg-controller:934]: update
[03:13:23][D][lg-controller:934]: update
[03:13:29][D][lg-controller:934]: update
[03:13:29][E][lg-controller:975]: discarding incomplete data 00
[03:13:35][D][lg-controller:934]: update
[03:13:35][E][lg-controller:521]: unknown fan mode, using Medium
[03:13:35][E][lg-controller:545]: unknown swing mode
[03:13:35][D][lg-controller:600]: sending A8.29.00.00.00.01.13.11.40.00.00.00.63 (13)
[03:13:35][D][climate:380]: 'LG Bild Wohnzimmer' - Sending state:
[03:13:35][D][climate:383]:   Mode: OFF
[03:13:35][D][climate:400]:   Swing Mode: UNKNOWN
[03:13:35][D][climate:403]:   Current Temperature: 18.50°C
[03:13:35][D][climate:409]:   Target Temperature: 18.50°C
[03:13:35][W][component:214]: Component <unknown> took a long time for an operation (0.55 s).
[03:13:35][W][component:215]: Components should block for at most 20-30ms.
[03:13:41][D][lg-controller:934]: update
[03:13:41][D][lg-controller:671]: received A8.29.00.00.00.01.13.11.40.00.00.00.63 (13)
[03:13:41][D][lg-controller:687]: verified send
[03:13:41][D][sensor:094]: 'Uptime': Sending state 37.53000 s with 0 decimals of accuracy
[03:13:47][D][lg-controller:934]: update
[03:13:53][D][lg-controller:934]: update
[03:13:56][D][esp32.preferences:114]: Saving 1 preferences to flash...
[03:13:56][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[03:13:59][D][lg-controller:934]: update
[03:13:59][E][lg-controller:521]: unknown fan mode, using Medium
[03:13:59][E][lg-controller:545]: unknown swing mode
[03:13:59][D][lg-controller:600]: sending A8.28.00.00.00.01.13.11.40.00.00.00.60 (13)
[03:13:59][W][component:214]: Component <unknown> took a long time for an operation (0.52 s).
[03:13:59][W][component:215]: Components should block for at most 20-30ms.
[03:14:04][D][climate:011]: 'LG Bild Wohnzimmer' - Setting
[03:14:04][D][climate:015]:   Mode: HEAT
[03:14:04][D][climate:380]: 'LG Bild Wohnzimmer' - Sending state:
[03:14:04][D][climate:383]:   Mode: HEAT
[03:14:04][D][climate:400]:   Swing Mode: UNKNOWN
[03:14:04][D][climate:403]:   Current Temperature: 18.50°C
[03:14:04][D][climate:409]:   Target Temperature: 18.50°C
[03:14:05][D][lg-controller:934]: update
[03:14:05][D][lg-controller:671]: received A8.28.00.00.00.01.13.11.40.00.00.00.60 (13)
[03:14:05][D][lg-controller:687]: verified send
[03:14:05][D][lg-controller:671]: received C8.10.00.00.00.01.13.00.00.00.40.00.79 (13)
```

JanM321, 2023-11-14:
I think I've seen this too, the first time I flashed the firmware version that saves the capabilities. Maybe we're setting the traits too late now?

drbugfinder, 2023-11-14:
I've seen this exact same issue also on my units yesterday, when we had a power-outage.

drbugfinder, 2023-12-19:
I noticed that this happens every time the unit is turned off via the controller (not only power-cycle/mains off). After turning it on again it half-opens in a position which also the LG app does not show correctly (for heating I always want it to point fully downwards)
[image]
[image]

JanM321, 2023-12-19:
> I noticed that this happens every time the unit is turned off via the controller (not only power-cycle/mains off). After turning it on again it half-opens in a position which also the LG app does not show correctly (for heating I always want it to point fully downwards)

The LG app shows it's in the default vane position, but in HA that should be 0 not 6. If I turn my unit off and on in HA it still remembers this setting.

Does it work if you use the LG app to turn the unit off? The ESP32 sends "mode = fan" when off, maybe that confuses the unit because changing the operation mode does reset the vertical vane position.

florianbrede-ayet, 2023-12-19:
Okay this is a tiny bit confusing.

**Artcool Gallery**
[image]

**My (problematic) 4-Way-Cassette**
[image]

Are the sensors reversed or did I somehow manage to reverse the pipes (which is unlikely since I wouldn't have had the fitting nuts)?
And yes, In is always 1+°C below Out for all temperature ranges I saw.

drbugfinder, 2023-12-20:
When I unplug the controller, it works normally through the LG app. The controller (at least for my setup) sometimes seems to have higher priority than settings via the remote or the LG app for settings which are set by both, but I guess that this is a timing issue. So for example setting the temperature on the remote is sometimes overwritten by the esp-controller setting. Or when turning the unit off via the remote, it will power-on again after a minute by itself. I'm not using the remote anymore, but the app has some features, which are (not yet) available via the esp-controller like:
- current power-consumption (only total power)
- my unit supports upper three and lower three vane position swing mode, so it swings only the upper or the lower half (this is also not selectable via the remote control, but through the app
The remote control also has some nice features, which are not available through the app (or the esp-controller yet):
- Jet mode (higher than maximum speed and full heating/cooling with more power than in normal modes)
- Comfort Air (I think this is just a kind of special preset for vane positions in heating/cooling mode)
- Power saving for the inside unit (reduces cooling/heating power)
- Quiet mode for the outside unit (reduces outside unit fan rotation speed)

(Maybe I should put this in a new issue...)

> Does it work if you use the LG app to turn the unit off? The ESP32 sends "mode = fan" when off, maybe that confuses the unit because changing the operation mode does reset the vertical vane position.

I haven't tested this, yet. I will try whis.

JanM321, 2023-12-20:
I think your unit may communicate a bit differently and for some reason settings get overwritten. We'd need the log output, ideally a few minutes before and after you use the remote, to see what it's doing.

For power saving: do you mean the 40/60/80% thing? Isn't that only available in cooling mode? It would be nice to have the log output for when you change this setting because I don't know if/how this gets sent to the wired controller, but I am interested in these settings.

JanM321, 2023-12-20:
Feel free to file new issues for these things. Some of them may not be exposed to the wired controller but others I can probably implement pretty quickly.

drbugfinder, 2023-12-21:
This is the log output when turning the unit off (12:53:43) and back on again (12:55:58). 
After that the fan state (in this case even both states, vertically and horizontally) is not shown in the app and is in some half-open position.

```
[12:53:36][D][lg-controller:1016]: update

[12:53:38][D][esp32.preferences:114]: Saving 1 preferences to flash...
[12:53:38][D][esp32.preferences:143]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
[12:53:38][D][sensor:094]: 'Uptime': Sending state 101080.84375 s with 0 decimals of accuracy
[12:53:42][D][lg-controller:1016]: update
[12:53:43][D][lg-controller:617]: sending A8.12.00.08.40.00.07.94.00.00.00.00.C8 (13)
[12:53:43][W][component:204]: Component <unknown> took a long time for an operation (0.51 s).
[12:53:43][W][component:205]: Components should block for at most 20-30ms.
[12:53:45][D][climate:011]: 'LG Office' - Setting
[12:53:45][D][climate:015]:   Mode: OFF
[12:53:45][D][climate:378]: 'LG Office' - Sending state:
[12:53:45][D][climate:381]:   Mode: OFF
[12:53:45][D][climate:386]:   Fan Mode: LOW
[12:53:45][D][climate:398]:   Swing Mode: OFF
[12:53:45][D][climate:401]:   Current Temperature: 23.50°C
[12:53:45][D][climate:407]:   Target Temperature: 22.00°C
[12:53:48][D][lg-controller:1016]: update
[12:53:48][D][lg-controller:701]: received A8.12.00.08.40.00.07.94.00.00.00.00.C8 (13)
[12:53:48][D][lg-controller:717]: verified send
[12:53:49][D][lg-controller:617]: sending A8.09.00.08.40.00.07.94.00.00.00.00.C1 (13)
[12:53:49][W][component:204]: Component <unknown> took a long time for an operation (0.51 s).
[12:53:49][W][component:205]: Components should block for at most 20-30ms.
[12:53:54][D][lg-controller:1016]: update
[12:53:54][D][lg-controller:701]: received A8.09.00.08.40.00.07.94.00.00.00.00.C1 (13)
[12:53:54][D][lg-controller:717]: verified send
[12:54:00][D][lg-controller:1016]: update
[12:54:06][D][lg-controller:1016]: update
[12:54:12][D][lg-controller:1016]: update
[12:54:13][D][lg-controller:617]: sending A8.08.00.08.40.00.07.94.00.00.00.00.C6 (13)
[12:54:13][W][component:204]: Component <unknown> took a long time for an operation (0.51 s).
[12:54:13][W][component:205]: Components should block for at most 20-30ms.
[12:54:18][D][lg-controller:1016]: update
[12:54:18][D][lg-controller:701]: received A8.08.00.08.40.00.07.94.00.00.00.00.C6 (13)
[12:54:18][D][lg-controller:717]: verified send
[12:54:24][D][lg-controller:1016]: update
[12:54:30][D][lg-controller:1016]: update
[12:54:36][D][lg-controller:1016]: update
[12:54:36][D][lg-controller:701]: received C8.10.00.00.40.00.07.1B.00.00.00.00.6F (13)
[12:54:36][D][binary_sensor:036]: 'Preheat': Sending state OFF
[12:54:36][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[12:54:36][D][climate:378]: 'LG Office' - Sending state:
[12:54:36][D][climate:381]:   Mode: OFF
[12:54:36][D][climate:386]:   Fan Mode: LOW
[12:54:36][D][climate:398]:   Swing Mode: OFF
[12:54:36][D][climate:401]:   Current Temperature: 23.50°C
[12:54:36][D][climate:407]:   Target Temperature: 22.00°C
[12:54:36][D][lg-controller:701]: received C8.10.00.00.40.00.07.1B.00.00.00.00.6F (13)
[12:54:36][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[12:54:36][D][climate:378]: 'LG Office' - Sending state:
[12:54:36][D][climate:381]:   Mode: OFF
[12:54:36][D][climate:386]:   Fan Mode: LOW
[12:54:36][D][climate:398]:   Swing Mode: OFF
[12:54:36][D][climate:401]:   Current Temperature: 23.50°C
[12:54:36][D][climate:407]:   Target Temperature: 22.00°C
[12:54:37][D][lg-controller:617]: sending A8.08.00.00.40.00.07.14.00.00.00.00.5E (13)
[12:54:37][W][component:204]: Component <unknown> took a long time for an operation (0.62 s).
[12:54:37][W][component:205]: Components should block for at most 20-30ms.
[12:54:38][D][esp32.preferences:114]: Saving 1 preferences to flash...
[12:54:38][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[12:54:38][D][sensor:094]: 'Uptime': Sending state 101140.83594 s with 0 decimals of accuracy
[12:54:42][D][lg-controller:1016]: update
[12:54:42][D][lg-controller:701]: received A8.08.00.00.40.00.07.14.00.00.00.00.5E (13)
[12:54:42][D][lg-controller:717]: verified send
[12:54:48][D][lg-controller:1016]: update
[12:54:54][D][lg-controller:1016]: update
[... 58 lines cut ...]
[12:55:58][D][climate:015]:   Mode: HEAT
[12:55:58][D][climate:378]: 'LG Office' - Sending state:
[12:55:58][D][climate:381]:   Mode: HEAT
[12:55:58][D][climate:386]:   Fan Mode: LOW
[12:55:58][D][climate:398]:   Swing Mode: OFF
[12:55:58][D][climate:401]:   Current Temperature: 23.50°C
[12:55:58][D][climate:407]:   Target Temperature: 22.00°C
[12:56:00][D][lg-controller:1016]: update
[12:56:01][D][lg-controller:617]: sending A8.13.00.00.40.00.07.14.00.00.00.00.43 (13)
[12:56:01][W][component:204]: Component <unknown> took a long time for an operation (0.51 s).
[12:56:01][W][component:205]: Components should block for at most 20-30ms.
[12:56:06][D][lg-controller:1016]: update
[12:56:06][D][lg-controller:701]: received A8.13.00.00.40.00.07.14.00.00.00.00.43 (13)
[12:56:06][D][lg-controller:717]: verified send
[12:56:12][D][lg-controller:1016]: update
[12:56:18][D][lg-controller:1016]: update
[12:56:24][D][lg-controller:1016]: update
[12:56:25][D][lg-controller:617]: sending A8.12.00.00.40.00.07.14.00.00.00.00.40 (13)
[12:56:25][W][component:204]: Component <unknown> took a long time for an operation (0.51 s).
[12:56:25][W][component:205]: Components should block for at most 20-30ms.

```

> For power saving: do you mean the 40/60/80% thing? Isn't that only available in cooling mode? 
You're right, only in cooling mode.

drbugfinder, 2023-12-21:
And this is after setting the position to down (from the previous, unknown state) via the app:

```
[13:28:06][D][lg-controller:1016]: update
[13:28:06][D][lg-controller:701]: received A8.12.00.00.40.04.07.94.00.00.00.00.CC (13)
[13:28:06][D][lg-controller:717]: verified send
[13:28:12][D][lg-controller:1016]: update
[13:28:12][D][lg-controller:701]: received C8.13.00.00.40.04.07.9C.00.00.00.00.97 (13)
[13:28:12][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[13:28:12][D][climate:378]: 'LG Office' - Sending state:
[13:28:12][D][climate:381]:   Mode: HEAT
[13:28:12][D][climate:386]:   Fan Mode: LOW
[13:28:12][D][climate:398]:   Swing Mode: OFF
[13:28:12][D][climate:401]:   Current Temperature: 24.00°C
[13:28:12][D][climate:407]:   Target Temperature: 22.00°C
[13:28:12][D][climate:378]: 'LG Office' - Sending state:
[13:28:12][D][climate:381]:   Mode: HEAT
[13:28:13][D][climate:386]:   Fan Mode: LOW
[13:28:13][D][climate:398]:   Swing Mode: OFF
[13:28:13][D][climate:401]:   Current Temperature: 24.00°C
[13:28:13][D][climate:407]:   Target Temperature: 22.00°C
[13:28:13][D][lg-controller:701]: received CA.00.00.00.00.00.00.06.00.02.F1.21.B1 (13)
[13:28:13][D][select:015]: 'Airflow 1 Up/Down': Sending state 6 (Down) (index 6)
[13:28:13][D][select:015]: 'Airflow 2 Up/Down': Sending state 0 (Default) (index 0)
[13:28:13][D][select:015]: 'Airflow 3 Up/Down': Sending state 0 (Default) (index 0)
[13:28:13][D][select:015]: 'Airflow 4 Up/Down': Sending state 0 (Default) (index 0)
[13:28:13][D][number:012]: 'Fan Speed Slow': Sending state 0.000000
[13:28:13][D][number:012]: 'Fan Speed Low': Sending state 0.000000
[13:28:13][D][number:012]: 'Fan Speed Medium': Sending state 0.000000
[13:28:13][D][number:012]: 'Fan Speed High': Sending state 0.000000
[13:28:13][D][lg-controller:701]: received C8.13.00.00.40.04.07.9C.00.00.00.00.97 (13)
[13:28:13][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[13:28:13][D][climate:378]: 'LG Office' - Sending state:
[13:28:13][D][climate:381]:   Mode: HEAT
[13:28:13][D][climate:386]:   Fan Mode: LOW
[13:28:13][D][climate:398]:   Swing Mode: OFF
[13:28:13][D][climate:401]:   Current Temperature: 24.00°C
[13:28:13][D][climate:407]:   Target Temperature: 22.00°C
[13:28:13][W][component:204]: Component <unknown> took a long time for an operation (0.20 s).
[13:28:13][W][component:205]: Components should block for at most 20-30ms.
[13:28:18][D][lg-controller:1016]: update
[13:28:24][D][lg-controller:1016]: update
[13:28:24][D][lg-controller:701]: received C8.13.00.00.40.04.07.06.00.00.00.97.C8 (13)
[13:28:24][E][lg-controller:711]: invalid checksum C8.13.00.00.40.04.07.06.00.00.00.97.C8 (13)
[13:28:24][D][lg-controller:701]: received 12.00.00.40.04.07.9C.00.00.00.00.94.C8 (13)
[13:28:24][E][lg-controller:711]: invalid checksum 12.00.00.40.04.07.9C.00.00.00.00.94.C8 (13)
[13:28:30][D][lg-controller:1016]: update
[13:28:36][D][lg-controller:1016]: update
[13:28:38][D][esp32.preferences:114]: Saving 1 preferences to flash...
[13:28:38][D][esp32.preferences:143]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
[13:28:38][D][sensor:094]: 'Uptime': Sending state 103180.84375 s with 0 decimals of accuracy
[13:28:42][D][lg-controller:1016]: update
[13:28:42][E][lg-controller:1057]: discarding incomplete data 12.00.00.40.04.07.9C.00.00.00.00.94 (12)
[13:28:48][D][lg-controller:1016]: update
[13:28:49][D][lg-controller:617]: sending A8.12.00.00.40.04.07.94.00.00.00.00.CC (13)
[13:28:49][W][component:204]: Component <unknown> took a long time for an operation (0.51 s).
[13:28:49][W][component:205]: Components should block for at most 20-30ms.
[13:28:54][D][lg-controller:1016]: update
[13:28:54][D][lg-controller:701]: received A8.12.00.00.40.04.07.94.00.00.00.00.CC (13)
[13:28:54][D][lg-controller:717]: verified send
[13:29:00][D][lg-controller:1016]: update
[13:29:06][D][lg-controller:1016]: update
[13:29:12][D][lg-controller:1016]: update
[13:29:13][D][lg-controller:617]: sending A8.12.00.00.40.04.07.94.00.00.00.00.CC (13)
[13:29:13][W][component:204]: Component <unknown> took a long time for an operation (0.51 s).
[13:29:13][W][component:205]: Components should block for at most 20-30ms.
[13:29:18][D][lg-controller:1016]: update
[13:29:18][D][lg-controller:701]: received A8.12.00.00.40.04.07.94.00.00.00.00.CC (13)
[13:29:18][D][lg-controller:717]: verified send
[13:29:24][D][lg-controller:1016]: update
[... 16 lines cut ...]
[13:29:25][W][component:204]: Component <unknown> took a long time for an operation (0.10 s).
[13:29:25][W][component:205]: Components should block for at most 20-30ms.
[13:29:30][D][lg-controller:1016]: update
[13:29:36][D][lg-controller:1016]: update
[13:29:37][D][lg-controller:617]: sending A8.12.00.00.40.04.07.94.00.00.00.00.CC (13)
[13:29:37][W][component:204]: Component <unknown> took a long time for an operation (0.51 s).
[13:29:37][W][component:205]: Components should block for at most 20-30ms.
[13:29:38][D][esp32.preferences:114]: Saving 1 preferences to flash...
[13:29:38][D][esp32.preferences:143]: Saving 1 preferences to flash: 1 cached, 0 written, 0 failed
[13:29:38][D][sensor:094]: 'Uptime': Sending state 103240.84375 s with 0 decimals of accuracy
[13:29:42][D][lg-controller:1016]: update
[13:29:42][D][lg-controller:701]: received A8.12.00.00.40.04.07.94.00.00.00.00.CC (13)
[13:29:42][D][lg-controller:717]: verified send
[13:29:48][D][lg-controller:1016]: update
[... cut ...]
[13:30:00][D][lg-controller:1016]: update
[13:30:01][D][lg-controller:617]: sending A8.12.00.00.40.04.07.94.00.00.00.00.CC (13)
[13:30:01][W][component:204]: Component <unknown> took a long time for an operation (0.51 s).
[13:30:01][W][component:205]: Components should block for at most 20-30ms.
[13:30:06][D][lg-controller:1016]: update
[13:30:06][D][lg-controller:701]: received A8.12.00.00.40.04.07.94.00.00.00.00.CC (13)
[13:30:06][D][lg-controller:717]: verified send
[13:30:12][D][lg-controller:1016]: update

```

JanM321, 2023-12-21:
@drbugfinder there are some checksum errors at the end while receiving a message from the AC. Do you see that regularly? It could explain some of your issues, because if the AC is notifying the controller of a change but these messages get lost, then the controller won't update its state.

Regarding the vane position: you could try changing [this line](https://github.com/JanM321/esphome-lg-controller/blob/main/esphome/lg-controller.h#L514) (line 514) from
```
b |= (2 << 2);
```
to
```
b |= (4 << 2);
```
With this change it will send `mode = heating` instead of `mode = fan` when the unit is off. This is just a wild guess though. I should probably change the code to preserve the previous mode value, like the LG controller does, but it's worth a try.

drbugfinder, 2024-01-22:
@JanM321 i tried this change, but all units still remain in the unknown vane position after power-on.

JanM321, 2024-01-22:
> @JanM321 i tried this change, but all units still remain in the unknown vane position after power-on.

The controller could send an `AA` message after mode changes, as a workaround, but it would only restore the vertical position. The horizontal direction the controller doesn't know anything about unfortunately.

> I see these checksum errors every few minutes (sometimes one every few hours, sometimes two within 10 minutes...) on all of my three units

It could be a hardware issue but it's weird you see it with all of your units. I've never seen them here.