JanM321/esphome-lg-controller #19: Using with IR remote cause problem

dzungpv, 2023-11-22:
I try you code on a wall unit, It is OK when using esphome with Home Assistant to control it.
But problem occurs when using IR remote, the esphome controller override all the command, for example when unit is OFF,
using IR turn it on but few seconds later it change to off.

JanM321, 2023-11-22:
Please post the esphome debug log for when this happens, including the exact time when you use the remote.

dzungpv, 2023-11-22:
> Please post the esphome debug log for when this happens, including the exact time when you use the remote.
My indoor model is: V13ENSN, outdoor unit: B10END1N

Every-time using the IR remote, it will revert back what state appear on HA esphome, after about 20 seconds. When it happen, logs send: 
```
[1;31m[E][lg-controller:584]: target temperature 20.000000 , thermistor: 26[0m
[0;36m[D][lg-controller:603]: sending A8.22.00.00.00.00.05.14.40.00.00.00.76 (13)[0m
```
Some time the unit show **CH03** error on the LCD, according to LG docs it is: "Fault Code C3 or CH03: Communication (Indoor Wired R/Control) Data Communication Fault"
But send control from esphome, it will be ok again.

This is picture when it error:

[image]
[image]

Do you have any unit using both IR remote and wall mount controller?

JanM321, 2023-11-22:
To see what's going on I need to see the full debug log. From at least a few minutes before you use the remote to a few minutes after.

I wonder if the AC isn't sending messages to the controller, or if they get lost somehow, but I can't say that for sure without seeing the log output.

dzungpv, 2023-11-22:
> To see what's going on I need to see the full debug log. From at least a few minutes before you use the remote to a few minutes after.
> 
> I wonder if the AC isn't sending messages to the controller, or if they get lost somehow, but I can't say that for sure without seeing the log output.

I will send you full logs later, because I installed it remotely for a friend, not mine.

JanM321, 2023-11-22:
From the message it's sending to the AC, it really looks like it's a problem with receiving messages from the AC. This is because it still has the bit set that means we didn't receive settings from the AC yet. So either the AC isn't sending anything back, or it's sending something we don't expect.

The full debug log would tell us more. If you're able to power cycle the AC and then get the debug log immediately when it's powering on that would be useful too.

dzungpv, 2023-11-27:
@JanM321, It is my fault, because I charge some of your code, now it is work perfect. I will close this.