JanM321/esphome-lg-controller #20: Investigate PREMTB100 protocol changes

JanM321, 2023-11-29:
I found a relatively cheap PREMTB100 controller (it's newer than the the PREMTB001 I've been using so far) and I've started reverse engineering it a little to see if it has major differences. A few obvious things I noticed:

* In addition to sending `A8` and `AC` messages, it now also sends `AE 80` messages regularly. Mostly has zeroes but the last bytes seem to represent some sort of timer like value that keeps increasing:
```
ae 80 00 00 00 00 00 00 02 00 13 03 13
```
* `AC` has zeroes except for byte 7 that has the high bit set if settings are changed (similar to PREMTB001) but byte 9 now always contains the room temperature (stored as `temperature * 2`). Like this:
```
ac 00 00 00 00 00 00 80 00 22 00 00 1b
0x22 / 2 => 17C
```
I don't know yet what this is for.

I made my ESP32 send this locally by adding the following at the end of `send_status_message`:
```cpp
uint8_t buf[MsgLen] = {};
buf[0] = 0xAC;
if (send_buf_[1] & 0x1) {
    buf[7] = 0x80;
}
buf[9] = uint8_t(temp * 2);
buf[12] = calc_checksum(buf);
ESP_LOGD(TAG, "sending %s", format_hex_pretty(buf, MsgLen).c_str());
serial_->write_array(buf, MsgLen);
```
I don't see an obvious change in behavior yet. It would be interesting to try different values to see how it affects the AC. It's possible my units just ignore it.

I don't have much time for this the coming days but I'll get back to this soon.

CC @florianbrede-ayet because you might find this interesting.

JanM321, 2023-11-29:
@florianbrede-ayet, in your logs in #16 I noticed the unit sends a higher temperature value in `CC` byte 9, like 20-25 degrees or so when the setpoint is lower than that. Could this be the return air temperature or something? If you could monitor this a bit that would be really helpful because the `AC` temperature value is probably similar.

florianbrede-ayet, 2023-11-30:
> @florianbrede-ayet, in your logs in #16 I noticed the unit sends a higher temperature value in `CC` byte 9, like 20-25 degrees or so when the setpoint is lower than that. Could this be the return air temperature or something? If you could monitor this a bit that would be really helpful because the `AC` temperature value is probably similar.
> 
> edit: or maybe it's just the unit's room temperature value?

I just checked my logs and it's the unit's room temperature thermistor, it matches the `C8` room temperature.

I've now included the room temperature in my `AC` messages and set the upper bit in my handshake `A8` byte 10 and see if that changes anything in my unit's behaviour.

JanM321, 2023-12-04:
I'm also trying to figure out more bits of byte 10 of the `C8`/`A8` message.

* Bit 0x80 is set by the controller when initializing. Same as PREMTB001. (The unit sets 0x40 once when responding to this?). We should probably do the same.
* Bit 0x04 is always set by PREMTB100 as far as I can tell. I think the PREMTB001 sets this too initially but it'll let the unit clear it? It'll then set it again when turning unit off => on.

So PREMTB100 will consistently send 0x84 when initializing and 0x04 after.

Florian, I'm curious if you see a pattern with bit 0x4 of byte 10 and PREMTB001.

JanM321, 2023-12-04:
> In addition to sending A8 and AC messages, it now also sends AE 80 messages regularly. Mostly has zeroes but the last bytes seem to represent some sort of timer like value that keeps increasing:

So this is not a timer, it's yet another field for the room temperature! It works like this:
```
ae 80 00 00 00 00 00 00 02 00 12 09 1e => 0x12 0x09 (= 18 9) => 18.9 C
ae 80 00 00 00 00 00 00 02 00 17 06 18 => 0x17 0x06 (= 23 6) => 23.6 C
```
They probably added this to let the controller send the room temperature with 0.1 degrees precision. Interestingly the PREMTB100 display still rounds to 0.5 degrees.

In my logs, byte 11 always has a value from 0 to 9 which makes sense.

JanM321, 2023-12-26:
I added some information from things I learned from PREMTB100 to protocol.md.

closed 2023-12-26 by JanM321