JanM321/esphome-lg-controller #21: LG AC and Heat Pump Support

bzumik1, 2023-12-04:
Hi, I plan to buy **LG AC UM36F + UUC1 Comfort** and **LG Heat Pump HN0913T.NK0 + HU051MR.U44**. Can I ask of they are supported?

JanM321, 2023-12-05:
> Hi, I plan to buy **LG AC UM36F + UUC1 Comfort** and **LG Heat Pump HN0913T.NK0 + HU051MR.U44**. Can I ask of they are supported?

I expect the AC units to be compatible, but I haven't seen these models before so I can't guarantee it will work. The air/water heat pump will likely need a different solution. I think it's possible to connect via modbus but I don't know more about this.

JanM321, 2023-12-05:
If you're able to test the controller with these units please let me know so I can add them to the README.

closed 2023-12-05 by JanM321

bzumik1, 2023-12-05:
Ok as I will receive them I will test that.

JanM321, 2023-12-05:
It *might* work. The protocol implemented here seems AC specific, but maybe the heat pump one is similar enough? There was a similar question in #14.