JanM321/esphome-lg-controller #23: hardware-tiny lot of extended lib type component

echoTheo, 2023-12-05:
This is not a bug but a possible improvement.

I try to order the tiny hardware PCB on JLCBCB but a lot of enxended library components are used.

[image]

An extra fee is apply on each extended lib type component. 

@florianbrede-ayet, could it be possible to use a maximum of basic lib type component to reduce the cost of the PCB ?

bleughb, 2023-12-05:
Most of them should be changeable to BASIC components
 As long as the values and tolerances are the same, go find a cheaper / BASIC version
it's easy enough to filter the JLCPCB choice menus, probably take you half a day to get them all changed over, create a new BOM and post it here for everyone to benefit

If footprint is different, just change it on the PCB layout.
OR, research and find ones with same footprint that are BASIC and cheaper
consider soldering the pinheaders yourself maybe? those are Through Hole so only available on extended

Component research and choice is probably the hardest and certainly the most tedious aspect of PCB design!

florianbrede-ayet, 2023-12-05:
@echoTheo some of the extended parts are actually *preferred extended parts* you're not being charged for.

Some others are non-replaceable (regulator & transceiver).

What is replaceable are the capacitors, but I had bad luck with cheap caps (multiple boards I had assembled in 2018-2020 failed already with blown caps), so I chose TDK & Panasonic instead.

As @bleughb said, you can just update the `bom.csv` with cheaper basic parts matching the footprint.

For the diode, JLCPCB didn't have a basic part available with more than 150mA continuous current.
While this is debatable, I wanted to play safe since the controller can exceed 200mA.

JanM321, 2024-02-03:
Closing because this has been answered.

closed 2024-02-03 by JanM321