JanM321/esphome-lg-controller #24: Making use of the WIFI dongle connector on the indoor unit PCB

janeczku, 2023-12-11:
Many current LG AC indoor units seem to have a connector socket on their PCBs to attach WIFI and Zigbee dongles. This socket, typically marked CN_WF, CN_WIFI or CN_LINK appears to provide a standard UART interface instead of the 1-wire bus on the CN_REMO socket.

Using this socket for the esphome controller would probably have some advantages over using CN_REMO:

- Wired LG remote controllers can still be used because we don't occupy the CN_REMO connector
- Cheaper PCB design (e.g. no transceiver needed)

Is there any information on the serial communication protocol used on the CN_WIFI port?

Found some details on the physical characteristics of the CN_WIFI connector here: https://fccid.io/BEJPCRCUDT2/User-Manual/User-Manual-1911954.pdf

JanM321, 2023-12-11:
> Is there any information on the serial communication protocol used on the CN_WIFI port?

There was a little discussion about this in the [Home Assistant forum thread](https://community.home-assistant.io/t/lg-ac-wired-controller-integration-via-esphome-esp32/582954). Unfortunately I don't think there's any information about this available.

The main reason for starting this project was to make it possible to use a HA temperature sensor because especially in heating mode the internal one just doesn't work well for me. I don't know if the Wifi interface can do this, but probably not because LG has no hardware/software for this AFAIK.

> Wired LG remote controllers can still be used because we don't occupy the CN_REMO connector

The same is true for the Wifi controller though :) My units came with a WIfi module but no wired controller, so with CN-REMO I can still use the LG app. This is important because not all features are exposed to the wired controller.

It is possible to connect two CN-REMO controllers but one of them needs to be in 'slave' mode. This would require some software changes though. It shouldn't be too hard to implement but I've never tried this.

If the Wifi interface can be used with an external room temperature sensor it would be very interesting, because at least for my units it is more feature complete than the wired controller protocol. Without this, I'd need to connect to both CN-REMO and CN-WIFI which seems a bit excessive and error-prone.

JanM321, 2024-01-15:
I'm closing this because even though this would be really interesting to explore (not all features are exposed to the wired controller unfortunately), I think this should be a separate project if someone manages to figure out how this interface works.