JanM321/esphome-lg-controller #25: invalid checksum

Arunas-g, 2023-12-20:
Hello. I followed everything according to the instructions, but the air conditioner control does not work. Indoor unit PC12SQ. Logs from ESPhome are attached
Tell me, what am I doing wrong?
[logs_template__turn_on_off.txt](https://github.com/JanM321/esphome-lg-controller/files/13724656/logs_template__turn_on_off.txt)
[logs_template_logs_rest.txt](https://github.com/JanM321/esphome-lg-controller/files/13724657/logs_template_logs_rest.txt)
[logs_template_long.txt](https://github.com/JanM321/esphome-lg-controller/files/13724658/logs_template_long.txt)
[logs_template_very_verbose.txt](https://github.com/JanM321/esphome-lg-controller/files/13724991/logs_template_very_verbose.txt)

JanM321, 2023-12-20:
In the first log I see multiple "invalid checksum" errors. The messages the controller is sending it should normally receive also, but they are being corrupted for some reason. Could it be an issue with the cable? Which PCB are you using?

Arunas-g, 2023-12-20:
> In the first log I see multiple "invalid checksum" errors. The messages the controller is sending it should normally receive also, but they are being corrupted for some reason. Could it be an issue with the cable? Which PCB are you using?

Tiny hardware. I made the cable myself.
Contacts from the board:
+12 to RED
GND to BLACK
LIN to YELLOW
As I understand it, nothing is read from the air conditioner at all?

JanM321, 2023-12-20:
> Tiny hardware. I made the cable myself. Contacts from the board: +12 to RED GND to BLACK LIN to YELLOW As I understand it, nothing is read from the air conditioner at all?

I just looked at the 'long' log and around 12:27:25 the controller did manage to send a non corrupted message. The AC then responds to this (the C8 and C9 messages for example are from the AC).

So it's really close to working, it's just that most messages are getting corrupted. It also looks like receiving from the AC works better than sending. Which ESP32 are you using? Maybe an issue with the TX pin or connection to the transceiver on the PCB?

JanM321, 2023-12-20:
Yeah, looking at it more, it receives messages from the AC perfectly. Looks very similar to my units, no checksum issues at all.

It just gets stuck sending an AA message because this doesn't work right.

The ESP32 has separate pins for receiving and sending that are "combined" by the transceiver on the PCB.

Arunas-g, 2023-12-22:
> Yeah, looking at it more, it receives messages from the AC perfectly. Looks very similar to my units, no checksum issues at all.
> 
> It just gets stuck sending an AA message because sending doesn't work right.
> 
> The ESP32 has separate pins for receiving and sending that are "combined" by the transceiver on the PCB, so it's most likely a hardware issue with the send part.

I have another air conditioner, model LG TC07GQR. It looks the same and the cable to CN_REMO is the same. My board doesn't work with it either. I have another set of ESP32 + tiny controller, it doesn’t work exactly the same)
I tried to read the output signal with an oscilloscope. Some data is coming. Video in attachment. The screenshot shows one of the packets read from the LIN contact. The air conditioner was not connected.

Video:
[https://youtu.be/PNevPlBduw8](url)

Screenshot: 
[image]

My ESP32:
[image]

Could I have uploaded the firmware incorrectly?

JanM321, 2023-12-22:
So the second PCB + ESP32 fails with the same checksum errors when connected to the AC? Do you have a log file? Maybe it's an issue with the cable?

The firmware should be fine. It looks more like a hardware issue to me.

Arunas-g, 2023-12-23:
> So the second PCB + ESP32 fails with the same checksum errors when connected to the AC? Do you have a log file? Maybe it's an issue with the cable?
> 
> The firmware should be fine. It looks more like a hardware issue to me.

The second air conditioner TC07GQR also does not respond to commands from the PCB. The cable is homemade, made from two cables and soldered in the middle with tin. Photo as an attachment. I recorded one log with TC07GQR at rest and a second log when trying to turn it on/off from Home Assistant.
[logs_template_TC07GQR_rest.txt](https://github.com/JanM321/esphome-lg-controller/files/13760033/logs_template_TC07GQR_rest.txt)
[logs_template_TC07GQR_on_off_via_ESP.txt](https://github.com/JanM321/esphome-lg-controller/files/13760034/logs_template_TC07GQR_on_off_via_ESP.txt)
Cable:
[image]

Arunas-g, 2023-12-23:
Maybe the LIN transceiver is not a suitable modification?
[image]

JanM321, 2023-12-23:
> Maybe the LIN transceiver is not a suitable modification?

Did you go with a different LIN transceiver than the one in bom.csv?

Looking at the checksum errors, there is a bit of a pattern to it (it affects 0x00 bytes mostly).

JanM321, 2023-12-23:
The LIN transceiver must not have the "TXD dominant timeout" feature. The TJA1027 is fine. A timeout like this does match the checksum errors you're seeing.

Arunas-g, 2024-01-10:
> Ага, если посмотреть дальше, сообщения от АС он принимает отлично. Выглядит очень похоже на мои устройства, никаких проблем с контрольной суммой.
> 
> Он просто застревает при отправке сообщения AA, потому что отправка работает неправильно.
> 
> ESP32 имеет отдельные контакты для приема и отправки, которые «объединены» трансивером на печатной плате, поэтому, скорее всего, это аппаратная проблема с частью отправки.

I connected the ESP32 to the air conditioner and listened to the signal with an oscilloscope on ESP pins 25, 26 and on the LIN pin. On the LIN contact I did not see a single response from the air conditioner. When the ESP sends a packet of commands to the air conditioner, I see the same signal on all pins (25, 26 and LIN), but I do not see any responses from the air conditioner (or I do not understand something). Maybe the response from the air conditioner comes immediately after receiving the command and on the oscilloscope it looks like “one command”. I tested it on two of my air conditioners, the result is the same. Video attached
What else can I check to diagnose the problem?

LIN pin
[https://youtu.be/YzlR2Z7z900](url)

25 pin
[https://youtu.be/0cXhGTpUHMM](url)

26 pin
[https://youtu.be/I6PoE2UUWI4](url)

JanM321, 2024-01-10:
The AC will only send a response after it receives a valid message. There's a problem somewhere, maybe with your LIN transceiver or cable, that results in corrupt messages sent to the AC, so it doesn't send anything.

Your earlier logs show the AC sometimes responds, but only if a non-corrupt messages is sent and that's uncommon.

It  looks like an issue with the LIN transceiver or TXD pin, although the TJA1027 *should* work...

florianbrede-ayet, 2024-01-18:
@Arunas-g very interesting failure, it looks like specific bytes are quite consistently corrupted in the echoed message.
The TJA1027 is usually fine as Jan said, the TJA1029 is the variant with dominant timeout feature.

Are you located in the EU?
I still have a spare of the boards I had produced which work fine.

Arunas-g, 2024-01-26:
@JanM321 thanks for the help! @florianbrede-ayet thanks for the offer. Yesterday I received an order with new TJA1027 parts from another batch, re-soldered and now everything works as it should on two air conditioners TC07GQR and PC12SQ
I probably came across a defective batch of TJA1027, which is why there were “invalid checksum” errors. Thanks for your help, I think the issue can be closed
New TJA1027
[image]
Correct log
[image]

JanM321, 2024-01-26:
@Arunas-g That's great to hear! Yes the log looks good now 😄 

> on two air conditioners TC07GQR and PC12SQ

Thanks. I'll add these to the README as supported models (TC07GQR I haven't seen yet).