JanM321/esphome-lg-controller #27: Farenheit temperature sensors not converted to Celsius

kchen, 2023-12-31:
I installed my first controller, and found two issues relating to Farenheit usage:

* My temperature sensors are in Farenheit, but when a temperature of 77 F was reported, the controller software would interpret that as 77 C, and then send a reading of 35 C to the LG unit. I tried calling temperature_sensor->get_unit_of_measurement in LGController::get_room_temp() to detect whether a conversion was needed, but that seemed to give me an empty string. The bottom "Note:" box on https://esphome.io/components/sensor/homeassistant.html seems to imply that the unit of measurement is lost and unavailable.
* The thermostat that the controller software creates is set to 0.5 degree increments, and so when the thermostat is configured in Farenheit, this ends up being 0.5 degree Farenheit increments, which is undesired. The thermostat otherwise works fine and I didn't need to configure any conversions.

I'm new to ESPHome, so I'm not sure how to cleanly implement supporting Farenheit in addition to Celsius. I considered a substitution, but then didn't quickly find a way to assume Celsius as a default without breaking everyone else's pre-existing configuration. For now, I've just hardcoded these two changes in my local copy.

kchen, 2024-01-01:
In response to the second bullet point, I tried setting visual_current_temperature_step and visual_target_temperature_step to 1, and that seemed to work fine. I also tried 0.9 (to be equivalent to 0.5 degrees C), but that caused weird behavior in the thermostat card, where it would briefly show the 0.9 degree change as I pressed + and -, but then jump to the nearest degree within a second or so.

JanM321, 2024-01-01:
Are you using a formula now to convert the room temperature and target temperature from F to C and back?

I think a boolean in the YAML file would be nice for this. If true, we'd change visual min/max/step values and do the conversion when getting/setting room and target temperature.

The protocol also has some flags for C/F that I should look into. Probably not really an issue for you because your AC was already set to Fahrenheit mode.

kchen, 2024-01-01:
Yep, I'm using a formula to convert the room temperature in get_room_temp(). I've included my current local diff with the formula below.

The target temperature seems to automatically work without any changes on my part. My Home Assistant thermostat card displays the temperature in F and I configure the target in F, but esphome-lg-controller I guess seems to automatically receive the target temperature in C.

Good to hear about the protocol supporting F -- that would potentially fix some inconsistencies. (Right now, if I set the thermostat card in Home Assistant to 78 F, the controller sends 25.5 C (77.9 F), but then the light on the indoor unit displays 77 F. The controller still gets 25.5 C when reading back from the indoor unit, even though 77 F would be 25.0 C.)

I also just found one more issue, which is the fractional target temperature, since it's currently being compared to exactly 0.5, and Fahrenheit conversions include may include fractions other than 0.5. This change is also included in the diff below. The fractional target temperature change should also be useful for Celsius, in case of potential floating point arithmetic error. (This potential floating point arithmetic error is why there's a comparison to 0.49 rather than 0.5.)

```diff
diff --git a/esphome/lg-controller.h b/esphome/lg-controller.h
index 79ad6a9..0a16074 100644
--- a/esphome/lg-controller.h
+++ b/esphome/lg-controller.h
@@ -189,8 +189,8 @@ class LgController final : public climate::Climate, public Component {
         supported_traits_.set_supports_action(false);
         supported_traits_.set_visual_min_temperature(MIN_TEMP_SETPOINT);
         supported_traits_.set_visual_max_temperature(MAX_TEMP_SETPOINT);
-        supported_traits_.set_visual_current_temperature_step(0.5);
-        supported_traits_.set_visual_target_temperature_step(0.5);
+        supported_traits_.set_visual_current_temperature_step(1);
+        supported_traits_.set_visual_target_temperature_step(1);
 
         // Only override defaults if the capabilities are known
         if (nvs_storage_.capabilities_message[0] != 0) {
@@ -454,6 +454,12 @@ public:
 private:
     optional<float> get_room_temp() const {
         float temp = temperature_sensor_->get_state();
+
+        ESP_LOGD(TAG, "Temperature sensor is %f F", temp);
+
+        // Convert from Farenheit to Celsius.
+        temp = (temp - 32) * 5/9;
+
         if (isnan(temp) || temp == 0) {
             return {};
         }
@@ -464,7 +470,11 @@ private:
             return 35;
         }
         // Round to nearest 0.5 degrees.
-        return round(temp * 2) / 2;
+        temp = round(temp * 2) / 2;
+
+        ESP_LOGD(TAG, "get_room_temp() returning %f C", temp);
+
+        return temp;
     }
 
     static uint8_t calc_checksum(const uint8_t* buffer) {
@@ -575,10 +585,15 @@ private:
             target = MAX_TEMP_SETPOINT;
         }
 
+	// Round to nearest 0.5 C.  (Mostly just needed when converting from Farenheit.)
+	target = round(target * 2) / 2;
+	uint8_t target_base = uint8_t(target);
+	bool target_fractional = (target - target_base) > 0.49;
+
         // Byte 5. Unchanged except for the low bit which indicates the target temperature has a
         // 0.5 fractional part.
         send_buf_[5] = last_recv_status_[5] & ~0x1;
-        if (target - uint8_t(target) == 0.5) {
+        if (target_fractional) {
             send_buf_[5] |= 0x1;
         }
 
@@ -596,8 +611,8 @@ private:
             thermistor = ThermistorSetting::Unit;
             temp = 20;
         }
-        send_buf_[6] = (thermistor << 4) | ((uint8_t(target) - 15) & 0xf);
-        send_buf_[7] = (last_recv_status_[7] & 0xC0) | uint8_t((temp - 10) * 2);
+        send_buf_[6] = (thermistor << 4) | ((target_base - 15) & 0xf);
+        send_buf_[7] = (last_recv_status_[7] & 0xC0) | uint8_t(nearbyintl((temp - 10) * 2));
 
         // Byte 8. Request settings when controller turns on.
         if (is_initializing_) {
```

JanM321, 2024-01-01:
> Good to hear about the protocol supporting F -- that would potentially fix some inconsistencies.

IIRC it's just for F/C switching and it would still send/receive temperature values in C, but I'll double check.
 
> (Right now, if I set the thermostat card in Home Assistant to 78 F, the controller sends 25.5 C (77.9 F), but then the light on the indoor unit displays 77 F. The controller still gets 25.5 C when reading back from the indoor unit, even though 77 F would be 25.0 C.)

That reminds me, the LG ThingQ JSON protocol uses `TempFahToCel`/`TempCelToFah` tables sent by the unit (or the LG servers?). See [the code here](https://github.com/ollo69/ha-smartthinq-sensors/blob/167e4b28aaadfcb5617c2aa175a9d598f2e90a29/custom_components/smartthinq_sensors/wideq/core_util.py#L38-L68) and snippets below from my units (fortunately I had an old JSON file lying around).

So another option would be to use those tables and do a binary search or array lookup on them or something. I think I have to check how the PREMTB100 behaves. Maybe using a formula instead of the LG mapping would leave small inconsistencies like this?

```json
              "TempCelToFah": {
                "_comment": "## \uc12d\uc528 \ud654\uc528 \ubcc0\ud658 \ud14c\uc774\ube14",
                "data_type": "enum",
                "default": "60",
                "value_mapping": {
                  "0": 32,
                  "1": 34,
                  ...
                  "24": 76,
                  "25": 77,
                  "26": 78,
                  ...
                  "24.5": 76,
                  "25.5": 77,
                  "26.5": 79,
```
```json

              "TempFahToCel": {
                "_comment": "## \ud654\uc528 \uc12d\uc528 \ubcc0\ud658 \ud14c\uc774\ube14",
                "data_type": "enum",
                "default": "60",
                "value_mapping": {
                  "32": 0,
                  "33": 0.5,
                  ...
                  "75": 23.5,
                  "76": 24,
                  "77": 25,
                  "78": 26,
                  "79": 26.5,
```

JanM321, 2024-01-01:
The PREMTB100 behaves like the JSON file above: 78F => 26C.

Playing a bit with this in Python, for F to C the following function should match the LG conversion for all Fahrenheit values from 32 to 104:
```py
def f_to_c(f):
    if f <= 41:
        if f == 41:
            return 5
        return 0 + (f - 32) / 2
    if f <= 59:
        if f == 59:
            return 15
        return 6 + (f - 42) / 2
    if f <= 77:
        if f == 77:
            return 25
        return 16 + (f - 60) / 2
    if f <= 95:
        if f == 95:
            return 35
        return 26 + (f - 78) / 2
    return 36 + (f - 96) / 2
```
Normally each +1F = +0.5C, except for Fahrenheit values that map to integer Celsius values (in this case it "starts over").

JanM321, 2024-02-15:
Thinking about this a little more.. To properly support Fahrenheit we probably need:

1. Add either a Fahrenheit option to the YAML file or a binary switch.
2. In this mode, interpret the room temperature value as F and convert it to C given the LG mapping formulas above.
3. For the setpoint and values reported to HA, the conversion unfortunately happens in HA or ESPHome and this doesn't match the weird LG C/F mapping. Maybe we could adjust for this (real deg C <=> LG-compatible deg C) so everything is consistent.
4. In Fahrenheit mode, we should also set the protocol flag for in-Fahrenheit-mode.

It's an interesting engineering problem :) I'll look into it when I have some time. First on my list is master/slave support.

JanM321, 2024-02-16:
OK I added an option for Fahrenheit to the YAML file. I hope this works better now.

closed 2024-02-16 by JanM321

kchen, 2024-02-18:
Looks good to me so far. Thanks!

JanM321, 2024-02-18:
For posterity, this is the Fahrenheit <=> Celsius data from the ThinQ JSON file: https://gist.github.com/JanM321/15b5a8d23f08e5409b21a3b93f4cd562

commit 518515ec (2024-02-16): Add Fahrenheit mode based on YAML option (closes #27)