JanM321/esphome-lg-controller #28: Low-Medium and Medium-High fan speeds not supported

kchen, 2024-01-08:
My indoor unit supports the Low-Medium and Medium-High fan speeds (via the physical remote), but not via esphome-lg-controller. Surprisingly, changing to these fan modes is also not supported in the LG ThinQ Android app, although the "SmartThinQ LGE Sensors" Home Assistant integration does support them. (I think my indoor unit may also support fan mode 7 "power", although I'm not certain.)

(I see that lg-controller.h already has some support for these fan modes. I also see that the Climate component of ESPHome supports custom fan modes, which I assume are the relevant thing to use to implement this. I haven't dug through all the code to ensure that everything works the way I think it does, though.)

JanM321, 2024-01-08:
The protocol does have support for Low-Medium and Medium-High fan modes, but my units don't recognize these values if I send them manually. You could try this with your indoor unit (see protocol.md) but I expect it won't work, if your unit is similar to mine.

When using these fan modes with the wireless remote or ThinQ, the wired controller will just receive Medium, IIRC. At least the AC unit is smart enough to not reset it to the actual Medium fan mode later if the controller doesn't change it.

It's really unfortunate that the Wifi controller and wired controller have a different set of features.

florianbrede-ayet, 2024-01-16:
Some devices support customizable fan speeds through *Installer Setting 3*.
I played around with that a little and it was really nice, however the setting is persisting power cycles.
So every change will likely wear the controller's EEPROM / Flash.

Anyway, if *Installer Setting 3* is supported, you might find a speed configuration that satisfies your needs even with just 3 fan speeds available.

kchen, 2024-02-18:
I finally got around to testing this, and can confirm what you said. It appears that when I set the fan speed to 5 or 6, the change is ignored. In addition, it looks like the capabilities message also does not report supporting low-medium or medium-high.

I noticed that the capabilities message does report that my unit supports fan speed 7. While performing this testing, I had my unit on fan only mode, but when I set the fan speed to power, the unit automatically set itself to cooling mode and 64 F, so it appears that power is not a normal fan speed, so we probably wouldn't want it as one. (I've seen power mode in heat mode before using either the physical remote or the LG app, but similarly, I don't think it would be desirable as a normal fan option.)

closed 2024-02-18 by kchen

JanM321, 2024-02-18:
> I noticed that the capabilities message does report that my unit supports fan speed 7. While performing this testing, I had my unit on fan only mode, but when I set the fan speed to power, the unit [...]

That's good to know, I haven't tried this option yet. On the PREMTB100, 'Power Cooling' and 'Power Heating' are separate operation modes with fixed setpoint associated with it (30C / 86F for heating).

The ThinQ app has a Jet Mode switch. That would probably be the easiest way to implement this.

JanM321, 2024-02-18:
I did some testing and my units only support the Power Fan option in Cooling mode. I can turn on the Jet option in the ThinQ app in heating mode, but the unit just resets it after a few seconds and also sends 'medium' to the wired controller. It looks like having a wired controller connected prevents my units from using this in heating mode.

Maybe I'll get back to this in the summer. I also want to add a toggle for the 'auto dry' / 'auto clean' option but that's also easier to test when the units are in cooling mode.

landrysik, 2024-03-06:
> It looks like having a wired controller connected prevents my units from using this in heating mode.

Hi, thanks for all the work. I just started using this project and noticed that with the wired controller connected I can't set the Jet mode (at least while heating) on my AC (LG Standard plus, PC12SK). I would understand if it is not implemented in the code but how is it possible that I can't set this even with IR remote or LG ThinQ app? AC completely ignores the button press on IR remote or ThinQ app...

Do you think this can be fixed somehow? I used the Jet mode quite frequently as the high fan speed nicely mixes air around the whole room when nobody is home (so the noise is not an issue)
Thanks!