JanM321/esphome-lg-controller #30: Not an issue - TEsted working with LG Ducted

bleughb, 2024-02-04:
My LG Units are a 17.1kW Ducted
[B55AWYU3G5](https://www.manualslib.com/products/Lg-B55awyu3g5a-11273601.html) - Indoor Unit
and
[ABUW55G7SB0](Outdoor Unit)

and I have installed myself a PREMTB100 controller

Everything seems to be working just fine right now!

JanM321, 2024-02-05:
Thanks! I added it to the list. Good to hear it works with ducted units too.

Are there things you still use the PREMTB100 for because they're not supported by the controller? Maybe zone control?

closed 2024-02-05 by JanM321

bleughb, 2024-02-05:
> Thanks! I added it to the list. Good to hear it works with ducted units too.
> 
> Are there things you still use the PREMTB100 for because they're not supported by the controller? Maybe zone control

We use the controller for a quick easy override / kids can control it.
still haven't fully adopted Home Assistant everywhere, 
plus it looks good on the wall.

I'm not using the zone control - which is supported, but plan to hook up our 3 zones and add more in the future.

rajkundu, 2024-07-20:
I'm not sure if this is the right place to ask, but I have an LG air conditioning system that it is built into my apartment, almost like a "central" system (I believe it is a "ducted" system just like @bleughb's). Unfortunately, this means that I have no idea what my indoor or outdoor units are; all I know is that my thermostat is an LG PREMTB10U, and it has 3 wires - power, ground, and signal.

My understanding is that the usual use case of this community is with an LG wall-mount unit, connected to an outdoor unit, which supports a device like the PREMTB10U via the CN-REMO connection. However, most people do not have such a thermostat/remote installed by default, so they open up the indoor/wall unit and connect this device instead.

So, I am hoping that I should be able to replace my PREMTB10U with this project's PCB, pretty much exactly as it seems @bleughb has done. Does that sound right - that I could simply connect the same 3 wires from my PREMTB10U (power, ground, signal) to this project's PCB, and it would work?

I know you can't guarantee anything, of course, but what do you think? Anyway, thank you so, so much @JanM321 for your _**tremendous**_ work on this project!!!

JanM321, 2024-07-20:
> I am hoping that I should be able to replace my PREMTB10U with this project's PCB, pretty much exactly as it seems @bleughb has done. Does that sound right - that I could simply connect the same 3 [...]

I expect that to work. For what it's worth, in issue #1 someone else replaced a PREMTB10U controller with the PCB.

Some of the LG controllers *also* support an older protocol for which there is an ESPHome implementation [here](https://github.com/Flameeyes/esphome-lg-pqrcuds0), but I hope my project will work. The hardware side is the same, it's just a completely different message format.

> Regardless, thank you so, so much @JanM321 for your _**tremendous**_ work on this project!!!

You're welcome! Please let me know if it works :)

rajkundu, 2025-06-01:
I can confirm that this setup works, with the PREMTB10U in master mode and the ESPHome controller in slave mode. **Thank you!**

Bezzey, 2026-09-13:
@bleughb and @rajkundu how has performance been since it's now been a while? I'm researching and came across this project. I also have a B55AWYU3G5 Indoor Unit / ABUW55G7SB0 Outdoor Unit.

rajkundu, 2026-09-13:
@Bezzey It's been pretty good for me! Granted, I don't really use any advanced functions - just control the temperature and fan speed through Home Assistant.

Unrelated, but I made a batch of 5 PCBs a while ago with JLCPCB. Happy to sell at-cost + shipping (no profit). I'm in Boston, MA, USA. Let me know if interested, and no worries if not!

commit 3a8e0f5f (2024-02-05): Add supported ducted unit (closes #30)