JanM321/esphome-lg-controller #31: Hardware availability @ JLCPCB

jacobcurulli, 2024-02-05:
Hi,

I'm hoping for some help with the tiny hardware PCB. I've gone to order from JLCPCB but they don't have stock of the PANASONIC EEEFPC220UAR Capacitor. I've had a look but can't seem to find a suitable replacement as I'm not sure how far I can deviate from the specs. Does anyone more knowledgeable know what other capacitor might work?

Thanks!

JanM321, 2024-02-05:
@florianbrede-ayet do you still remember how you selected this one?

florianbrede-ayet, 2024-02-05:
@jacobcurulli I think I would just choose [C178597](https://jlcpcb.com/partdetail/Panasonic-EEEFT1V220AR/C178597) as replacement - although it's a little expensive @ $0.20 .

This is from the TPS82130 datasheet:
[image]

For our requirements, you can likely get away with any `SMD,D4xL5.8mm` capacitor that is `10+ μF at 10+ V`.

jacobcurulli, 2024-02-06:
Thank you for the help @florianbrede-ayet and @JanM321 
I’ve ordered the PCB’s and I’m excited to get it up and running. 

Has anyone tried getting an advantage air zone controller into home assistant?
[image]
[image]

JanM321, 2024-02-06:
> Thank you for the help @florianbrede-ayet and @JanM321 I’ve ordered the PCB’s and I’m excited to get it up and running.

Great. Please let us know if it works.

> Has anyone tried getting an advantage air zone controller into home assistant?

I haven't heard anything about this as part of this project. Some LG duct units and controllers also have support for zone control, but it's not implemented in the controller because I don't have experience with or access to such units.

kchen, 2024-02-18:
For what it's worth, I ran into a similar JLCPCB supply issue, and used EEEFN1V330R which seems to work fine. It's a 33 uF capacitor instead of a 22 uF one, but that seems probably fine based on the following portion of the datasheet:

> 8.2.1.2.3 Input and Output Capacitor Selection
> For the best output and input voltage filtering, low-ESR ceramic capacitors are required. The input capacitor
> minimizes input voltage ripple, suppresses input voltage spikes, and provides a stable system rail for the device.
> A 10-µF or larger input capacitor is required. The output capacitor value can range from 22 μF up to more
> than 400 μF. Higher values are possible as well and can be evaluated through the transient response. TI
> recommends larger soft start times for higher output capacitances.

rkotulan-ksy, 2024-02-19:
Can you please advise me what to replace Shenzhen Kinghelm Elec KH-2.54FH-1X19P-H8.5?
These are now sold out.

JanM321, 2024-02-22:
> Can you please advise me what to replace Shenzhen Kinghelm Elec KH-2.54FH-1X19P-H8.5? These are now sold out. What about this one https://datasheet.lcsc.com/lcsc/2306091006_HCTL-PM254-1-19-Z-8-5_C2897382.pdf ?

Yes, that part (C2897382) looks very similar and seems to be a good replacement (although I didn't compare the datasheets in detail). Please post here if it works so we can help others in the same position.

florianbrede-ayet, 2024-02-27:
I've updated hardware-tiny and created a PR: https://github.com/JanM321/esphome-lg-controller/pull/37
- C3 has been removed. As quoted above, the datasheet mentions a 22uf (HF) ceramic capacitor which is part of the reference design for ESP32 DevKits. The electrolytic output capacitor is not required.
- C2 had already been replaced with a stock part in the previous PR
- Small schematic fixes (PinSocket -> PinHeader, "No Connect")

All parts are currently in stock at LCSC - but it's fluctuating frequently. Maybe PCBWAY is a better manufacturer for the PCBs - the gerber, bom & position files should work for them as well.

kchen, 2024-02-27:
For anyone considering JLCPCB vs PCBWay:

JLCPCB was about $60 for 5 for me, including shipping. I received a quote from PCBWay (but did not complete an order), and PCBWay was $99.33 for 5 ($15.43 component cost, $30 assembly cost (which looks like it gives you free shipping?), $53.90 PCB cost). Don't know how much PCBWay is for a different quantity.

PCBWay seemed to be a more manual process after uploading the files, but that also meant that they took the BOM containing LCSC part numbers, and figured out what parts they should actually use.

kchen, 2024-02-27:
I guess PCBWay is $30 for assembly for 1-20 pieces, so I guess that price is fixed, and probably the component costs scale linearly.

JanM321, 2024-02-27:
> PCBWay seemed to be a more manual process after uploading the files, but that also meant that they took the BOM containing LCSC part numbers, and figured out what parts they should actually use.

That's also my experience: PCBWay is able to order parts they don't have (or propose replacements) but they're more expensive than JLCPCB and a bit slower. It was worth it for me when I started this project but it's great we now have a PCB that can be assembled by JLCPCB.

I wonder if there are good options for doing a bulk order of > 100 PCBs and DevKits and selling them somewhere. Maybe eBay or Tindie but I don't have the time to send packages every week and deal with problems.

rkotulan-ksy, 2024-03-01:
> Yes, that part (C2897382) looks very similar and seems to be a good replacement (although I didn't compare the datasheets in detail). Please post here if it works so we can help others in the same [...]

I can confirm that C2897382 work well.

TerryFrench, 2024-07-24:
FYI in case others find this here, I placed an order on JLPCB for 20 boards - I need 3 for myself, likely with 1 or 2 spares. 
Once I get them and install/test 3, I can sell them at cost (buyers will pay for shipping&packing, but I think that should be super cheap as small/lightweight). 
Note: I also replaced the out-of-stock C1 with a same size similar specs 10uF capacitor. (these components are +-20% anyway, and the circuit likely does not require something too precise).

rkotulan-ksy, 2024-07-30:
> FYI in case others find this here, I placed an order on JLPCB for 20 boards - I need 3 for myself, likely with 1 or 2 spares. Once I get them and install/test 3, I can sell them at cost (buyers will [...]

I would be interested in two pieces

TerryFrench, 2024-07-30:
> I would be interested in two pieces

I am supposed to receive them today. So I hope to install and test tonight. If all goes well, I'll post here. 
Question: would you need a cable made? I likely have enough connectors (to be verified, but for my Voron printer and some RC stuff, I got many connectors) and a good crimper. 

Note: I never tried it, but see if you can PM me. I'd rather not write my personal email here directly :-)

rkotulan-ksy, 2024-07-31:
> I am supposed to receive them today. So I hope to install and test tonight. If all goes well, I'll post here. Question: would you need a cable made? I likely have enough connectors (to be verified, [...]

Cabel would be fine. I use one with LG PREMTB001 in slave mode so cable (socket splitter) with two connectors on one side and one on the other. 
I did not find PM but my email is rudolf.kotulan at outlook.cz.

TerryFrench, 2024-08-02:
> Cabel would be fine. I use one with LG PREMTB001 in slave mode so cable (socket splitter) with two connectors on one side and one on the other. I did not find PM but my email is rudolf.kotulan at [...]

I got my 3 indoor unit working with the boards. 
Cables: I will likely not build special cables with some kind of socket splitter. 
Anyway, I just need to price that stuff (Add all I spent :-) ) 
How many boards, and will that be shipping (from the USA) to the Czech Republic? (not super sure about the country name, please assume I mean well). 
boards photos... https://github.com/jourdant/esphome-lgap/issues/10#issuecomment-2264531969

w-marco, 2024-09-10:
Hi!

I also have received 5 Tiny PCBs but only needed 2 (both are working beautifully). 

This means I have 3 spares, based in Germany (so EU wide shipping would be easy). 

If anyone would be interested, you can contact me via mail(at)marco-wack.de or Telegram (at)w_marco, so we could figure out payment and shipping.

kristoficko, 2024-10-01:
I wonder what has gone wrong here :) JLCPCB tiny-hardware + aliexpress [ESP32-WROOM-32E](https://vi.aliexpress.com/item/1005007055074944.html)
[image]
[image]

ChriD, 2024-10-01:
Wrong ESP
Missing pinout labels ;-)

kristoficko, 2024-10-01:
This is much bigger issue than missing pinout labels. 22mm instead of 23.5mm :( Chinese crap :(
[image]
[image]

bjw-s, 2024-10-14:
I've ordered 5 PCB's (because MOQ) on JLCPCB, and I only need 2. In case anyone is interested, I'm willing to sell the other 3 at cost + shipping. I'm based in NL, so shipping within the EU is probably easiest.

PVi1, 2024-10-16:
> I've ordered 5 PCB's (because MOQ) on JLCPCB, and I only need 2. In case anyone is interested, I'm willing to sell the other 3 at cost + shipping. I'm based in NL, so shipping within the EU is probably easiest.

I would like to purchase 2 units from you. I am located in the EU. Please send me an email at lg@vilhan.eu so we can arrange the details.

Thank you.

bjw-s, 2024-10-17:
> Hello @bjw-s, I would like to purchase 2 units from you. I am located in the EU. Please send me an email at [lg@vilhan.eu](mailto:lg@vilhan.eu) so we can arrange the details.

Email sent 🙂

kLeR1k, 2024-10-17:
Can't find in ease TLIN1027DRQ1. So trying to find more popular one.
Kicad project for Tiny hardware for U2 contents TJA1027, but datasheet says that it has dominant time-out function. Or it has very large time-out, so it not a problem?
Is it possible to use TJA1027 as replacement? or any other applicant?

Mrsash2020, 2024-10-21:
> I've ordered 5 PCB's (because MOQ) on JLCPCB, and I only need 2. In case anyone is interested, I'm willing to sell the other 3 at cost + shipping. I'm based in NL, so shipping within the EU is probably easiest.

Do you still have any spare? Will take 1 for my split. I am in Australia. Drop me an email on emailspam2024@gmail.com

qurpi, 2024-10-29:
Anyone looking to sell some boards? Happy to order some as well with other people, looking to split the cost a little.

qurpi, 2024-10-29:
PANASONIC EEEFC1C100AR 10uF 16V 45mA@120Hz ±20% SMD,D4xL5.4mm is not in stock, any other alternatives? Will this work (ST(Semtech) CK1C100MCRC54)? Sorry I have no experience in electronics.

Mrsash2020, 2024-10-29:
I am happy to have one. Does it com assembled? If anyone orders a few, happy to pay for one.

kLeR1k, 2024-11-06:
> Can't find in ease TLIN1027DRQ1. So trying to find more popular one. Kicad project for Tiny hardware for U2 contents TJA1027, but datasheet says that it has dominant time-out function. Or it has [...]
@florianbrede-ayet Can you help with this please? AFAIK you a pro at this question

qurpi, 2024-11-07:
Ordered, only need 1 board. Will update once I receive them and then I can start shipping out the other 4 boards.

s50en, 2024-11-12:
> Ordered, only need 1 board. Will update once I receive them and then I can start shipping out the other 4 boards.

Would be interrested in one of the Boards skilift9@gmail.com

Mrsash2020, 2024-11-13:
> Ordered, only need 1 board. Will update once I receive them and then I can start shipping out the other 4 boards.

I'll take 1 but maybe two if I can use it for the fujitsu split as well. [emailspam2024@gmail.com](mailto:emailspam2024@gmail.com)

gabi1324, 2024-11-30:
> PANASONIC EEEFC1C100AR 10uF 16V 45mA@120Hz ±20% SMD,D4xL5.4mm is not in stock, any other alternatives? Will this work (ST(Semtech) CK1C100MCRC54)? Sorry I have no experience in electronics.

Hi! I'm also trying to order the PCBs but the Panasonic capacitor is out of stock. Did you find out if the Semtech one is ok?

teh2wsx, 2024-11-30:
Hello @bjw-s 
Do you still have one unit left? If yes, I'd like to purchase it from you. Please drop me a line @ teh2wsx@gmail.com
I'm based in EU. 
Thanks in advance for your help

bjw-s, 2024-12-02:
> Hello @bjw-s Do you still have one unit left? If yes, I'd like to purchase it from you. Please drop me a line @ [teh2wsx@gmail.com](mailto:teh2wsx@gmail.com) I'm based in EU. Thanks in advance for your help

Email sent :)

Mrsash2020, 2024-12-05:
I'll take one as well if you have it.

bjw-s, 2024-12-06:
I'm all out of boards (only have 1 left that I am keeping as a spare for myself)

JanM321, 2024-12-14:
@ExquisiteMetropolis which parts are currently missing? Maybe others subscribed to this issue can help out.

ExquisiteMetropolis, 2024-12-14:
Excellent suggestion.
Please see the screenshot below of what is shown once the BOM and Placement File are uploaded 
[image]

JanM321, 2024-12-14:
> Excellent suggestion. Please see the screenshot below of what is shown once the BOM and Placement File are uploaded !

I'd suggest ordering the hardware-tiny PCB instead. For that one most parts are available for me now except C1, a Panasonic capacitor. You can select a replacement for it by using the Search icon next to it.

@qurpi mentioned one option here: https://github.com/JanM321/esphome-lg-controller/issues/31#issuecomment-2444519985 Did that one work for you, qurpi?

ExquisiteMetropolis, 2024-12-14:
Fortunate that it failed, as I was under the impression I was ordering the tiny one.
Should I use the files in this folder? https://github.com/JanM321/esphome-lg-controller/tree/main/hardware-tiny/production

lagomCat, 2025-06-08:
And I want to try this version of the electronic board))

[image]

dritchie, 2026-09-17:
If anyone's still watching this thread: I recently got the Feather hardware configuration working, and I have a few extra Feather boards that I'd be willing to sell for $30 each (+ shipping). Happy to ship within the continental US. Reply here with your email address if you're interested, and I'll follow up.

Kalored, 2026-10-16:
Hey I have just taken my S24EQ Unit online in HA and I am sitting here with 3 PCB for Feather S3 (without parts) (latest revision July 26) and 1 Module which is just fully equipped with all parts (only missing the Feather S3 Board). 
I would ship within EU.
For bare PCB 10€ each + shipping
For populated PCB 30€ + shipping

[image]

[image]

kearfy, 2026-10-16:
Hey @Kalored, I'd like to buy the equipped board! I live in the Netherlands, could you shoot me an email to micha@devrie.sh ? Thanks! 🙏