JanM321/esphome-lg-controller #33: Support for a LG LZ-H100GBA5 Heat Recovery Ventilation Units

rht321, 2024-02-11:
Hi, 
I have connected one of your "hardware-tiny" boards to an LG LZ-H100GBA5 Heat Recovery Ventilation Unit.  
The wired control is a PREMTB100.
I can see the esphome-lg-controller sending a 'A8' message and the lg-unit responding with a 'A8' message followed by a D0, D1, D3, D4 and D7 messages.
I would like advice on how to reverse engineer this.  From your code I get the feeling I can wire the PREMTB100 and the "hardware-tiny" board in parallel?  
My thinking here is to use the PREMTB100 to generate the commands to the lg-unit and use the esphome-lg-controller to sniff the link.  
For that to work I assume I will need to change the esphome-lg-controller to a slave unit.  Is there a setting for this?
Thanks for any help and the work you guys have put into this, the progress is brilliant.

JanM321, 2024-02-11:
It's probably easiest to connect both the PREMTB100 and the ESPHome controller, but change the ESPHome code to never send any messages (add an early return to the `update` function, before the `while (true)` loop). You should then be able to sniff the messages between the PREMTB100 and the unit.

I don't know how similar these messages are to the HVAC protocol, but I think D0 is the equivalent of C8 (status message), D1 is similar to C9 (capabilities message), etc. See protocol.md for details.

> and the lg-unit responding with a 'A8' message

This is likely coming from the controller. It's a single wire bus so the controller will also receive the messages it sends.

JanM321, 2024-02-11:
Btw, D0-D7 messages are interesting. I wonder if the "Product type" bit in protocol.md is actually two bits instead of one.

If that's the case, then your unit sends D0-D7 messages (instead of C8-CF for AC units) and the controller sends B0-B7 (instead of A8-AF for AC units). Let me know if you see the PREMTB100 send B0 messages and I can improve the docs.

rht321, 2024-02-11:
[image]

So have disabled the send message.  From a fresh boot up I get the above log

rht321, 2024-02-11:
[image]
Also captured B0 and B4 messages

rht321, 2024-02-11:
the 0xB0 and 0xC0 messages

[image]

JanM321, 2024-02-12:
Great progress! I improved the documentation for the first message byte in protocol.md based on this.

What kind of features does your unit have that you want to make available in HA?

These messages are pretty different from the HVAC messages, even though the format is *similar*. If you want to make an ESPHome controller for your unit, I think it makes most sense for it to be a separate project, because trying to handle both message formats with the same code will likely end up being more complicated than having a version of the code specialized for ventilation units.

closed 2024-05-01 by JanM321

Sakab001, 2025-10-22:
Hi, I also have the same heat-recovery ventilator, and I’d like to integrate it into Home Assistant. I was wondering if you’ve managed to create the YAML for controlling the ventilator yet, or how far you’ve gotten with the project? I’m currently at this point:

[image]