JanM321/esphome-lg-controller #35: using home assistant temperature sensor.

cwtufts, 2024-02-15:
I am trying to figure out how to use a home assistant temperature sensor entity. Which file should that be placed in?

JanM321, 2024-02-15:
The template YAML file has this line:
```yaml
temperature_sensor_entity_id: sensor.my_sensor # XXX
```
Here you should put the entity id of the Home Assistant temperature sensor you want the controller to use.

If you want to use a template sensor, you can put those in Home Assistant's configuration file. I have mine in configuration.yaml.

cwtufts, 2024-02-15:
Okay that is what I have. For some reason it is reading 95 F when the temperature is 73.89. Could this be an issue of the temp being in F instead of C?

[image]

JanM321, 2024-02-15:
> Okay that is what I have. For some reason it is reading 95 F when the temperature is 73.89. Could this be an issue of the temp being in F instead of C?

Oh right, Fahrenheit isn't well supported atm. See also issue #27

JanM321, 2024-02-16:
I made some changes to better support Fahrenheit. You can set the Fahrenheit option to "true" in the template yaml file with the latest version. Please let me know if this works better now.

JanM321, 2024-02-22:
Closing because the new Fahrenheit mode fixes this.

closed 2024-02-22 by JanM321