JanM321/esphome-lg-controller #38: Jet mode (heating) blocked with wired controller connected

landrysik, 2024-03-07:
Hi, thanks for all the work. I just started using this project and noticed that with the wired controller connected I can't set the Jet mode (at least while heating) on my AC (LG Standard plus, PC12SK). I would understand if it is not implemented in the code but how is it possible that I can't set this even with IR remote or LG ThinQ app? AC completely ignores the button press on IR remote or ThinQ app...

Do you think this can be fixed somehow? I used the Jet mode quite frequently as the high fan speed nicely mixes air around the whole room/house when nobody is home (so the noise is not an issue)
Thanks!

JanM321, 2024-03-07:
I've noticed this too and I'm afraid it can't be fixed, at least not in heating mode :/

My AC reports that it only supports Power Cooling but not Power Heating (the LG wired controllers use these names instead of "Jet Mode" but I think it's the same thing), and it looks like the AC internally blocks this feature when a wired controller is connected. Trying to enable it anyway through the wired controller also doesn't work.

There are some other things that LG could have easily supported but just didn't (for example the Low/Medium and Medium/High fan modes are supported by most controllers and the protocol, but they just didn't expose them).

JanM321, 2024-03-07:
It might be worth checking how long this feature remains blocked if the controller is disconnected or stops sending messages. If Jet Mode becomes available again after a few minutes, a workaround would be for the controller to have an option to stop sending messages.

kchen, 2024-03-07:
In my experience, if there is a controller connected and then it becomes disconnected, after at most a few minutes, the indoor unit displays an error code (on its display and in the app) and effectively turns off. It seemed that if I wanted to leave the controller disconnected, I needed to powercycle the unit. Reconnecting the controller without powercycling works fine.

(I noticed this both because of some loose controller wiring I had at one point, and then also when I intentionally disconnected the loose wiring to fix it, the error also came up.)

landrysik, 2024-03-07:
> In my experience, if there is a controller connected and then it becomes disconnected, after at most a few minutes, the indoor unit displays an error code (on its display and in the app)

Yes I can confirm this - I just tested to OTA flash empty ESPhome code and after few minutes the display is blinking CH 03. I tried turning it on/off with remote and after reflashing controller code using the controller but no lock. Had to to 230V power cycle

belly-89, 2024-05-13:
Jet mode seems also to be unavailable in cooling mode! Too bad!