JanM321/esphome-lg-controller #39: Over heating settings - what exactly does it change? + EEV lifetime?

landrysik, 2024-03-11:
Hi, I am trying to understand the logic behind the AC's internal algorithm that regulates the actual compressor power. It is (should) be closely related with the "over heating setting" but what those numbers even mean? (+4/+6°C). What happens when the tempeature is (presumably) +4°C from setpoint and what happens at +6°C? What are the default (0) values?
I am using LG Standard plus PC12SK

I have it set to +-1°C in hopes it will better regulate indoor temperature and I first thought it will just get to the setpoint +1°C and slow down the compressor and if it gets -1°C from setpoint it will speed up. But the "black box" is more complicated than this. When the temperature started to rise quickly I was still -1°C from setpoint and at this point the AC lowered compressor power.

**Does anybody have more observations? Maybe we can get a better picture how it works together.**
One observation I can share is from before I started using this wired controller: 
I hardwired the actual thermistor input to fixed resistor to simulate 31°C. When set to 30°C and turned ON, the AC started at nominal power (around 900W input) and stayed there forever. When I wanted lower power I set the fan speed to low and the AC lowered power to around 500W within few minutes. When I then set fan speed back to HIGH it kept the 500W again forever.
After defrost it started again at 900W. 

It is honestly really hard to get relevant data with a repeatability to understand the behaviour because sometimes even after power cycle the AC sets different power level even with the same setpoint + same measured temperature. Maybe outside temperature plays a role aswell? Or pipe temperatures/pressures/ time from the last compressor start?...

I tested the workaround template sensor that fakes the temperature reading but in my conditions it didn't work that great. The AC was constantly modifying the power up and down.*  Maybe it should be "tuned" to specific room and the thermal inertia. 
>  To work around 2), if the temperature drops below `setpoint - 0.1` and it has been heating for more  than 10 minutes, it forces the unit to work harder by reporting `setpoint - 2.5`.

I got a feeling that 2.5°C with just 0.1°C difference was too excessive. The AC started ramping the power too quickly and too much and overshoot the target temperature. It got me thinking that maybe report setpoint -2.5°C but only for a few minutes to get the AC power going up and then lower it to only setpoint +1 or even real temp so it won't ramp up that much? I have to test it further but we currently have quite a warm weather...

### * EEV?
And one more thing to consider. As we see the AC is really lazy changing the compressor speeds. It likes to set one speed and let it run for hours even when the temperature is fluctuating. This got me thinking that maybe the manufacturer set the software this **poorly on purpose!** I am not a HVAC expert but the EEV (Electronics Expansion Valve) needs to change position every time the pressure changes to maintain ideal pressure difference for refrigerant evaporation. So with every compressor speed change the EEV needs to change aswell. And those EEV's have for sure limited cycle lifespan. So in typical conditions, even "shitty" EEV and compressor system can last projected 20 years or so...But once we start to mess with the AC by reporting fluctuating temperature and force it to change the compressor speed every 10minutes, those 20years can quickly shrink to much less.... Maybe I am just overly cautious but be aware of this aspect...

JanM321, 2024-03-11:
> Hi, I am trying to understand the logic behind the AC's internal algorithm that regulates the actual compressor power. It is (should) be closely related with the "over heating setting" but what [...]

+4/+6 means it stops heating at `setpoint + 6` and then starts heating again when the room temperature reaches `setpoint + 4`.

> What are the default (0) values? I am using LG Standard plus PC12SK

It's the default behavior for the unit. For my units that's +1/+3 but it could be different for other units.

> Maybe outside temperature plays a role aswell?

My units do modulate better (less Watt) when it's cold outside. I agree that the behavior of the unit is hard to predict unfortunately :/

JanM321, 2024-03-12:
> This lower wattage does not come from any active modulation. It is purely physics. Lower temperature = lower pressures and lower heat transfer. This means that at the constant speed the compressor does not work so hard and has lower power input. 

I don't think it's that. A few degrees C can be the difference between < 300W or > 450W. This seems to be a multi-split issue though - single split units and units from other brands don't seem to have this.

SiboutVanLoo, 2024-07-16:
> I don't think it's that. A few degrees C can be the difference between < 300W or > 450W. This seems to be a multi-split issue though - single split units and units from other brands don't seem to [...]

Yes it can, depending on the indoor setpoint vs indoor temp. And time cycle delays, (internal PID)
Combined with Outdoor temp and RF-gas properties.
Single split units are better at modulating if correctly sized in power for the room.
Multi splits have a bigger "engine" they need to manage demand and offer more, especially if only 1 unit of 4 is running. (for example.)

TerryFrench, 2024-08-02:
I'll just chime in with some of my understanding. I might be wrong of course, so I'm trying to share, not to convince :-) 

- The efficiency of the HVAC is better when the blower in the indoor unit is at max speed or close to it. That means that real-world experience vs. testing by the manufacturer in a lab with noisy fan setting are different. 
- for multi units (I've one: one ODU with support for 4 IDU, but I piped only 3), the various manuals show that performance is not as good as single split units, and also the numbers are best when the load of all indoor units does match the outdoor unit's rating & design. Rule of thumb, if the load is < 50% of the ODU Capacity (cooling or heating), efficiency is degraded. And likely at 100% or some pretty high percentage, the efficiency is maximized. 
- EEV: they are made to be used all the time. Same as modulating the compressor's rpm. But on cheap boards, it's likely some lookup tables for some values based on the inputs. i.e. the MB is pretty dumb, all calculations were pre-made during the R&D time. 
- In the experiment above, with 30C-31C... if that was for cooling, and you are faking the indoor temperature, "maybe" it is valid. 
Basically the HVAC systems we have are still quite basic in design, and you can expect the firmware to only handle specific conditions. All the sensors on the ODU are important and working together. You can also only test cooling mode indoor (besides install & emergency modes) when the outdoor temperature is within some range. Same for testing the heat-pump mode. So it's hard to get reproducible data. 
Also note that in heat-pump mode, the ODU has to manage freezing coils, so it's behavior will seem less constant and predictable. And again, you have to take the real outdoor conditions and cannot fake them.