JanM321/esphome-lg-controller #41: Half working with DC18RQ.NSK

robertklep, 2024-03-16:
I'm trying to get your controller code working using @AussieMakersGeek's [level shifter/opto isolator](https://github.com/AussieMakerGeek/LG_Aircon_MQTT_interface#connections) setup with a DC18RQ.NSK model.

It seems to be working _somewhat_, in that I see certain sensors being updated in HA (for instance, the outdoor unit and preheat states). Also, I can change the airflow sensor in HA and the AC picks it up.

However, a lot of things don't work: reading/setting the mode, reading/setting the target temperature, etc.

The ESP32 is logging a lot of errors:
```
[12:09:27][D][lg-controller:1305]: update
[12:09:27][E][lg-controller:1324]: did not receive message we just sent
[12:09:33][D][lg-controller:1305]: update
[12:09:39][D][lg-controller:1305]: update
[12:09:45][D][lg-controller:1305]: update
[12:09:45][E][lg-controller:1346]: discarding incomplete data A9
[12:09:51][D][lg-controller:1305]: update
[12:09:52][D][lg-controller:861]: sending AA.00.00.00.00.00.00.01.00.02.F1.21.EA (13)
[12:09:57][D][lg-controller:1305]: update
[12:09:57][D][lg-controller:905]: received AA.C8.30.00.00.40.00.16.00.00.00.00.00 (13)
[12:09:57][E][lg-controller:915]: invalid checksum AA.C8.30.00.00.40.00.16.00.00.00.00.00 (13)
[12:09:57][E][lg-controller:1324]: did not receive message we just sent
```

I'm not sure where to look. I'm perfectly willing to accept that the hardware setup that I'm using might be causing these issues, but it seems strange that some TX/RX's work (consistently) and others don't, which suggests that it might be caused by my AC perhaps using a different protocol version (if that makes sense).

JanM321, 2024-03-16:
> which suggests that it might be caused by my AC perhaps using a different protocol version (if that makes sense).

That's unlikely. Your unit is very similar to other units that work fine with this controller.

In the log output you posted, the controller is trying to send an AA message (which controls the airflow up/down setting among other things), but there might be a conflict with the AC sending a C8 message (status message) at the same time. In this case the controller will keep sending this message until it succeeds.

Did you change the RX/TX pin numbers? In that case make sure you also change `RxPin` in the header file.

JanM321, 2024-03-16:
If you can pastebin the full log output (including controller startup + a few minutes after that) I can take a look to see if there's anything unusual.

robertklep, 2024-03-16:
Thanks!

Here's the log: https://pastebin.com/Qk08fFkK

I did change the RX/TX pins but edited the header file accordingly.

A few things that might be relevant:
* the original schematic is showing two 270Ω resistors, I only had 220Ω. Probably not an issue though?
* the original schematic is powering the level shifter from the 5V pin from an Arduino, but since I'm using an ESP32 I assumed I need to drive it from 3.3V. However, I didn't change the resistor to the optocoupler, perhaps that's problematic? I do see the LED flash whenever a message is being sent.

JanM321, 2024-03-16:
> Here's the log: https://pastebin.com/Qk08fFkK

14:09:45: the controller sends its initial message and reads it back on the bus. The AC doesn't respond to this for a while (this means messages aren't received by the AC or are corrupted somewhere)
14:13:58: finally the unit responds by sending all its settings (part of initialization)
14:14:22: controller tries to send an AA message (it always does this after initialization) but it never reads it back on the bus, so it gets stuck trying to send this. The controller is still receiving messages from the AC occasionally.

This explains what you're seeing. The controller is stuck trying to send an AA message (this has the airflow up/down setting but not the other settings) because it fails to read it back on the bus. The controller is still processing status messages from the AC so sensors are updating.

It's very close to working (you can both send and receive messages sometimes) but communication is buggy because something isn't quite right.

Are you using a 12V / 3.3V level shifter? It could be the resistor on the ESP32 side maybe, but I'm not as familiar with this schematic.

robertklep, 2024-03-16:
The level shifter I'm using should support 3.3V on the low side and 12V on the high side, but I don't necessarily trust it. I'll see if I can source a proper 12/3.3V level shifter just to rule out that this is the cause.

I'll try and see if I can the resistor on the ESP32 side to match the lower voltage.

Thanks for now, I'll update this issue when I find out more.

robertklep, 2024-03-20:
I took the easy way out and bought two of your controllers, courtesy of @belly-89 😅

I do have an additional aircon unit (one we don't often use) that I might try when I receive the other level shifters, but I'll close the issue for now. Thanks again 😊

robertklep, 2024-03-20:
FWIW, you can add the DC18RQ.NSK to the list of supported devices. I didn't have an ESP32-DevKit module available so I hooked it up to an ESP32-C3 SuperMini board, which works fine (I'll clean up the wiring at a later date 😁).

[image]

closed 2024-03-20 by robertklep

JanM321, 2024-03-22:
> FWIW, you can add the DC18RQ.NSK to the list of supported devices. I didn't have an ESP32-DevKit module available so I hooked it up to an ESP32-C3 SuperMini board, which works fine (I'll clean up the wiring at a later date 😁).

Done! I'm glad it works now with the PCB.

belly-89, 2024-03-22:
> FWIW, you can add the DC18RQ.NSK to the list of supported devices. I didn't have an ESP32-DevKit module available so I hooked it up to an ESP32-C3 SuperMini board, which works fine (I'll clean up the wiring at a later date 😁).
> 
> ![F1B7FAA2-C541-471D-9600-76C169510A93](https://private-user-images.githubusercontent.com/1190916/314492419-0dc14952-0014-43d6-b332-f0a7d3d5e10d.jpeg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTExNDUyMDcsIm5iZiI6MTcxMTE0NDkwNywicGF0aCI6Ii8xMTkwOTE2LzMxNDQ5MjQxOS0wZGMxNDk1Mi0wMDE0LTQzZDYtYjMzMi1mMGE3ZDNkNWUxMGQuanBlZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzIyVDIyMDE0N1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTExODczZTczZDVjNzU5YzZkY2I5ZGUxNzQzNzJlMmRiODhjYzA3ZjI3ZDM4NmIzMTgzMmRiNDY5Y2ViNzBkNDQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.HQtMEHCiiidZcpjiDdVPd-T-V4dfFrCGS65YL-hFwrY)

Nice to see it working! 🙂