JanM321/esphome-lg-controller #43: Quick temperature sensor choice

pieca2, 2024-03-31:
Hi,

it isn't a real issue, but I'm using an Aquara temperature sensor and it is too slow to properly control my HVAC quickly. it could wait 10 min to report a change from 1°C...

What temperature sensor are you recommending for a fast response when temperature is changing and get a proper control ?

Thanks for your help

landrysik, 2024-03-31:
I think the best temperature sensor for this purpose is LYWSD03MMC or similar BLE thermometer supported with firmware from pvvx. You can directly receive the BLE signal using the same ESP32. It updates every few seconds and have 0.0X °C resolution

JanM321, 2024-04-01:
I use the Xiaomi Mi 2 sensors with pvvx firmware. You can flash the firmware here: https://pvvx.github.io/ATC_MiThermometer/TelinkMiFlasher.html

I use this with an M5Stack Atom Lite. You can easily set these up as bluetooth proxy: https://esphome.io/projects/?type=bluetooth

landrysik, 2024-04-03:
With some code modification (haven't tried it yet) it would be possible to receive the BLE transmissions by the ESP32 and directly use this value. Use this code instad BT proxy.
https://esphome.io/components/sensor/xiaomi_ble.html#lywsd03mmc
As an advantage this will work independently on HA server. You could install this in your friend's AC even if they don't use HA at all

JanM321, 2024-04-03:
> With some code modification (haven't tried it yet) it would be possible to receive the BLE transmissions by the ESP32 and directly use this value.

True, I've thought about doing this or enabling the bluetooth proxy feature, but I like the flexibility of a HA sensor (I actually use a template sensor). The ESP32 also uses more power when BLE is enabled. That's probably okay but when I started this project I wanted to play it safe and keep it to a minimum.

JanM321, 2024-04-08:
I added a link to this issue to the Tips section in the README.

I think we can close this but anyone feel free to add additional comments.

closed 2024-04-08 by JanM321

pieca2, 2024-04-08:
**Issue solved** using an ESP32 mini D1 with a Dallas DS18B20 sensor directly soldered on the board with a resistor & Esp home.
This DIY sensor report temperature each 30 seconds with a great precision and from know my HVAC temperature is really stable and well regulated !!!!

rajkundu, 2025-04-02:
I'm so sorry if this is either obvious or already answered elsewhere, but I couldn't seem to find the answer: Is there any way to use the built-in LG temperature sensor instead of a custom sensor on HA? Is it okay to have an empty or invalid value for `temperature_sensor_entity_id` in template.yaml?

My current goal is to try and use the ESP controller in slave mode while preserving my original, wall-mounted PREMTB10U in master mode.

JanM321, 2025-04-02:
> I'm so sorry if this is either obvious or already answered elsewhere, but I couldn't seem to find the answer: Is there any way to use the built-in LG temperature sensor instead of a custom sensor on [...]

I'm not 100% sure but I *think* you can use an invalid sensor id such as `sensor.invalid`...

> My current goal is to try and use the ESP controller in slave mode while preserving my original, wall-mounted PREMTB10U in master mode.

In slave mode the temperature sensor isn't used, the unit will get the temperature from either the unit itself or the master controller.

rajkundu, 2025-04-02:
Thank you very much! I will try using `sensor.invalid` for now and proceed from there.

rnivz, 2025-04-20:
Hi,

It would be great if it were possible to use the Xiaomi LYWSD03MMC (pvvx) with the ESP32 board.
Has anyone already implemented this or a guide with the necessary steps? Including the required adjustments to the YAML file?

JanM321, 2025-04-21:
> It would be great if it were possible to use the Xiaomi LYWSD03MMC (pvvx) with the ESP32 board. Has anyone already implemented this or a guide with the necessary steps? Including the required adjustments to the YAML file?

Someone added support for using the ESP32 as a bluetooth proxy (I haven't tested this myself). For this you need to uncomment [these lines](https://github.com/JanM321/esphome-lg-controller/blob/89ce8bd4a95a039eef131e744925303344950f85/esphome/base.yaml#L9-L16). Changing from arduino to esp-idf might not be possible with flashing over wifi but I'm not sure about that. This would let the ESP32 forward the pvvx messages to Home Assistant.

rajkundu, 2025-06-01:
> Thank you very much! I will try using `sensor.invalid` for now and proceed from there.

I can confirm that this works - I finally set everything up today! Thank you so much, @JanM321, for this phenomenal repo. It has made it unbelievably easy to accomplish a dream that I have had since moving into my apartment, both on a hardware and software level. Had I known it was this easy, I would have done it months earlier. Thank you!!

JanM321, 2025-06-01:
Thanks to @leoshusar (PR #108) the temperature sensor is now optional, so if you don't use your own sensor you can comment it out or delete it in the YAML file.