JanM321/esphome-lg-controller #49: Feature requests Tiny PCB

rrooggiieerr, 2024-04-13:
Great ESPHome lib! I ordered the PCBs, installed the code and now got it working on my 2 HVAC units :-)

I think this one is for @florianbrede-ayet 

First thumbs up for making the PCB! I've got a couple of ideas/feature requests for the Tiny PBC.

I wanted to do some measurements on the PCB before adding my ESP to it, making sure the voltages were right. But I was missing test pads. Thus my feature request would be to add test pads on the rear or both sides of the PCB. These test pads should expose 12V, Ground, 5 Volts, and maybe even RX, TX and LIN. 12V, Ground and LIN was easy enough to probe from the JST connector soldering points, so those would have lower priority. I was mainly looking for a 5 volt point

Second, would it be possible to add a second LIN channel? I think there is enough space on the board to fit the extra LIN bus converter especialy since you can use some room of the antenna area. Although fitting a second JST connectors might be tight. To power the board you could use the primary LIN bus so no extra components would be needed for the power supply. I think only an extra LIN bus converter and JST tag would need to be added to the BOM. My use case for the second LIN channel would be to control my second HVAC from the same ESP since they are installed on the same wall back to back so a short cable would be needed to wire them to the same PCB.

Maybe the voltage and ground pins could be added to the silkscreen because those and the RX/TX are basically the only ones one needs to solder. Adding the voltage pin to the silkscreen would maybe eliminate the need for test pads, but why not both.

Also the revision number should be updated on the silkscreen.

And then my last request would be to use slightly larger components. I first tried to hand solder the parts, bit naïve, most components worked out, even the tiny diodes. The voltage converter chip was obviously totally impossible. I understand that you designed to board to be assembled by the JLCPCB, but having a bit bigger components would make it easier to rework or tinker with the board.

Also, your PCB design doesn't need a 12 Volt LIN bus, everything above +/- 5.5 Volts should work right? I'm asking because I've got a couple of AXA Remote window openers which also use a LIN bus converter for communication and they run on 8 volts. So I want to use your PCB to control these window openers too.

So to sum it up, I basically redesigned your whole PCB. You're welcome :-) Keep up the good work!

rrooggiieerr, 2024-04-13:
I just thought of something else...

Since it's not possible to control everything over the LIN bus, disabling the display light comes to mind, how about adding the circuitry required to add an IR sender. The `climate_ir_lg` component or other IR Remote component could then be used to control these settings.

Even when not controlling the HVAC with the IR, a HVAC is often positioned on a central location and thus makes it a great place to hide an IR transmitter to control other devices.

I think the right resistor and a connector header should be enough, the IR sender could then be on a cable placed in sight of the IR receiver of the HVAC.

JanM321, 2024-04-17:
It's *possible* to add a second LIN transceiver and/or an IR transmitter, but most people aren't going to use those features. We've tried to keep the PCBs as simple as possible: do one thing and do it well. This also makes the PCBs cheaper to assemble and is less error-prone.

martiinezz, 2024-09-12:
I'm wondering why does Tiny PCB duplicate ESP32 module 3.3V creation? Would be wiser to make 5V on the board and provide it to ESP32 module - the rest will happen as designed in the module itself. It would simplify PCB and make it less expensive.

florianbrede-ayet, 2024-09-12:
> I'm wondering why does Tiny PCB duplicate ESP32 module 3.3V creation? Would be wiser to make 5V on the board and provide it to ESP32 module - the rest will happen as designed in the module itself. It would simplify PCB and make it less expensive.

The ESP32 onboard 3.3V linear regulator has an efficiency of around 65% from 5V -> 3.3V and is usually the cheapest part currently sourceable in China.
The regulator on the PCB is a TI switching step down converter with an efficiency of 85%+ for 17V -> 3.3V and will easily offset its part costs in a few months.

lagomCat, 2025-06-07:
> Линейный регулятор ESP32 на плате 3,3 В имеет эффективность около 65% от 5 В -> 3,3 В и обычно является самой дешевой деталью, которую в настоящее время можно найти в Китае.
> Регулятор на печатной плате представляет собой импульсный понижающий преобразователь TI с эффективностью 85%+ для 17 В -> 3,3 В и легко окупит стоимость детали за несколько месяцев.

Hello! Maybe you can answer my question: why is a 1 kΩ pull-up resistor used on the board? When working with the wiring of the CN_REMO connector on the LG board, there is a voltage divider on the LIN bus formed by 1 kΩ and 12 kΩ (sometimes 4.7 kΩ). As a result, when receiving a packet from the air conditioner unit on the ESP32, the high logic level is about 12 volts, and the low logic level is around 11 volts. This does not correspond at all to the TJA1027 specification. If packet reception succeeds, it’s some kind of miracle.

In my opinion, there should be a resistor of at least 10 kΩ, as allowed by the TJA1027 datasheet for a short line. But even that would not ensure the required low logic voltage level.