JanM321/esphome-lg-controller #52: LG Controller as External Component

rrooggiieerr, 2024-04-18:
Since Custom Components are deprecated and will be removed from ESPHome in a future release I put some time in refactoring the library to a custom component.

https://esphome.io/custom/custom_component
https://esphome.io/components/external_components

Before I submit a pull request I'd like you to take a look at what I did:
https://github.com/rrooggiieerr/esphome-lg-controller

The following minimal yaml should give you a working HVAC module. 
```
external_components:
  - source:
      type: local
      path: components
    components: [lg_controller]

uart:
  tx_pin: GPIO25
  rx_pin: GPIO26
  baud_rate: 104

lg_controller:

climate:
  - platform: lg_controller
    name: Heatpump
```

The External Component is not dependant on aditional components, however they can be added if needed:
```
sensor:
  - platform: lg_controller
    name: Error Code
    entity_id: error_code
  - platform: lg_controller
    name: Pipe Temperature In
    entity_id: pipe_temperature_in
    entity_category: diagnostic
  - platform: lg_controller
    name: Pipe Temperature Mid
    entity_id: pipe_temperature_mid
    entity_category: diagnostic
  - platform: lg_controller
    name: Pipe Temperature Out
    entity_id: pipe_temperature_out
    entity_category: diagnostic

binary_sensor:
  - platform: lg_controller
    name: Defrost
    entity_id: defrost
  - platform: lg_controller
    name: Preheat
    entity_id: preheat
  - platform: lg_controller
    name: Outdoor Unit
    entity_id: outdoor

switch:
  - platform: lg_controller
    name: Air Purifier
    entity_id: air_purifier
  - platform: lg_controller
    name: Internal Thermistor
    entity_id: internal_thermistor

select:
  - platform: lg_controller
    name: Airflow 1 Up/Down
    entity_id: vane_position_1
  - platform: lg_controller
    name: Airflow 2 Up/Down
    entity_id: vane_position_2
  - platform: lg_controller
    name: Airflow 3 Up/Down
    entity_id: vane_position_3
  - platform: lg_controller
    name: Airflow 4 Up/Down
    entity_id: vane_position_4
  - platform: lg_controller
    name: Over Heating # Installer setting 15.
    entity_id: overheating

number:
  - platform: lg_controller
    entity_id: fan_speed_slow
    name: "Fan Speed Slow"
    icon: mdi:fan-chevron-down
    mode: box
  - platform: lg_controller
    entity_id: fan_speed_low
    name: "Fan Speed Low"
    icon: mdi:fan-speed-1
    mode: box
  - platform: lg_controller
    entity_id: fan_speed_medium
    name: "Fan Speed Medium"
    icon: mdi:fan-speed-2
    mode: box
  - platform: lg_controller
    entity_id: fan_speed_high
    name: "Fan Speed High"
    id: fan_speed_high
    icon: mdi:fan-speed-3
    mode: box
  - platform: lg_controller
    entity_id: sleep_timer
    name: "Sleep Timer"
    icon: mdi:timer-outline
    mode: box
```

JanM321, 2024-04-21:
Thanks for looking into this!

This should be done at some point, but I want to play with this a bit more myself before taking a large refactoring. I wonder if there's a more incremental way to move away from custom components but I haven't looked into it yet.

JanM321, 2024-04-27:
I rewrote the code as an external component in 942f2f61b7f4654a65bd2fbd1bd9f1cb27b0ecc5.

Thanks for reporting the deprecation issue. I'm glad I was able to fix this without having to reorganize the code too much, but at some point it probably makes sense to split up `lg-controller.h` a bit more.

closed 2024-04-27 by JanM321