JanM321/esphome-lg-controller #53: Vane control issue

sergeipr, 2024-04-19:
Hi, 

I just wanted to say thanks for all your hard work on this project. It's been a lifesaver.

I have successfully connected an esp controller to ARNU09GSBL2 wall unit. This is one of 4 units (3 wall + 1 ducted) connected to MultiV.
All seems to work good, except for vane control. When unit turns on the vane is at topmost position. I can turn the swing mode, but can't set the static angle. None of 4 available settings (although there is only one vane) changes anything.

Here is the C9 message `C9.84.EA.1F.80.32.00.84.03.00.00.00.DA`
It seems there are some bytes set that are 'unknown' in protocol.md

And here is a log for vane position change attempt:
```shell
[21:42:19][D][lg-controller:1305]: update
[21:42:19][D][lg-controller:905]: received A8.08.00.00.00.01.0C.14.00.00.00.00.84 (13)
[21:42:19][D][lg-controller:921]: verified send
[21:42:20][D][climate:011]: 'LG AC Office' - Setting
[21:42:20][D][climate:015]:   Mode: COOL
[21:42:20][D][climate:396]: 'LG AC Office' - Sending state:
[21:42:20][D][climate:399]:   Mode: COOL
[21:42:20][D][climate:404]:   Fan Mode: LOW
[21:42:20][D][climate:416]:   Swing Mode: OFF
[21:42:20][D][climate:419]:   Current Temperature: 25.50°C
[21:42:21][D][climate:425]:   Target Temperature: 27.50°C
[21:42:25][D][lg-controller:1305]: update
[21:42:26][D][lg-controller:814]: sending A8.03.00.00.00.01.0C.14.00.00.00.00.99 (13)
[21:42:26][W][component:237]: Component <unknown> took a long time for an operation (516 ms).
[21:42:26][W][component:238]: Components should block for at most 30 ms.
[21:42:30][D][esp32.preferences:114]: Saving 1 preferences to flash...
[21:42:30][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[21:42:30][D][sensor:094]: 'Uptime': Sending state 1316.33704 s with 0 decimals of accuracy
[21:42:31][D][lg-controller:1305]: update
[21:42:31][D][lg-controller:905]: received A8.03.00.00.00.01.0C.14.00.00.00.00.99 (13)
[21:42:31][D][lg-controller:921]: verified send
[21:42:37][D][lg-controller:1305]: update
[21:42:40][D][select:062]: 'Airflow 1 Up/Down' - Setting
[21:42:40][D][select:115]: 'Airflow 1 Up/Down' - Set selected option to: 5
[21:42:40][D][select:015]: 'Airflow 1 Up/Down': Sending state 5 (index 5)
[21:42:40][D][lg-controller:513]: Setting vane 1 position: 5
[21:42:43][D][lg-controller:1305]: update
[21:42:43][D][lg-controller:905]: received C8.03.00.00.00.01.0C.5F.00.00.00.00.62 (13)
[21:42:43][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[21:42:43][D][climate:396]: 'LG AC Office' - Sending state:
[21:42:43][D][climate:399]:   Mode: COOL
[21:42:43][D][climate:404]:   Fan Mode: LOW
[21:42:43][D][climate:416]:   Swing Mode: OFF
[21:42:43][D][climate:419]:   Current Temperature: 25.50°C
[21:42:43][D][climate:425]:   Target Temperature: 27.50°C
[21:42:43][D][lg-controller:905]: received C8.03.00.00.00.01.0C.5F.00.00.00.00.62 (13)
[21:42:43][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[21:42:43][D][climate:396]: 'LG AC Office' - Sending state:
[21:42:43][D][climate:399]:   Mode: COOL
[21:42:43][D][climate:404]:   Fan Mode: LOW
[21:42:43][D][climate:416]:   Swing Mode: OFF
[21:42:43][D][climate:419]:   Current Temperature: 25.50°C
[21:42:43][D][climate:425]:   Target Temperature: 27.50°C
[21:42:43][D][lg-controller:905]: received C8.03.00.00.00.01.0C.1F.00.00.00.00.A2 (13)
[21:42:43][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[21:42:43][D][climate:396]: 'LG AC Office' - Sending state:
[21:42:43][D][climate:399]:   Mode: COOL
[21:42:43][D][climate:404]:   Fan Mode: LOW
[21:42:43][D][climate:416]:   Swing Mode: OFF
[21:42:43][D][climate:419]:   Current Temperature: 25.50°C
[21:42:43][D][climate:425]:   Target Temperature: 27.50°C
[21:42:43][D][lg-controller:905]: received C8.03.00.00.00.01.0C.1F.00.00.00.00.A2 (13)
[21:42:43][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[21:42:43][D][climate:396]: 'LG AC Office' - Sending state:
[21:42:43][D][climate:399]:   Mode: COOL
[21:42:43][D][climate:404]:   Fan Mode: LOW
[21:42:43][D][climate:416]:   Swing Mode: OFF
[21:42:43][D][climate:419]:   Current Temperature: 25.50°C
[21:42:43][D][climate:425]:   Target Temperature: 27.50°C
[21:42:44][D][lg-controller:861]: sending AA.00.00.00.00.00.00.05.00.02.F1.21.96 (13)
[21:42:44][W][component:237]: Component <unknown> took a long time for an operation (714 ms).
[21:42:44][W][component:238]: Components should block for at most 30 ms.
[21:42:49][D][lg-controller:1305]: update
[21:42:49][D][lg-controller:905]: received AA.00.00.00.00.00.00.05.00.02.F1.21.96 (13)
[21:42:49][D][lg-controller:921]: verified send
[21:42:50][D][lg-controller:814]: sending A8.02.00.00.00.01.0C.14.00.00.00.00.9E (13)
[21:42:50][W][component:237]: Component <unknown> took a long time for an operation (521 ms).
[21:42:50][W][component:238]: Components should block for at most 30 ms.
[21:42:55][D][lg-controller:1305]: update
[21:42:55][D][lg-controller:905]: received A8.02.00.00.00.01.0C.14.00.00.00.00.9E (13)
[21:42:55][D][lg-controller:921]: verified send
[21:42:56][D][select:062]: 'Airflow 1 Up/Down' - Setting
[21:42:56][D][select:115]: 'Airflow 1 Up/Down' - Set selected option to: 1 (Up)
[21:42:56][D][select:015]: 'Airflow 1 Up/Down': Sending state 1 (Up) (index 1)
[21:42:56][D][lg-controller:513]: Setting vane 1 position: 1
[21:43:01][D][lg-controller:1305]: update
[21:43:02][D][lg-controller:861]: sending AA.00.00.00.00.00.00.01.00.02.F1.21.EA (13)
[21:43:02][W][component:237]: Component <unknown> took a long time for an operation (517 ms).
[21:43:02][W][component:238]: Components should block for at most 30 ms.
[21:43:07][D][lg-controller:1305]: update
[21:43:07][D][lg-controller:905]: received AA.00.00.00.00.00.00.01.00.02.F1.21.EA (13)
[21:43:07][D][lg-controller:921]: verified send
[21:43:11][D][api.connection:197]: ESPHome Logs 2024.4.0 (10.100.102.7) requested disconnected
```

Any ideas on how to fix this? Your help would be awesome.
Thanks again for everything!

JanM321, 2024-04-21:
What happens if you change the vane position using ThinQ or the wireless remote? What messages does the wired controller receive then?

> P.S. the red led on the esp is lit when connected to AC, is this ok?

In my case that only happens when connected to USB (or maybe any 5V power source) but not when connected to the AC. Which ESP32 board are you using?

JanM321, 2024-04-21:
> Here is the C9 message `C9.84.EA.1F.80.32.00.84.03.00.00.00.DA`

It looks like your unit doesn't support vertical vane control (low bit of byte 4, byte 4 is 0x80 for this unit). We don't check this currently in the code though. The code also assumes that if the two high bits of byte 5 are both zero we have 4 vanes, but maybe that's only the case if the "vane control" bit is set.

sergeipr, 2024-04-21:
> What happens if you change the vane position using ThinQ or the wireless remote? What messages does the wired controller receive then?

> It looks like your unit doesn't support vertical vane control (low bit of byte 4, byte 4 is 0x80 for this unit). We don't check this currently in the code though. The code also assumes that if the [...]

I don't have ThinQ dongle, but checked with wireless remote and the unit even doesn't beep when I push the button and no messages are received. So I think you are correct, no vertical vane control for this unit.
My only option is to enable swing and try to stop it at wanted angle. With wireless remote this works great, but with controller it has some delay before the command is received, thus the angle is random. 
Maybe there is a way to time the swing on/off commands to achieve some sort of vane control? But not sure if it possible on single wire

> > P.S. the red led on the esp is lit when connected to AC, is this ok?
> 
> In my case that only happens when connected to USB (or maybe any 5V power source) but not when connected to the AC. Which ESP32 board are you using?

I use [eps32 devkitc v4](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/hw-reference/esp32/get-started-devkitc.html#esp32-devkitc-v4-getting-started-guide). Checked [rrooggiieerr installation pictures](https://github.com/JanM321/esphome-lg-controller/discussions/51#discussion-6498998), and on last pic it looks like led is lit as well. Weird as the board docs state it should turn on when the USB or an external 5V power supply is connected to the board

sergeipr, 2024-04-21:
> I don't have ThinQ dongle, but checked with wireless remote and the unit even doesn't beep when I push the button and no messages are received. So I think you are correct, no vertical vane control [...]

Maybe that's why https://esp32.com/viewtopic.php?t=13504

JanM321, 2024-04-27:
@sergeipr I just made a change that should fix this (it should now detect 0 vanes instead of 4). Let me know if this works.

closed 2024-04-27 by JanM321

sergeipr, 2024-04-27:
Works like a charm! No vane entities exposed to HA. 
Can confirm successful installation on 2 models - **ARNU12GSBL2** and **ARNU09GSBL2**

Thank you for your hard work! 🙏🏼