JanM321/esphome-lg-controller #54: Would this work with a LG WH27S?

PvdGulik, 2024-04-29:
Hello, verry intresting project. I have a WH27S waterpumpboiler, which has a CN_REMO connector.
Do you think this project would work?

Thanks!

[image]

JanM321, 2024-04-29:
The hardware side is probably similar, but I expect the protocol to be different.

This looks relevant: https://github.com/cribskip/esp8266_lgawhp ?

PvdGulik, 2024-04-29:
I do have a HM093MR.U44. This works perfectly with https://github.com/paqpaqpaq/ThermaV_overshoot_logic or with the above one.
But now I 'm  looking for a LG WH27S.F5 which does not have a modbus connection, as the above one needs.
That's why a found yours. Could I try to use the CN_REMO connection without breaking my WaterPumpboiler?

Thanks!

JanM321, 2024-04-30:
> Could I try to use the CN_REMO connection without breaking my WaterPumpboiler?

I can't tell you this because I know nothing about this kind of device. Do you know if LG has official CN-REMO controllers for it? There's not a lot of information available.

If it's a 12V connection similar to the one for AC units it __probably__ won't break it, but I would be surprised if it uses the same protocol as the one implemented here. LG also used different protocols for AC units and ventilation products for example (see issue #33).

JanM321, 2024-05-01:
Closing because out of scope for this project if it uses a different protocol.

closed 2024-05-01 by JanM321