JanM321/esphome-lg-controller #57: LG H12AL (ASNW126MMS6)

DavidAntonin, 2024-05-23:
Hi, I have finally created my own PCB for the ESP32-S2 and have had partial success with my LG Prestige split air conditioner. I can switch modes, temperature, fan speed and set horizontal and vertical oscillation. I miss the ability to set a specific air direction - I can set 6 vertical positions and 5 horizontal from the remote. 
Occasionally I get invalid checksum for a reason unknown to me (e.g. once every 30 minutes).
I am sending the log below.

Since my unit does not have power metering, I would like to use the PZEM-004T . The problem is that it also communicates via UART. ESPHome allows to use several UARTs with their own id. In the YAML file it even says "id: ac_serial". However, in the source files this id is not used and so when trying to add another UART with a different id, the program cannot be compiled. Would it be possible to modify the source code to use id: ac_serial? Thank you

`[14:02:23][C][homeassistant.sensor:031]:   Entity ID: 'sensor.my_sensor'
[14:02:27][D][lg-controller:1351]: update
[14:02:27][E][lg-controller:1392]: discarding incomplete data 00.00.00.07.1E.00.00.00.00.58 (10)
[14:02:33][D][lg-controller:1351]: update
[14:02:33][E][lg-controller:743]: unknown fan mode, using Medium
[14:02:33][D][lg-controller:846]: sending A8.21.00.00.00.00.07.14.40.00.80.00.F1 (13)
[14:02:33][W][component:237]: Component lg_controller.climate took a long time for an operation (509 ms).
[14:02:33][W][component:238]: Components should block for at most 30 ms.
[14:02:39][D][lg-controller:1351]: update
[14:02:45][D][lg-controller:1351]: update
[14:02:51][D][lg-controller:1351]: update
[14:02:51][D][lg-controller:943]: received A8.21.00.00.00.00.07.14.40.00.80.00.F1 (13)
[14:02:51][D][lg-controller:959]: verified send
[14:02:51][D][lg-controller:943]: received C8.20.00.00.00.00.07.1E.00.00.40.00.18 (13)
[14:02:51][D][binary_sensor:036]: 'Defrost': Sending state OFF
[14:02:51][D][binary_sensor:036]: 'Preheat': Sending state OFF
[14:02:51][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[14:02:51][D][binary_sensor:036]: 'Outdoor Unit': Sending state OFF
[14:02:51][D][binary_sensor:036]: 'Auto Dry Active': Sending state OFF
[14:02:51][D][climate:396]: 'Klima LG Prestige' - Sending state:
[14:02:51][D][climate:399]:   Mode: OFF
[14:02:51][D][climate:416]:   Swing Mode: OFF
[14:02:51][D][climate:419]:   Current Temperature: 25.00°C
[14:02:51][D][climate:425]:   Target Temperature: 22.00°C
[14:02:51][D][switch:055]: 'Air Purifier': Sending state OFF
[14:02:51][D][climate:396]: 'Klima LG Prestige' - Sending state:
[14:02:51][D][climate:399]:   Mode: OFF
[14:02:51][D][climate:404]:   Fan Mode: MEDIUM
[14:02:51][D][climate:416]:   Swing Mode: OFF
[14:02:51][D][climate:419]:   Current Temperature: 25.00°C
[14:02:51][D][climate:425]:   Target Temperature: 22.00°C
[14:02:51][D][lg-controller:943]: received C8.20.00.00.00.00.07.1E.00.00.40.00.18 (13)
[14:02:51][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[14:02:51][D][lg-controller:1074]: ignoring because pending change
[14:02:51][D][lg-controller:943]: received C8.20.00.00.00.00.07.1E.00.00.40.00.10 (13)
[14:02:51][E][lg-controller:953]: invalid checksum C8.20.00.00.00.00.07.1E.00.00.40.00.10 (13)
[14:02:51][D][lg-controller:943]: received C9.C4.EA.1F.80.30.00.C0.02.00.00.00.5D (13)
[14:02:51][D][lg-controller:943]: received C9.C4.EA.1F.80.30.00.C0.02.00.00.00.5D (13)
[14:02:51][D][lg-controller:943]: received CA.E0.00.00.00.00.00.00.00.02.F1.21.EB (13)
[14:02:51][D][select:015]: 'Airflow 1 Up/Down': Sending state 0 (Default) (index 0)
[14:02:51][D][select:015]: 'Airflow 2 Up/Down': Sending state 0 (Default) (index 0)
[14:02:51][D][select:015]: 'Airflow 3 Up/Down': Sending state 0 (Default) (index 0)
[14:02:51][D][select:015]: 'Airflow 4 Up/Down': Sending state 0 (Default) (index 0)
[14:02:51][D][switch:055]: 'Auto Dry': Sending state OFF
[14:02:51][D][number:012]: 'Fan Speed Slow': Sending state 0.000000
[14:02:51][D][number:012]: 'Fan Speed Low': Sending state 0.000000
[14:02:51][D][number:012]: 'Fan Speed Medium': Sending state 0.000000
[14:02:51][D][number:012]: 'Fan Speed High': Sending state 0.000000
[14:02:51][D][lg-controller:943]: received CA.E0.00.00.00.00.00.00.00.02.F1.21.EB (13)
[14:02:51][D][select:015]: 'Airflow 1 Up/Down': Sending state 0 (Default) (index 0)
[14:02:51][D][select:015]: 'Airflow 2 Up/Down': Sending state 0 (Default) (index 0)
[14:02:51][D][select:015]: 'Airflow 3 Up/Down': Sending state 0 (Default) (index 0)
[14:02:51][D][select:015]: 'Airflow 4 Up/Down': Sending state 0 (Default) (index 0)
[14:02:51][D][number:012]: 'Fan Speed Slow': Sending state 0.000000
[14:02:51][D][number:012]: 'Fan Speed Low': Sending state 0.000000
[14:02:51][D][number:012]: 'Fan Speed Medium': Sending state 0.000000
[14:02:51][D][number:012]: 'Fan Speed High': Sending state 0.000000
[14:02:51][D][lg-controller:943]: received CB.00.00.FF.FF.75.00.00.00.00.00.00.6B (13)
[14:02:51][D][select:015]: 'Over Heating': Sending state 0 (Default) (index 0)
[14:02:51][D][sensor:094]: 'Pipe Temperature Mid': Sending state 24.00000 °C with 0 decimals of accuracy
[14:02:51][D][lg-controller:943]: received CB.00.00.FF.FF.75.00.00.00.00.00.00.6B (13)
[14:02:51][D][select:015]: 'Over Heating': Sending state 0 (Default) (index 0)
[14:02:51][D][sensor:094]: 'Pipe Temperature Mid': Sending state 24.00000 °C with 0 decimals of accuracy
[14:02:51][W][component:237]: Component lg_controller.climate took a long time for an operation (147 ms).
[14:02:51][W][component:238]: Components should block for at most 30 ms.
[14:02:51][D][sensor:094]: 'Uptime': Sending state 50.55000 s with 0 decimals of accuracy
[14:02:57][D][lg-controller:1351]: update
[14:02:57][D][lg-controller:899]: sending AA.E0.00.00.00.00.00.00.00.02.F1.21.CB (13)
[14:02:57][W][component:237]: Component lg_controller.climate took a long time for an operation (509 ms).
[14:02:57][W][component:238]: Components should block for at most 30 ms.
[14:03:03][D][lg-controller:1351]: update
[14:03:03][D][lg-controller:943]: received AA.E0.00.00.00.00.00.00.00.02.F1.21.CB (13)
[14:03:03][D][lg-controller:959]: verified send
[14:03:03][D][lg-controller:934]: sending AB.00.00.FF.FF.75.00.00.00.00.00.00.4B (13)
[14:03:03][W][component:237]: Component lg_controller.climate took a long time for an operation (515 ms).
[14:03:03][W][component:238]: Components should block for at most 30 ms.
[14:03:09][D][lg-controller:1351]: update
[14:03:09][D][lg-controller:943]: received AB.00.00.FF.FF.75.00.00.00.00.00.00.4B (13)
[14:03:09][D][lg-controller:959]: verified send
[14:03:09][D][lg-controller:943]: received C8.20.00.00.00.00.07.1E.00.00.00.00.58 (13)
[14:03:09][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[14:03:09][D][lg-controller:1074]: ignoring because pending change
[14:03:09][D][lg-controller:846]: sending A8.21.00.00.00.00.07.14.00.00.40.00.71 (13)
[14:03:09][W][component:237]: Component lg_controller.climate took a long time for an operation (525 ms).
[14:03:09][W][component:238]: Components should block for at most 30 ms.
[14:03:10][D][sensor:094]: 'WiFi Signal': Sending state -30.00000 dBm with 0 decimals of accuracy
[14:03:15][D][lg-controller:1351]: update
[14:03:15][D][lg-controller:943]: received A8.21.00.00.00.00.07.14.00.00.40.00.71 (13)
[14:03:15][D][lg-controller:959]: verified send
[14:03:21][D][lg-controller:1351]: update
[14:03:27][D][lg-controller:1351]: update
[14:03:33][D][lg-controller:1351]: update
[14:03:33][D][lg-controller:846]: sending A8.20.00.00.00.00.07.14.00.00.40.00.76 (13)
[14:03:33][W][component:237]: Component lg_controller.climate took a long time for an operation (509 ms).
[14:03:33][W][component:238]: Components should block for at most 30 ms.
`

JanM321, 2024-05-23:
> I miss the ability to set a specific air direction - I can set 6 vertical positions and 5 horizontal from the remote.

The horizontal angle can't be set because the protocol and LG wired controllers don't support it, unfortunately.

There's protocol support for setting the vertical direction, but I see in your log that the unit reports that it doesn't have this feature. Maybe LG chose not to implement this for some reason. They've done this with other features too, for example low-medium and medium-high fan speeds are available in the protocol but most units ignore these values even though they support these modes with the wireless remote.

>  Occasionally I get invalid checksum for a reason unknown to me (e.g. once every 30 minutes).

Maybe a hardware issue or something with the wiring.

> Would it be possible to modify the source code to use id: ac_serial?

Does adding `uart_id: ac_serial` to the climate's YAML code work?

DavidAntonin, 2024-05-24:
Thank you for the airflow information. Would it be possible to try to send a setup command for vertical flow - even if the unit doesn't publish it? With the fan it is also strange, using the remote I can set 5 speeds + breeze. 
There was an idea in a previous post, to add an IR transmitter, to send commands that are not supported by the wired controller. 

As for the UART, it is already listed in base.yaml:
```
uart:
  - id: ac_serial
    tx_pin: GPIO25
    rx_pin: GPIO26 # Keep in sync with RxPin in lg-controller.h
    baud_rate: 104
```
However, the actual component does not use id: ac_serial, it refers to UART in general. 
Therefore, if I change the code to:
```
uart:
  - id: ac_serial
    tx_pin: GPIO25
    rx_pin: GPIO26 # Keep in sync with RxPin in lg-controller.h
    baud_rate: 104
  - id: pzem_serial
    tx_pin: GPIO5
    rx_pin: GPIO6
    baud_rate: 9600
```
an error occurs during compilation.
I don't know if it is possible to specify which UART (id) to use in the custom component. This is how it is described in the ESPHome UART documentation 
```
  - uart.write:
      id: my_second_uart
      data: 'data to send'

```

JanM321, 2024-05-24:
@DavidAntonin What I was suggesting is to add `uart_id: ac_serial` here: https://github.com/JanM321/esphome-lg-controller/blob/main/esphome/base.yaml#L32

Does that work?

DavidAntonin, 2024-05-25:
@JanM321 Thank you, adding uart_id: ac_serial works and now I have power metering working. Now I will try add IR transmiter to get more functions of my AC

JanM321, 2024-06-01:
Great! I added `uart_id: ac_serial` to base.yaml. It's also nicer to be more explicit about this.

commit 819ba068 (2024-06-01): Add explicit uart_id to support configurations with multiple UARTs (fixes #57)
closed 2024-06-01 by JanM321