JanM321/esphome-lg-controller #60: Current Temperature missing

Arunas-g, 2024-07-05:
Hello. I have a problem getting "Current Temperature" from my LG TC07GQR air conditioner. I have another LG air conditioner, model PC12SQ, it works perfectly. In the logs, instead of the temperature, it says “nan°C”

`[15:01:01][D][climate:419]:   Current Temperature: nan°C`
`[15:01:01][D][climate:425]:   Target Temperature: 26.00°C`

At the same time, the air conditioner itself works normally, reacts to changes in “Target Temperature” and automatically changes the air flow rate. Air conditioner control is organized using "hardware-tiny". Please help me understand the reason for the incorrect display of the target temperature.

I attach logs and screenshots
[logs_Current Temperature nan C.txt](https://github.com/user-attachments/files/16108666/logs_Current.Temperature.nan.C.txt)
[image]

JanM321, 2024-07-05:
It's using `sensor.ac_lg_linas` as room temperature sensor. Is that a sensor you have available in Home Assistant? Can you show information about that one?

JanM321, 2024-07-05:
Basically your options are:
1. Set the "Internal Thermistor" setting to On to use the temperature value measured by the AC.
2. Make sure `sensor.ac_lg_linas` is a valid temperature sensor in Home Assistant because it will be *read* by the controller.

Arunas-g, 2024-07-05:
I apologize for the trouble :) I turned on "Internal Thermistor" and after that "Current Temperature" is displayed correctly in the logs.
Thank you!

[image]

[image]

JanM321, 2024-07-05:
No problem. I’m glad that works :)

closed 2024-07-05 by JanM321