JanM321/esphome-lg-controller #62: LW1017ERSM1 Compatibility?

royaltongue, 2024-07-14:
I saw in the `ReadMe` that anything compatible with the PREMTA200 or PREMTB100 should be compatible with this repo, but I can't find anywhere a list of devices that are compatible with those controllers.

If it helps, this A/C uses the [EBR85871801](https://lgparts.com/products/ebr85871801) RF board.

JanM321, 2024-07-14:
Unfortunately there isn’t a complete list available as far as I know. What’s your indoor unit model?

JanM321, 2024-07-14:
Nevermind I see it’s in the title, LW1017ERSM1

That’s a window unit and these are likely pretty different from the split units. You could check if there’s a CN-REMO connector on the pcb.

royaltongue, 2024-07-14:
Ah I didn't realize this was meant for the split units, my mistake.

[Here's a picture of the main PCB](https://imgur.com/a/wnweLQO). Doesn't look like there's a CN-REMO connector. 

Just in case, I won't close the issue, but feel free to close it if you took a look and come to the same conclusion.

closed 2024-11-17 by JanM321