JanM321/esphome-lg-controller #65: Issue with lg_lvn361hv4 hysteresis  and unable to set settings

mogorman, 2024-08-01:
Hi I have a lg central air handler heat pump setup.  My esphome-lg-controller is working in that it turns on and enables compressor, gets pipe temp.

But I can't seem to change the fan speed like I could with the builtin controller, I also am having problems with getting the outdoor unit activated. I have to set the target_temp more than 10 degrees colder than the actual temperature in my home for it to activate.  I thought this was just a problem with the hysteresis so I made a template for the temperature sensor 
```yaml
        value_template: >
          {% set temperature = state_attr('climate.house_thermostat', 'current_temperature') %}
          {% set setpoint = states.climate.first_floor_heatpump_first_floor_heatpump.attributes.temperature|float %}
          {% if (not is_state('climate.first_floor_heatpump_first_floor_heatpump', 'cool')) %}
            {{ temperature | int}}
          {% else %}
            {% if setpoint  < temperature -1 %}
              {{ (temperature + 12) | int }}
            {% else %}
              {{ temperature | int}}
            {% endif %}
          {% endif%}
```

but even with this it didn't activate I still had to set the temperature to 60f for it to activate even when it thought temperature was in 90s.  So I am not sure exactly what is going on. Is there any debug information I could provide to hopefully solve some of this?

JanM321, 2024-08-02:
What's the official LG controller for your unit? Can you pastebin the output from `esphome logs your-yaml-file`, after running it for a few minutes?

It's possible LG uses a different protocol for these air handler devices.

mogorman, 2024-08-02:
The remote I use is [PREMTC00U](https://lghvac.com/residential-light-commercial/product-type/model-details/?productTypeId=a2x44000003XR2A&modelId=01t4400000Ari4c&modelNumber=PREMTC00U&iscommercial=false&class=Controls)
The unit is [lvn361hv4](https://www.lg.com/ca_en/support/product-support/cs-LVN361HV4.ENWBEUS/)
the outdoor unit is [luu350hhv](https://www.lg.com/ca_en/support/product-support/cs-LUU360HHV.EWGBEUS/)

[logs_first_floor_heatpump_logs.txt](https://github.com/user-attachments/files/16473075/logs_first_floor_heatpump_logs.txt)

some logs.  I set fan speeds with no change, i also set some temperature points in logs, as well as finally setting it at a very low setpoint which finally caused the out door unit to fire.
[logs_first_floor_heatpump_logs.txt](https://github.com/user-attachments/files/16473108/logs_first_floor_heatpump_logs.txt)

thank you for looking into this.

JanM321, 2024-08-03:
It shows that your "fan mode" is set to Auto all the time. Try changing that setting, it's part of the climate entity. It should have an option for Low / Medium / High / Auto.

The "Fan Speed High" and "Fan Speed Low" values are advanced settings that only apply if you set the fan mode to "low" or "high" first.

JanM321, 2024-08-03:
The temperature thing is weird because what I'm seeing in the log all makes sense. It's sending a current temperature value that's a lot higher than the target value.

Does it work better if you set the "Internal Thermistor" setting to On?

mogorman, 2024-08-03:
i just tried playing with fan settings. and still see no change in fan speed, and can confirm that my current meter shows same 316 w draw when its just running this fan and no compressor
[logs_first_floor_heatpump_logs(1).txt](https://github.com/user-attachments/files/16480776/logs_first_floor_heatpump_logs.1.txt)

JanM321, 2024-08-03:
Something I just thought of to check: the unit is sending its own temperature value in the `CC` message and that's only 20-21C (68-69F). So maybe that's the "current temperature" value it's using internally? That might explain the behavior you're seeing.

The esphome controller currently doesn't send this type of message to the AC, because it hasn't been necessary for other units. I wonder if your unit requires this.

If you want to test this, you could add the following code before [this line](https://github.com/JanM321/esphome-lg-controller/blob/9964e0ad8316ac971dcfcd284e475792635b8043/esphome/components/lg_controller/lg-controller.h#L868) in lg-controller.h
```cpp
uint8_t buf[MsgLen] = {};
buf[0] = 0xAC;
buf[6] = uint8_t(target);
if (send_buf_[1] & 0x1) {
    buf[7] = 0x80; // settings changed
}
if (send_buf_[5] & 0x1) {
    buf[8] = 0x50; // target + 0.5C
}
buf[9] = uint8_t(temp * 2);
buf[12] = calc_checksum(buf);
ESP_LOGD(TAG, "sending %s", format_hex_pretty(buf, MsgLen).c_str());
UARTDevice::write_array(buf, MsgLen);
```

mogorman, 2024-08-03:
setting it to internal thermistor, did seem to have it activate close to what I would expect, i set it to 72 and the internal thermistor is at 75 and it kicked on.

mogorman, 2024-08-03:
i think this code change caused pip temperature in to change. it was reading 9c right before code deployed ,and changed to -5c after , also the pipe out temp went from 10c to 17c as well.
[logs_first_floor_heatpump_run(1).txt](https://github.com/user-attachments/files/16480911/logs_first_floor_heatpump_run.1.txt)

i changed the sending line to MOG sending so it was easier to spot in logs but otherwise no change

mogorman, 2024-08-03:
i think this fixed different bug with pipe temperature i thought i have had. the in and out temp were almost always the same which seemed awful, since applying this patch they seem to have diverged 10c or so 
[image]

mogorman, 2024-08-04:
that def fixed my setpoint issue today, was hard to tell as so hot only got to temp once today

JanM321, 2024-08-04:
So with that code change it works as expected with Internal Thermistor set to Off?

mogorman, 2024-08-04:
Correct. Solved issue with temperature. Any ideas on fan speed settings?

JanM321, 2024-08-04:
What if you change it from Cooling to Fan Only and reset the fan speed settings by setting them all to 0. Do you then see any difference between Low and High?

Do you remember which options you had with the LG controller? It’s weird this doesn’t work.

mogorman, 2024-08-05:
fan only mode I measured the following. i felt and saw no change with these settings.
with 0 for all settings
```
auto: 318w
low: 315w
med: 316w
high: 316w 
```
so i assume same within noise values
[logs_first_floor_heatpump_logs(2).txt](https://github.com/user-attachments/files/16490566/logs_first_floor_heatpump_logs.2.txt)

now trying with 240 245 and 250 as the values for the fan speeds, capture shows me setting these values as well.
```
auto: 315w
low: 315w
med: 316w
high: 314w
```
[logs_first_floor_heatpump_logs(3).txt](https://github.com/user-attachments/files/16490597/logs_first_floor_heatpump_logs.3.txt)

mogorman, 2024-08-05:
I was able to set fan speeds with the factory controller in the past

JanM321, 2024-08-05:
Your unit is likely different from the AC units that others are using with this controller. I suspect it uses the newer protocol messages more that my controller doesn't send or process.

One option would be to connect both controllers at the same time and put the ESPHome controller in 'slave' mode, so you can see what's being sent by the LG controller if you change settings.

mogorman, 2024-08-05:
is that just setting 
  is_slave_controller: "true"

and connecting the remote to same bus line?

JanM321, 2024-08-05:
> is that just setting is_slave_controller: "true"
> 
> and connecting the remote to same bus line?

Yes.