JanM321/esphome-lg-controller #67: Identifying  the connector ?

yiannosi, 2024-08-25:
Hi

I am thinking  to try this in my LG indoor unit. Can anyone help  to identify the right connector in the unit's PCB ?
Any pictures?

sergeipr, 2024-08-26:
> Hi
> 
> I am thinking to try this in my LG indoor unit. Can anyone help to identify the right connector in the unit's PCB ? Any pictures?

Hi, 
It's dark green connector, 
Check this [discussion](https://github.com/JanM321/esphome-lg-controller/discussions/51#discussion-6498998) pictures

yiannosi, 2024-08-26:
Thanks for your reply.

It appears that there is no green connector on the PCB.. see pic below: 
[image]
Below is the AC name plate and model 

[image]

JanM321, 2024-08-26:
What's the label next to the 3-pin white connector on the left hand side? It doesn't say CN-REMO anywhere?