JanM321/esphome-lg-controller #68: Cable connection color confirmation

Hadatko, 2024-08-27:
Hi, the cable i bought based on requirements and connector i should connect to looks differ in colors.
[image]
I wanted confirmation how to wire to the board. 
Like: My red cable is gnd, My black wire is LIN and my yellow wire is 12V.

Hadatko, 2024-08-27:
Yeah i made is as i wrote and it is working.

closed 2024-08-27 by Hadatko

apppie123, 2025-04-20:
The cable you used in the picture is a JST-XH?
I'm a bit confused.

- In the LG manual of the PZCWRCG3 -which I believe is the extension cable for the wired remote display- I read that it is a MOLEX cable(typenumber not specified).
- The connector on the PCB inside the indoor unit looks like a JST-XH connector but with locking feature.
- I read that people had issues using the recommended cable from adafruit(Adafruit 4873) to the green extension cable in your picture, but that the same adafruit cable does fit into the connector on the PCB.

JanM321, 2025-04-21:
@apppie123 This isn't documented very well by LG unfortunately. I use the LG cables (PZCWRC1 I think?) and the Adafruit 4873 works well with those.

When I started this project I didn't know how often I'd need to access the controller so using these long LG cables was the simplest option. JST-XH was the closest connector I could find at the time, but it's not an exact match.

joeblack2k, 2025-06-13:
> Yeah i made it as i wrote and it is working.

in the end, did you match the colors ? so "Red,Yellow,Black" ?

joeblack2k, 2025-06-13:
I'm not very experienced with electronics, so I'm trying to understand it correctly. On the air conditioner side, red = 12V, black = ground, and yellow = data/line, right?
If I follow that same order when connecting to the board, everything should work fine, correct?

On the ESP side, it's:

Pin 1 = GND (black)

Pin 2 = line bus (yellow)

Pin 3 = 12V (red)

So I just need to make sure the wires from the air conditioner are connected in that order to the ESP32 using Dupont cables.

> I think, for my own good feel i changed order of the colored cables in connector. So i matched colors and purpose at the end. On other side of cable i didn't have connector just cables so i soldered them to the connector on board with esp in right order.
> 
> > > Yeah i made it as i wrote and it is working.
> > 
> > 
> > in the end, did you match the colors ? so "Red,Yellow,Black" ?