JanM321/esphome-lg-controller #76: Comparability with PQCSZ250S0

Shahar288, 2024-10-09:
I have an LG PQCSZ250S0 controller connected to 10 ducted units which I don’t know their model, and wanted to ask if you think it would work with this ESP32 setup (I know you can’t tell me for sure, but a ruff estimate will go a long way).
I don’t know if it’s relevant, but I have an old CoolMaster (the white one without the lan port) connected to the LG PQCSZ250S0, I will disconnect it if needed, but if it’s possible to have them both it will be desirable. I also saw you recommended someone to use this ESP32 setup: https://github.com/Flameeyes/esphome-lg-pqrcuds0. By the way, my goal is to connect HA to the ac, so if there’s a way to use the CoolMaster to connect to HA it would be great (I can’t use the integration as it requires ip, but my CoolMaster doesn’t have lan and communicates with Telnet  through serial rs232 (I have a dongle for usb a, but don’t know how to read it in HA)) sorry for deviating from the subject, but do you think it will work, or know of something that might will?
Thank you either way

JanM321, 2024-10-12:
Is the PQCSZ250S0 connected to all of the indoor units, or to the outdoor unit? I skimmed the manual and it does have a master/slave setting, so if that's for the controller maybe you can connect both the ESPHome controller and this one, setting one to 'master' and the other to 'slave'.

The ESPHome controller must be connected to the indoor unit and currently only supports a single indoor unit per controller though.

Shahar288, 2024-10-13:
Thanks!
It won’t be practical for me to to connect each indoor unit with an esp32 controller, so I’ll need to program a custom integration to talk with the PQCSZ250S0. Thanks again for your help and for creating this amazing tool.

closed 2024-11-17 by JanM321