JanM321/esphome-lg-controller #78: Turn display off

evertide, 2024-10-09:
Feature request?

First of all great work on the project!. The biggest gain is no sound with changes! !

the current remote control (IR) can switch off the display whilst staying powered on.
This is great during the night.

Can this be done with the pcb as well?

regards

JanM321, 2024-10-12:
Unfortunately this isn't possible, at least with my indoor units. Newer controllers have some settings for this in newer protocol messages but I haven't seen them actually being used.

You could test this by changing the display light setting with the IR remote and checking the esphome log output.

The LG ThinQ integration for Home Assistant has this setting. The Wifi module uses a very different protocol with a different set of features.

evertide, 2024-10-14:
This is the log when switching the light off
[19:26:01][D][lg-controller:1354]: update
[19:26:07][D][lg-controller:1354]: update
[19:26:07][D][lg-controller:946]: received C8.43.00.00.00.04.03.55.00.00.00.00.32 (13)
[19:26:07][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[19:26:07][D][climate:396]: 'Mancave 2 Airco' - Sending state:
[19:26:07][D][climate:399]:   Mode: COOL
[19:26:07][D][climate:404]:   Fan Mode: HIGH
[19:26:07][D][climate:416]:   Swing Mode: OFF
[19:26:07][D][climate:419]:   Current Temperature: 20.50°C
[19:26:07][D][climate:425]:   Target Temperature: 18.00°C
[19:26:07][D][climate:396]: 'Mancave 2 Airco' - Sending state:
[19:26:07][D][climate:399]:   Mode: COOL
[19:26:07][D][climate:404]:   Fan Mode: HIGH
[19:26:07][D][climate:416]:   Swing Mode: OFF
[19:26:07][D][climate:419]:   Current Temperature: 20.50°C
[19:26:07][D][climate:425]:   Target Temperature: 18.00°C
[19:26:07][D][lg-controller:946]: received C8.43.00.00.00.04.03.55.00.00.00.00.32 (13)
[19:26:07][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[19:26:07][D][climate:396]: 'Mancave 2 Airco' - Sending state:
[19:26:07][D][climate:399]:   Mode: COOL
[19:26:07][D][climate:404]:   Fan Mode: HIGH
[19:26:07][D][climate:416]:   Swing Mode: OFF
[19:26:07][D][climate:419]:   Current Temperature: 20.50°C
[19:26:07][D][climate:425]:   Target Temperature: 18.00°C

This is when switching it on again
[19:25:31][D][lg-controller:1354]: update
[19:25:31][D][lg-controller:946]: received A8.42.00.00.00.04.03.54.00.00.00.00.10 (13)
[19:25:31][D][lg-controller:962]: verified send
[19:25:37][D][lg-controller:1354]: update
[19:25:37][D][lg-controller:946]: received C8.42.00.00.00.04.03.56.00.00.00.00.32 (13)
[19:25:37][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[19:25:37][D][climate:396]: 'Mancave 2 Airco' - Sending state:
[19:25:37][D][climate:399]:   Mode: COOL
[19:25:37][D][climate:404]:   Fan Mode: HIGH
[19:25:37][D][climate:416]:   Swing Mode: OFF
[19:25:37][D][climate:419]:   Current Temperature: 21.00°C
[19:25:37][D][climate:425]:   Target Temperature: 18.00°C
[19:25:37][D][climate:396]: 'Mancave 2 Airco' - Sending state:
[19:25:37][D][climate:399]:   Mode: COOL
[19:25:37][D][climate:404]:   Fan Mode: HIGH
[19:25:37][D][climate:416]:   Swing Mode: OFF
[19:25:37][D][climate:419]:   Current Temperature: 21.00°C
[19:25:37][D][climate:425]:   Target Temperature: 18.00°C
[19:25:37][D][lg-controller:946]: received C8.42.00.00.00.04.03.56.00.00.00.00.32 (13)
[19:25:37][D][sensor:094]: 'Error Code': Sending state 0.00000  with 0 decimals of accuracy
[19:25:37][D][climate:396]: 'Mancave 2 Airco' - Sending state:
[19:25:37][D][climate:399]:   Mode: COOL
[19:25:37][D][climate:404]:   Fan Mode: HIGH
[19:25:37][D][climate:416]:   Swing Mode: OFF
[19:25:37][D][climate:419]:   Current Temperature: 21.00°C
[19:25:37][D][climate:425]:   Target Temperature: 18.00°C

JanM321, 2024-10-15:
Unfortunately it's not sending any settings for the display light, just the regular status messages. I see the same thing with my units.

I really wish LG had designed this to at least offer the same features as the IR remote.

evertide, 2024-10-15:
Thanks, Lets close the issue then. ( But keep it on the wish list with future firmware updates ;-) )

closed 2024-10-15 by evertide