JanM321/esphome-lg-controller #79: Invalid checksum

mizrachiran, 2024-10-14:
Hello and thank you for this great project,

I'm using esp32 with the tiny hardware (ordered from JLCPCB).
Wiring and setup was very easy and straight forward but i couldn't bring it to the finish line. I get checksum error, below are my logs.

Would appreciate your support.

[19:55:01][E][lg-controller:956]: invalid checksum 00.71.00.25.06.00.A2.98.00.24.00.00.00 (13)
[19:55:01][D][lg-controller:946]: received 71.00.25.06.00.A2.98.00.24.00.00.00.71 (13)
[19:55:01][E][lg-controller:956]: invalid checksum 71.00.25.06.00.A2.98.00.24.00.00.00.71 (13)
[19:55:07][D][lg-controller:1354]: update
[19:55:07][D][lg-controller:946]: received 00.25.06.00.A2.98.00.24.00.00.00.71.00 (13)
[19:55:07][E][lg-controller:956]: invalid checksum 00.25.06.00.A2.98.00.24.00.00.00.71.00 (13)
[19:55:07][D][lg-controller:946]: received 25.06.00.A2.98.00.24.00.00.00.71.00.25 (13)
[19:55:07][E][lg-controller:956]: invalid checksum 25.06.00.A2.98.00.24.00.00.00.71.00.25 (13)
[19:55:07][D][lg-controller:946]: received 06.00.A2.98.00.24.00.00.00.71.00.25.06 (13)
[19:55:07][E][lg-controller:956]: invalid checksum 06.00.A2.98.00.24.00.00.00.71.00.25.06 (13)
[19:55:07][W][component:237]: Component lg_controller.climate took a long time for an operation (51 ms).
[19:55:07][W][component:238]: Components should block for at most 30 ms.
[19:55:13][D][lg-controller:1354]: update
[19:55:13][D][lg-controller:946]: received 00.A2.98.00.24.00.00.00.71.00.25.06.00 (13)
[19:55:13][E][lg-controller:956]: invalid checksum 00.A2.98.00.24.00.00.00.71.00.25.06.00 (13)
[19:55:13][D][lg-controller:946]: received A2.98.00.24.00.00.00.71.00.25.06.00.A2 (13)
[19:55:13][E][lg-controller:956]: invalid checksum A2.98.00.24.00.00.00.71.00.25.06.00.A2 (13)
[19:55:13][D][lg-controller:946]: received 98.00.24.00.00.00.71.00.25.06.00.A2.98 (13)
[19:55:13][E][lg-controller:956]: invalid checksum 98.00.24.00.00.00.71.00.25.06.00.A2.98 (13)

mizrachiran, 2024-10-14:
made another try, disconnected the LG wall controller and set slave to false.
Now logs looks promising but the AC doesn't response to any state change. all the sensors in HA are unavailable.
I hope this helps debugging this issue.

[20:09:25][D][lg-controller:1354]: update
[20:09:26][E][lg-controller:770]: unknown swing mode
[20:09:26][D][lg-controller:849]: sending A8.03.00.00.00.01.12.22.40.00.80.00.F5 (13)
[20:09:26][W][component:237]: Component lg_controller.climate took a long time for an operation (515 ms).
[20:09:26][W][component:238]: Components should block for at most 30 ms.
[20:09:31][D][lg-controller:1354]: update
[20:09:31][D][lg-controller:946]: received A8.03.00.00.00.01.12.22.40.00.80.00.F5 (13)
[20:09:31][D][lg-controller:962]: verified send
[20:09:32][E][lg-controller:872]: Unexpected missing previous CA/AA message
[20:09:32][W][component:237]: Component lg_controller.climate took a long time for an operation (514 ms).
[20:09:32][W][component:238]: Components should block for at most 30 ms.
[20:09:32][D][esp32.preferences:114]: Saving 1 preferences to flash...
[20:09:32][D][esp32.preferences:143]: Saving 1 preferences to flash: 0 cached, 1 written, 0 failed
[20:09:35][D][sensor:094]: 'Uptime': Sending state 166.67999 s with 0 decimals of accuracy
[20:09:37][D][lg-controller:1354]: update
[20:09:43][D][lg-controller:1354]: update
[20:09:49][D][lg-controller:1354]: update
[20:09:50][E][lg-controller:770]: unknown swing mode
[20:09:50][D][lg-controller:849]: sending A8.02.00.00.00.01.12.22.40.00.80.00.CA (13)
[20:09:50][W][component:237]: Component lg_controller.climate took a long time for an operation (514 ms).
[20:09:50][W][component:238]: Components should block for at most 30 ms.
[20:09:55][D][lg-controller:1354]: update
[20:09:55][D][lg-controller:946]: received A8.02.00.00.00.01.12.22.40.00.80.00.CA (13)
[20:09:55][D][lg-controller:962]: verified send
[20:10:01][D][lg-controller:1354]: update
[20:10:07][D][lg-controller:1354]: update
[20:10:13][D][lg-controller:1354]: update
[20:10:14][E][lg-controller:770]: unknown swing mode
[20:10:14][D][lg-controller:849]: sending A8.02.00.00.00.01.12.22.40.00.80.00.CA (13)
[20:10:14][W][component:237]: Component lg_controller.climate took a long time for an operation (514 ms).
[20:10:14][W][component:238]: Components should block for at most 30 ms.
[20:10:19][D][lg-controller:1354]: update
[20:10:19][D][lg-controller:946]: received A8.02.00.00.00.01.12.22.40.00.80.00.CA (13)
[20:10:19][D][lg-controller:962]: verified send
[20:10:25][D][lg-controller:1354]: update

JanM321, 2024-10-14:
> made another try, disconnected the LG wall controller and set slave to false. Now logs looks promising but the AC doesn't response to any state change. all the sensors in HA are unavailable. I hope this helps debugging this issue.

In this log file, the AC is not sending any messages and if it's also not responding to the controller, there's probably something wrong with the wiring, especially the signal one (yellow).

If the AC sends a message you'll see "received C..." instead of "received A..."

JanM321, 2024-10-14:
Looking at the log in the first message, I think your unit uses the older 6-byte protocol instead of the 13-byte one implemented here. There's more information on these two protocols in the README.

Unfortunately the 6-byte protocol is very different. It has an ESPHome implementation here: https://github.com/Flameeyes/esphome-lg-pqrcuds0

The hardware should be the same so you could try the code there.

mizrachiran, 2024-10-14:
Thank you so much. 
Indeed, it's the other protocol. I flashed the other firmware and it worked.
Thanks again! @JanM321

closed 2024-10-14 by mizrachiran

JanM321, 2024-10-15:
Nice, I'm glad that worked and that the hardware is compatible with both firmwares.