JanM321/esphome-lg-controller #82: Plug & Play kit?

ExquisiteMetropolis, 2024-11-07:
Not an issue, but a question from a fellow Dutchy.
I'd love to have this unit in my PC09ST unit(s), so it's not dependent on an internet connection to control it via Home Assistant.
Would it in any way be possible to order a 'plug & play' kit via @JanM321? 
So it would just be a matter of connecting the ESP to the unit & my wifi and go from there.

JanM321, 2024-11-07:
>  Would it in any way be possible to order a 'plug & play' kit via @JanM321?

No I'm not offering this, but sometimes people have spare PCBs available. See some of the recent issues.

closed 2024-11-17 by JanM321

ExquisiteMetropolis, 2024-12-12:
I'm attempting to do it myself, but when I follow the steps to upload the several files to JLCPB, there is a shortage on parts, or no longer available. Is it possible to have the bom/placement file updated with parts that are on-stock/available?