JanM321/esphome-lg-controller #85: Compatibility with Ducted heating

Oxygen54, 2024-11-14:
Hi,

I have a ducted heating and want to know is this project could work with these :

- LG UUD3 U30 (Heat Pump outdoor unit)
- LG UM36F N20 (HVAC indoor unit)
- LG PREMTB100 / RS3 (As Master/Slave controller)

All of these 3 units are connected to the same bus and the actual system is this : 

[image]

In red the module output and green the LG bus.The IntesisBox LG-RC-MBS-1 apparently convert the LG signal to modbus RTU (EIA485)

Did I need to use this converter ? Can I use the LG signal with this project ?

Thanks

JanM321, 2024-11-15:
If you use the LG PREMTB100 today, then just replacing that controller with the ESPHome controller should work. You could probably also use both the PREMTB100 and the ESPHome controller, but most users don't have a Master/Slave setup so I can't say how well it would work especially with 3 controllers.

Keep in mind that this controller doesn't support zone control yet, so if you use that on the PREMTB100 it won't be available in HA.

I don't know anything about the relay board or duct units in general.

Oxygen54, 2024-11-17:
Yes, I'll handle the zone control on my own, but my main question is about the RS485. If I use this project and modify the communication to adapt it for the RS485 module, should it work? I saw on some forums that to use the RS485 module, it was simply a matter of adapting the communication.

qurpi, 2024-11-17:
It does work with ducted systems, just replaced mine ([here](LVN241HV4 and PREMTB100 (source: https://github.com/JanM321/esphome-lg-controller/discussions/86))

Anoxym, 2024-11-19:
Thanks for the answer :) , I need to find how to modify the code to use RS495