JanM321/esphome-lg-controller #88: Tiny PCB and backfeding

rrooggiieerr, 2024-11-22:
Could it be that the Tiny PCB can't handle backfeeding? I plugged in a USB in my ESP to do some debugging while power also being connected trough the Tiny PCB and it seems something in the Tiny PCB died. When removing the USB the ESP won't respond anymore

JanM321, 2024-11-22:
It's possible the ESP32 board died because they support being powered from 3.3V/5V/USB but not more than one of these:

https://docs.espressif.com/projects/esp-idf/en/v5.1/esp32/hw-reference/esp32/get-started-devkitc.html
> **Warning**
> The power supply must be provided using one and only one of the options above, otherwise the board and/or the power supply source can be damaged.

rrooggiieerr, 2024-11-22:
No, the ESP32 still works

rrooggiieerr, 2024-11-22:
I still have 6 of the Tiny PCB boards laying around, so not really a big problem for me. But maybe add a warning in the documentation not to wire both USB and CN_REMO

closed 2025-02-12 by rrooggiieerr