JanM321/esphome-lg-controller #89: Any available units?

MaikaiLife, 2024-12-05:
Before I go and purchase all the equipment to create these devices, I'd like to see if anyone has 8 of these.  I have 8x LCN097HV4 indoor cassettes that I need to be able to remote control.

Please let me know if you have some of these available and what you would want for them.  

Thank you!

MaikaiLife, 2024-12-10:
@pinhao @katshup @florianbrede-ayet @JanM321 

I've gone ahead and purchased some units, but I'm wondering since the LMCN077HV is on the list of supported devices if anyone could drop me a picture of what the connection looks like on the indoor unit so I know if I should get some sort of connectors such as the JST connectors?

ExquisiteMetropolis, 2024-12-14:
How did you mange to order them? I'm looking for 2 of these.

MaikaiLife, 2024-12-16:
I ordered them through JLCPB using these instructions.

https://docs.easyeda.com/en/PCB/Order-PCB/

One of the parts wasn't available so we had to swap out a part.

[image]
[image]
[image]

MaikaiLife, 2024-12-20:
I can tell you on my ceiling cartridges they have this little yellow connection block.  My plan is to connect right into that block with these connected to the boards.  https://www.amazon.com/dp/B0D4M6MTFT?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1
[image]

ExquisiteMetropolis, 2024-12-20:
[image]

Mine looks like this. Not sure where to plug the cable. The Amazon link you provided, I can't find the same thing in my area. Only cables with a plug on one end and loose wires on the other. I'm not into soldering or something. Plug & play would be nice. :-)

ExquisiteMetropolis, 2024-12-31:
I've received the 5 units. 
I'm going to use 2, and keep 1 as spare. 
Still waiting on my ESP's and need to figure out on how to connect it to my AC board... (see above) anyone have tip?

If anyone is interested in either of the additional units I have, let me know.
Note tested, no ESP included, no cabling), just the unit itself. 
Price: Not looking to make a profit on them. You pay what I paid for a single unit + shipping to whereever you are.

Scootyb, 2025-01-15:
If you still have the 2 units i would love to buy them from you.

ExquisiteMetropolis, 2025-01-17:
@Scootyb , yes I still have 2 units available. Please provide means of contact (preferably e-mail) and I'll reach out to you.

KrzysztofHajdamowicz, 2025-01-20:
> how to connect it to my AC board... (see above) anyone have tip?

I bought two set of 3pin JST XH2.54 connector ended with bare wires, like:
[image]

Then I soldered both cables, keep middle one connected to middle one and wire 1 to wire 3 and wire 3 to wire 1, in this case:
red -> yellow
black -> black
yellow -> red

MaikaiLife, 2025-01-20:
When I ordered my connectors on [Amazon](https://www.amazon.com/dp/B0D4M6MTFT?ref=ppx_yo2ov_dt_b_fed_asin_title&th=1), I removed the pins and moved them in the correct order so that all the colors were the same on both sides.  In other words, yellow = yellow, red = red, and black = black.  Mine had a very small tab you could press down gently with a tiny flathead screwdriver and slide the wire out of the connector to re-arrange them in the correct order.

[image]

robkap, 2025-01-24:
I also ordered 5 and only need 3. 
PM me if you're interested in cost sharing for my 2 extras.

ExquisiteMetropolis, 2025-02-09:
> > I also ordered 5 boards from JLCPCB and only need 3. PM me if you're interested in cost sharing for my 2 extras.
> 
> [@robkap](https://github.com/robkap) I'm interested in your remaining two boards; do you still have them? A single one would be good for a test as well. I'm based in the Netherlands. You can contact me at [duncan@projectfive.nl](mailto:duncan@projectfive.nl)

I also still have two available if needed.

Duncank, 2025-02-10:
> I also still have two available if needed.

@ExquisiteMetropolis thanks for the offer! If you can send me an email we can work this out!

gjongenelen, 2025-02-25:
I have 6 tiny pcbs available in NL. Contact me if you're interested via races_newt.7n@icloud.com

rajkundu, 2025-06-01:
I ordered 5 hardware-tiny boards from JLCPCB. Substituted C1 for [this 10µF capacitor](https://jlcpcb.com/partdetail/Dmbj-RVT1C100M0405_10UF16V/C970654) that was in stock instead, but just got everything set up and can verify that it works!

I'm using 2 of the boards and am happy to sell the other 3 at-cost ($13.46) + shipping! Price includes a few pairs of [3-pin JST XH cables from Amazon](https://www.amazon.com/dp/B09JP22SNK). I recommend using the same trick as @MaikaiLife mentioned above to make the cable colors match.

Please email me with your shipping address at raj@kundu.io if interested!

Hoekr0, 2025-12-06:
> > I ordered 5 hardware-tiny boards from JLCPCB. Substituted C1 for [this 10µF capacitor](https://jlcpcb.com/partdetail/Dmbj-RVT1C100M0405_10UF16V/C970654) that was in stock instead, but just got everything set up and can verify that it works!
> > I'm using 2 of the boards and am happy to sell the other 3 at-cost ($13.46) + shipping! Price includes a few pairs of [3-pin JST XH cables from Amazon](https://www.amazon.com/dp/B09JP22SNK). I recommend using the same trick as [@MaikaiLife](https://github.com/MaikaiLife) mentioned above to make the cable colors match.
> > I'm based in Boston, MA, USA. Please email me with your shipping address at [raj@kundu.io](mailto:raj@kundu.io) if interested!
> 
> [@Hoekr0](https://github.com/Hoekr0) I still have all 3 of these boards available. If you want a few, see what the cheapest shipping you can find from Boston, MA, USA to NL. E.g., I'm seeing around $30 USD for shipping US to NL + $26.92 for 2 boards = $56.92 USD for 2 boards from me. Let me know if you are interested, but absolutely no worries if not!

Hi @rajkundu Thanks for your offer, but I think there will be customs clearance fees, and in that case, I'd almost be better off ordering directly from JLCPCB. I hope you can make someone else happy with it in the US!

Hoekr0, 2025-12-06:
> > I can also order but because of custums and shipping it will be a bit "expensive" for me. I'm not completely familiar with this yet, so whether I'll get it working is still a question, and that's why I think it's an "expensive" gamble for me.
> 
> Unfortunately I ordered 5 units two weeks ago. I already need all of the units. I hope you find more people in the Netherlands or somewhere in the EU. for me it was around 70 euros including customs and shipping to NL.

Ai That's unfortunate to hear. If you know anyone else who wants something, please get in touch. Now that I'm working on it, I would have preferred to have them yesterday :-P, but I'm really not in a hurry.

The price on the website was over €90. I think the PCBs cost just over €50, so that's €40 for shipping and customs clearance. I think that's a bit much for something I might not even get working. I also have the option of installing a TADO system that works via IR for €60. It's guaranteed to work, but I'd rather not have one. If there really is no other option, I could also buy the LG Wi-Fi module myself, which I think is around €100 now.

So if I can split the cost with someone, I'd be willing to take the risk and order them. 5 or 10, depending on demand.