JanM321/esphome-lg-controller #91: How to install it on ESPHome on Home Assistant

Anoxym, 2024-12-21:
Hi,

I try to install the code to my ESP32 via ESPHome on Home Assistant but there is error : Component not found: climate.
I put base.yaml in **/homeassistant/esphome/custom_components/lg_controller** with this : 

```
external_components:
  - source:
      type: local
      path: /config/esphome/custom_components/lg_controller
```

climate.py & lg-controller.h in **/homeassistant/esphome/custom_components/lg_controller** too.

template.yaml as a new device in ESHome Builder with :  `<<: !include ../esphome/custom_components/lg_controller/base.yaml` to include base.yaml

What is wrong ?

Thanks

Anoxym, 2024-12-24:
Nobody have an idea?

JanM321, 2024-12-24:
Is there a reason you can’t use the instructions in the README?

This describes an alternative option: https://github.com/JanM321/esphome-lg-controller/discussions/55

Anoxym, 2024-12-24:
Sorry, the issue was actually that I hadn't created a components folder in ESPHome, and I had placed lg_controller in custom_components.

closed 2024-12-24 by Anoxym