JanM321/esphome-lg-controller #92: ESP32-S3-WROOM-1U

Anoxym, 2024-12-22:
Hi,

I have an ESP32-S3 and I would like to know if I can use the TX and RX pins on GPIO43 and GPIO44? Will this cause any issues? I understood that it's better to use GPIO17 and GPIO18, but they are already in use. 

[image]

JanM321, 2024-12-22:
I can't say for sure but it's probably fine based on this picture.

Anoxym, 2024-12-22:
ok thanks ;) I thought it was better not to use TX/RX but rather GPIOs to control the high or low state during signal transmission.

Anoxym, 2024-12-26:
Unknown as state tells that the serial link doesn't works ? The cables are connected behind the PREMTB100.

[image]

JanM321, 2024-12-26:
You should look at the ESPHome logs to see what's going on. You changed the GPIO numbers in the YAML file right?

JanM321, 2024-12-26:
> Did I need to put the PREMTB100 in slave mode ?

The controller doesn't work well if there are multiple 'master' controllers but you should still get sensor data etc in Home Assistant so it's likely unrelated.

Anoxym, 2024-12-27:
Ok thanks! I only have this in log : 

[01:21:48][D][lg-controller:1354]: update
[01:21:48][D][lg-controller:1450]: line busy, not sending yet

JanM321, 2024-12-27:
It’s trying to send messages to the AC but not receiving anything, so something with the hardware side isn’t right.

Anoxym, 2024-12-27:
Maybe GPIO43/44 doesn't work with the code ? The only difference is that I didn't connect the 3.3V to the ESP since it's already powered. I connected the LIN TX to TX GPIO43 and the LIN RX to RX GPIO44.

Anoxym, 2024-12-27:
I have changed the wiring and now I have : 

```
[16:28:33][D][lg-controller:1354]: update
[16:28:33][D][lg-controller:1450]: line busy, not sending yet
```

lagomCat, 2025-06-07:
> Unknown as state tells that the serial link doesn't works ? The cables are connected behind the PREMTB100. And I did not connect the 3.3V to the ESP as it has its own power supply, I only connected TX, RX, and GND to the ESP.

Am I understanding correctly that you are not using the TJA1027 transceiver as suggested in the project, but instead found the Rx and Tx points on the LG wired wall controller PCB and connected the ESP32 directly to them?
If that's the case, please note that those lines likely operate at 5V logic levels, while the ESP32 requires 3.3V logic. This voltage mismatch could potentially cause communication issues or even damage the ESP32.