JanM321/esphome-lg-controller #93: Little help needed to flash the ESPHome

ExquisiteMetropolis, 2025-01-04:
So, I've got all components now. ESPHome module, TinyPCB and a cable to connect it to the LG ac.
I've hooked the ESP Home module onto the TinyPCB and connected it via USB to my Windows 11 PC.
Within Home Assistant on my Raspberry Pi, I already had ESPHome Device Builder running to update ESPHome's in my environment (those came pre-flashed). 

I've updated template.yaml to suit my needs.
When I click install it gives the error that 'base.yaml' isn't to be found.
So I missed the first step 'clone the repository'. However, I don't see an option for that within Home Assistant or the ESPHome Device Builder website (that comes with the addon). 

Who can point me in the right direction on how to clone the repository so I can flash the ESPHome.

JanM321, 2025-01-04:
>  So I missed the first step 'clone the repository'. However, I don't see an option for that within Home Assistant or the ESPHome Device Builder website (that comes with the addon).

'cloning' refers to downloading the source code in this GitHub repository with a Git client, eg `git clone <url>` in a console. You can also use Code => Download ZIP on this website to download the latest version as a ZIP file.

There are alternative installation instructions here, if you prefer a non-terminal workflow: https://github.com/JanM321/esphome-lg-controller/discussions/55

ExquisiteMetropolis, 2025-01-05:
Thanks, that worked for me. Was able to flash the config onto the ESPHome.
Next issue that now arises, is that the cable advice I got from another topic, was inacurate for my LG unit (LG PC09ST NSJ).
Either side of this connector won't fit into this outlet (seems the only logical one to plug the unit into).

[image]
[image]

https://www.amazon.nl/dp/B0CS6ZP7PV/ref=pe_28126711_487805961_TE_item?th=1&language=en_GB

Which cable should I purchase?

porly1985, 2025-01-07:
I used simple [Jumper Cable](https://www.amazon.de/AZDelivery-Jumper-Arduino-Raspberry-Breadboard/dp/B074P726ZR/ref=sr_1_1_sspa?__mk_de_DE=%C3%85M%C3%85%C5%BD%C3%95%C3%91&crid=35QYKKNOZMU3J&dib=eyJ2IjoiMSJ9.QMvsCv-OL1-GLDfzEN_1jwfE4gsdlUXTnngNmuBV_fDhpS6-fMJHqZUky1RFhiNP-l-1g6NfDHb3QoDrt39qc-pQD5SxPlkv9YtLxBr7Tdhfv6mmLYf8pLRsc-BxXcQ8t4o6MB7PVXeGquPhfpifVW18eU0CbPWXg3iEoAc6Ln3WYjkGpl14V7v8atS9Zd6c79b2YlyQIPnr4knJm2DUmGu14K2RvRzhEMrZoNbANr0.dr9v_tilDAZGvB0dvLRkAfv9VeW-HZ_DHmzTWs_w8iM&dib_tag=se&keywords=ESP%2BKabel&nsdOptOutParam=true&qid=1736283604&sprefix=esp%2Bkabel%2Caps%2C102&sr=8-1-spons&sp_csd=d2lkZ2V0TmFtZT1zcF9hdGY&th=1). Both sides Female. Fits to the indoor unit and to PCB.

ExquisiteMetropolis, 2025-01-17:
In the end I used the jumper cables as well. Duced taped it to the connector.
All is working now.

[image]

@JanM321 you can add the LG PC09ST NSJ to the compatibility list. :)

closed 2025-02-09 by ExquisiteMetropolis