JanM321/esphome-lg-controller #94: Feature Request

dncbert, 2025-01-08:
I have been using your LG Controller SW for over a year now and it works great. Appreciate the work you put into this project.

I use Home Assistant to control all of my HVAC units. This includes a Mitsubishi mini split heat pump, a conventional heat pump system, and an LG mini split heat pump (your controller SW). I am including a screenshot from the Mitsubishi Climate Card and the LG Climate Card. On the Mitsubishi card it shows the current system "status" in the center (circled in red). On the LG card it shows the system "mode" in the center. The "status" shows "Heating" or "Idle" in heat mode, and "Cooling" or "Idle" in cooling mode. i.e. it tells you whether the system is actually "running". My conventional heat pump card also functions this way.

So, question here is whether your LG SW could be updated to publish/update the "status".  I believe the ESPHome Climate attribute is called "action". The following is from this ESPHome Climate page... https://esphome.io/components/climate/index.html#base-climate-configuration. If I were to try to implement this myself, would the best route be to utilize the status of the outdoor unit to set the "action" attribute?

// Current action (current on idle, cooling, heating, etc.), ClimateAction (enum)
id(my_climate).action

[image]

Thanks for your consideration.

JanM321, 2025-01-08:
Nice to hear from you again. I remember because you filed the very first issue (#1) after I published this :)

The code doesn't currently set the `ClimateAction` because we can't really tell with the LG units if they're actively cooling. We could use the status of the outdoor unit but that's not correct for multi-split units because the outdoor unit could be active for another indoor unit. In heating mode we *can* distinguish this with the 'preheating' flag (that's set to 'on' if the indoor unit is idle) but unfortunately for cooling mode there's no difference in status.

The best we can do is use the Mode and use Idle if `(Cooling || Heating) && (OutdoorUnit == Off || Preheating)` or something. I'm not sure if the 'Drying' mode turns the outdoor unit on - does anyone know? I've never used that mode :)

That still has the multi-split issue but maybe that's acceptable.

dncbert, 2025-01-08:
Ah, that makes sense. I only have a single unit system so I did not think of that multi-unit reality. After posting this I recognized that mini splits by design are generally running. They ramp up and down as the demands require vs. a conventional forced air system where it's essentially either conditioning or not (on/off). It actually makes me wonder what the Mitsubishi controller is doing. I guess it would be interesting if they are just setting the "action" to the "mode" and I just never noticed the difference. :) With a mini-split I imagine the only time the system ever really goes "idle" is if the room condition is satisfied with no load. So, for example, in the middle of the day during the winter, our sunroom will often be warmer than the set point due to solar gain, at which point the mini split does turn off (The outdoor unit for sure turns off. It seems like the indoor unit also goes into a sleep like state but perhaps the fan is still running slowly). In any case, I am not sure about the "drying" mode as I also have never used that setting. :)  Just throwing out another idea... is there any way to measure the indoor unit fan speed such that if the set point is satisfied and the fan speed is on the lowest setting than you infer it is effectively "idle".

lagomCat, 2025-06-07:
Drying mode is essentially the same as cooling mode, but in theory it's tuned so that the indoor unit stays closer to the dew point temperature more consistently, allowing for more efficient moisture removal (condensation).
At the same time, it is designed to avoid significantly lowering the room temperature.

lagomCat, 2025-06-07:
> I'm not sure if the 'Drying' mode turns the outdoor unit on - does anyone know? I've never used that mode :)

Drying mode is essentially the same as cooling mode, but in theory it's tuned so that the indoor unit stays closer to the dew point temperature more consistently, allowing for more efficient moisture removal (condensation), without significantly lowering the room temperature.
So yes, the outdoor unit does turn on in this mode.