JanM321/esphome-lg-controller #96: Compatibility with E12SQ

ricard0g0mes, 2025-01-12:
I'm trying to use this with my AC.
It's the LG E12SQ and it has the JST connector.
But in the power supply in the connector, i nesure about 30V instead of the mensioned 12V in the documentation.
Is that normal?

I have the PCB (hardware-tiny) assembled by JLCPCB, but it doen't work.
Could this model have different settings, supply...?

JanM321, 2025-01-13:
I'd expect the E12SQ to be compatible. You flashed the firmware? Does the ESP32 power on?

ricard0g0mes, 2025-01-13:
Yes and yes. I've used a M5 atom (connected to pinout in the PCB) and changed to the correct gpios of the m5. I also activated the webpage and it shows all the entities, but always trying to connect and failing. I will try with the esp32 dev board. But it should work with this...

ricard0g0mes, 2025-01-20:
Hello.
It's working. I think it was a faulty pcb from JLCPCB
By the way, i used esphome builder embebbed in home assistant.
If anyone needs it, here is the code:

`external_components:
  # use rtttl and dfplayer from ESPHome's dev branch in GitHub
  - source:
      type: git
      url: https://github.com/JanM321/esphome-lg-controller
      ref: dev
    components: [lg_controller]

substitutions:
  device_name: escritorio
  device_type: ac_lg
  device_id: ${device_type}_${device_name}
  
  # The input temperature sensor in HA.
  temperature_sensor_entity_id: sensor.escritorio_th_temperature # XXX
  
  # Set to "true" if the temperature sensor and Home Assistant use Fahrenheit instead of Celsius.
  fahrenheit: "false" # XXX

  # If you're connecting a single controller, this should be set to "false".
  is_slave_controller: "false"
  
esphome:
  name: ${device_id}

esp32:
  board: esp32dev
  framework:
    type: arduino
    #type: esp-idf
    
# Enable logging
logger:

# Enable Home Assistant API
api:
  encryption:
    key: "XXXXXXXXXXXXXXXX"

ota:
  - platform: esphome
    password: "XXXXXXXXXXXXXXXXXXXXXXX"

wifi:
  ssid: !secret wifi_ssid
  password: !secret wifi_password

  # Enable fallback hotspot (captive portal) in case wifi connection fails
  ap:
    ssid: "Ac-Lg-Esc Fallback Hotspot"
    password: "crz9FrYsVXNH"

captive_portal:

# Enable Web server (optional).
web_server:
  port: 80

uart:
  - id: ac_serial
    tx_pin: GPIO25
    rx_pin:
      number: GPIO26 # Keep in sync with rx_pin below.
      allow_other_uses: true
    baud_rate: 104

time:
  - platform: homeassistant
    id: ${device_id}_homeassistant_time

sensor:
  - id: temp_sensor
    platform: homeassistant
    internal: true
    entity_id: ${temperature_sensor_entity_id}  
  # Uptime sensor.
  - platform: uptime
    name: ${device_id} Uptime

  # WiFi Signal sensor.
  - platform: wifi_signal
    name: ${device_id} WiFi Signal
    id: ${device_id}_${device_id}_wifi_signal_db
    update_interval: 60s

  - platform: copy # Reports the WiFi signal strength in %
    source_id: ${device_id}_${device_id}_wifi_signal_db
    name: "${device_id} WiFi Signal Percent"
    filters:
      - lambda: return min(max(2 * (x + 100.0), 0.0), 100.0);
    unit_of_measurement: "Signal %"
    entity_category: "diagnostic"
    device_class: "" 

climate:
  - platform: lg_controller
    id: ${device_id}
    name: ${device_type} ${device_name}
    uart_id: ac_serial
    rx_pin:
      number: GPIO26 # Keep in sync with rx_pin above.
      allow_other_uses: true
    fahrenheit: ${fahrenheit}
    is_slave_controller: ${is_slave_controller}
    temperature_sensor: temp_sensor
    vane1:
      name: Airflow 1 Up/Down
      id: ${device_id}_vane_position_1
      icon: mdi:arrow-up-down
    vane2:
      name: ${device_id} Airflow 2 Up/Down
      id: ${device_id}_vane_position_2
      icon: mdi:arrow-up-down
    vane3:
      name: ${device_id} Airflow 3 Up/Down
      id: ${device_id}_vane_position_3
      icon: mdi:arrow-up-down
    vane4:
      name: ${device_id} Airflow 4 Up/Down
      id: ${device_id}_vane_position_4
      icon: mdi:arrow-up-down
    overheating:
      name: ${device_id} Over Heating
      id: ${device_id}_overheating
      icon: mdi:heating-coil
    fan_speed_slow:
      name: ${device_id} Fan Speed Slow
      id: ${device_id}_fan_speed_slow
      icon: mdi:fan-chevron-down
      mode: box
    fan_speed_low:
      name: ${device_id} Fan Speed Low
      id: ${device_id}_fan_speed_low
      icon: mdi:fan-speed-1
      mode: box
    fan_speed_medium:
      name: ${device_id} Fan Speed Medium
      id: ${device_id}_fan_speed_medium
      icon: mdi:fan-speed-2
      mode: box
    fan_speed_high:
      name: ${device_id} Fan Speed High
      id: ${device_id}_fan_speed_high
      icon: mdi:fan-speed-3
      mode: box
    sleep_timer:
      name: ${device_id} "Sleep Timer (minutes)"
      id: ${device_id}_sleep_timer
      icon: mdi:timer-outline
      mode: box
    error_code:
      name: ${device_id} Error Code
      id: ${device_id}_error_code
      icon: mdi:alert-circle-outline
    pipe_temp_in:
      name: ${device_id} Pipe Temperature In
      id: ${device_id}_pipe_temperature_in
      icon: mdi:thermometer
      unit_of_measurement: "°C"
    pipe_temp_mid:
      name: ${device_id} Pipe Temperature Mid
      id: ${device_id}_pipe_temperature_mid
      icon: mdi:thermometer
      unit_of_measurement: "°C"
    pipe_temp_out:
      name: ${device_id} Pipe Temperature Out
      id: ${device_id}_pipe_temperature_out
      icon: mdi:thermometer
      unit_of_measurement: "°C"
    defrost:
      name: ${device_id} Defrost
      id: ${device_id}_defrost
      icon: mdi:snowflake-melt
    preheat:
      name: ${device_id} Preheat
      id: ${device_id}_preheat
      icon: mdi:heat-wave
    outdoor:
      name: ${device_id} Outdoor Unit
      id: ${device_id}_outdoor
      icon: mdi:fan
    auto_dry_active:
      name: ${device_id} Auto Dry Active
      id: ${device_id}_auto_dry_active
      icon: mdi:fan-clock
    purifier:
      name: ${device_id} Air Purifier
      id: ${device_id}_air_purifier
      icon: mdi:pine-tree
    internal_thermistor:
      name: ${device_id} Internal Thermistor
      id: ${device_id}_internal_thermistor
      icon: mdi:thermometer
    auto_dry:
      name: ${device_id} Auto Dry
      id: ${device_id}_auto_dry
      icon: mdi:fan-clock    `

JanM321, 2025-01-25:
I'm glad it works now.

closed 2025-01-25 by JanM321