JanM321/esphome-lg-controller #99: Problem with the hardware-tiny

LakiTom, 2025-02-24:
Good morning, in the meantime I would like to congratulate you on the project, and for more than a year I have been following it and now I have decided to try it, I would just like to expose you to the problems I am experiencing:

1) I had hardware built by jlcpcb.com using the files as per the guide. I had to replace a couple of unattainable components, I follow the list of components:

[bom.csv](https://github.com/user-attachments/files/18942490/bom.csv)

Here the first problem arose, the TPS82130SILR once connected to REMO, did not convert 12v to 3v. Not having previously checked the first ESP32 card I burned it immediately.
Later I checked the problem and discovered that TPS82130SILR could not convert the volts, overheated and stopped working leaving the 12v board powered.
So I tried to find a solution, and I took TPS82130SILR out of a board and replaced it with a 3volt converter.

2) I configured esp32 as per project, it turns on home assistant recognizes it, but does not receive and transmits the signal, while the bluetooth proxy works well.

Adesso sono bloccato sul problema di come risolvere la ricezione e trasmissione del segnale, sempre se la modifica che ho fatto sulla scheda non sia un problema.

ESP32: ESP32-DevKitC-32E
Hardware: hardware-tiny
AC: LG S12ET.NSJ

[image]
[image]

d4m14ndx, 2025-02-26:
Strange, I had a set built this week by them. the only part i had to substitute from memory is the C1 capacitor, but they had an identical one in stock.

One thing I noticed when installing it, is the cable to the remote had the 12v + and - wires switched, so i just made up a quick patch / Y  cable to swap them back and everything worked fine including control.

LakiTom, 2025-03-01:
> Strange, I had a set built this week by them. the only part i had to substitute from memory is the C1 capacitor, but they had an identical one in stock.
>
> One thing I noticed when installing it, is [...]

Hi @d4m14ndx, I think I also had the same replacement for the C1, and even for me the cables were inverted. So we have had the same experience so far, easily solvable, no problem. I however, once connected i had the problems written above. I really don't know why, in fact I was hoping that someone who knew about electronics could tell me. Anyway, once the voltage is lowered, esp32 works, the only problem is that it doesn't read the data, it won't reach it, I don't know. He tries to send them when I change settings on home assistant, but there is no feedback.
@d4m14ndx, If it were possible for you, would you like to sell me one of yours so I can see if mine are defective?

d4m14ndx, 2025-03-02:
My Boards are converting the 12V properly.

One thing i learned later, is that unless you are specifically buying the ESP32 Devkit from Espressif directly, lots of people are making very similar but slightly different dev boards. For instance, mine was the Duinotech branded ones, and the header pins were about 3mm narrower than the Espressif devkit. 

Did you double check all of your pins on your ESP32 board are exactly the same as what the tinyboard scematic assumed? Are you able to post a photo of your ESP32?

LakiTom, 2025-03-16:
Sorry for my late reply and thank you for your help.

I bought the written card in the project precisely because I am ignorant about the subject and I didn't want to make mistakes, I attach the photos and the link to the documentation. 

[https://docs.espressif.com/projects/esp-dev-kits/en/latest/esp32/esp32-devkitc/user_guide.html](url)

[image]

[image]

Unfortunately I don't know how to diagnose the problem, I don't know the electronics so well to test the circuit that I had done to check for any errors. What I did is check the output of the air conditioner and arrange the cables for the input in the project circuit, then the 12 volts arrive inside the board correctly, but then the board does not convert the 12v to 3v, it tries lowering the voltage up to about 10v and then the component that I modified in the first post breaks. Changed in a rather ugly way that component a one found around that may not be adequate, the board comes out in 3v and esp32 works, except for the data signal.